RfidResearchGroup/proxmark3 #1495: [RELEASE 4.x] Checklist

doegox, 2021-09-03:
# Checklist

- [ ] CHANGELOG.md
- [ ] `make style`
- [ ] `make clean; make client CC=clang CXX=clang++ LD=clang++` on recent Debian or Ubuntu
- [ ] `mymanualchecks.sh`
- [ ] `mycppcheck.sh` no alarming warning?
- [ ] `mymakeclang.sh` no alarming error/warning ?
- [ ] `mystandalone_makes.sh` check that the script contains all standalone modes then compile all standalone modes (linux only)
- [ ] GitHub Actions - green across the board ( MacOS, Ubuntu, Windows)
- [ ] [Appveyor](https://ci.appveyor.com/project/RfidResearchGroup/proxmark3/history) green (PS)

# OS compilation and tests

```bash
#!/usr/bin/env bash

make clean && make -j PLATFORM=PM3GENERIC && tools/pm3_tests.sh --long
make clean && make -j PLATFORM=PM3RDV4 && tools/pm3_tests.sh --long
make clean && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && tools/pm3_tests.sh --long
sudo make install; pushd /tmp; proxmark3 -c 'data load -f lf_EM4x05.pm3;lf search -1'; popd; sudo make uninstall

( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3GENERIC && PM3BIN=./proxmark3 ../../tools/pm3_tests.sh client )
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4  && PM3BIN=./proxmark3 ../../tools/pm3_tests.sh client )
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && PM3BIN=./proxmark3 ../../tools/pm3_tests.sh client )
```

- [ ] RPI Zero
- [ ] WSL
- [ ] PSv3.9
- [ ] Archlinux
- [ ] Kali
- [ ] Debian
- [ ] Ubuntu20
- [ ] ParrotOS
- [ ] Fedora
- [ ] OpenSuse
- [ ] OSX
- [ ] Android
- [ ] Termux

# creating release
`make release RELEASE_NAME="ice awesome"`
last line of output,  gives you next command to run
Sample:  `git push && git push origin v4.15000`

Go to Github releases,  create release based on the new created tag and publish
update homebrew repo, file `proxmark3.rb` with a SHA256 sum of the file `v4.15000.tar.gz`

iceman1001, 2021-09-04:
I believe I got an updated version of the test build script from you @doegox   where it does exit 1.
if this is the prefered way,  I can update the release template to match it.

```
#!/usr/bin/env bash

make clean && make -j PLATFORM=PM3GENERIC && tools/pm3_tests.sh --long || exit 1
make clean && make -j PLATFORM=PM3RDV4 && tools/pm3_tests.sh --long || exit 1
make clean && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && tools/pm3_tests.sh --long || exit 1
make install; pushd /tmp; proxmark3 -c 'data load -f lf_EM4x05.pm3;lf search -1'; popd; make uninstall

( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3GENERIC && PM3BIN=./proxmark3 ../../tool
s/pm3_tests.sh client --long ) || exit 1
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4  && PM3BIN=./proxmark3 ../../tools/
pm3_tests.sh client --long ) || exit 1
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && PM3BIN=.
/proxmark3 ../../tools/pm3_tests.sh client --long ) || exit 1

```

doegox, 2021-09-04:
ok I changed it here and in the template

iceman1001, 2021-09-04:
@msoose  mentioned on discord it compiles just fine some week ago

iceman1001, 2021-09-04:
@Gator96100  mentioned on discord it compiles just fin on ProxSpace 3.10

iceman1001, 2021-09-04:
@tcprst  will run it on OSX and on the M2

iceman1001, 2021-09-05:
@doegox   don't forget to drop a line in CHANGELOG.md for all your commits

doegox, 2021-09-05:
"gardening for next release"?
Nah it's ok for me

tcprst, 2021-09-06:
> @tcprst will run it on OSX and on the M2

Tested on Intel and Apple Silicon without issues

iceman1001, 2021-09-10:
ccache is nice!

iceman1001, 2021-09-18:
New release is published.
homebrew is updated.   Still need to update the standalone modes possible in that file. 
https://github.com/RfidResearchGroup/homebrew-proxmark3/blob/master/proxmark3.rb#L26-L27

Now, its time to create a list of new standalone files in this release,  to inform all package maintainers.,

iceman1001, 2021-09-18:
Changelog has been updated to reflect it.

closed 2021-09-18 by iceman1001