RfidResearchGroup/proxmark3 #1497: cppcheck iso14443b.c redundant assignment

doegox, 2021-09-03:
in iso14443b.c
```c
        if (tag_is_active == false) {

            if (Handle14443bSampleFromReader(ci & 0x01)) {
                uint32_t eof_time = dma_start_time + (samples * 16) + 8; // - DELAY_READER_TO_ARM_SNIFF; // end of EOF
                if (Uart.byteCnt > 0) {
                    uint32_t sof_time = eof_time
                                        - Uart.byteCnt * 1 // time for byte transfers
                                        - 32 * 16          // time for SOF transfer
                                        - 16 * 16;         // time for EOF transfer
                    LogTrace(Uart.output, Uart.byteCnt, (sof_time * 4), (eof_time * 4), NULL, true);
                }
                // And ready to receive another command.
                Uart14bReset();
                Demod14bReset();
                reader_is_active = false;<--- reader_is_active is assigned
                expect_tag_answer = true;
            }

            if (Handle14443bSampleFromReader(cq & 0x01)) {

                uint32_t eof_time = dma_start_time + (samples * 16) + 16; // - DELAY_READER_TO_ARM_SNIFF; // end of EOF
                if (Uart.byteCnt > 0) {
                    uint32_t sof_time = eof_time
                                        - Uart.byteCnt * 1 // time for byte transfers
                                        - 32 * 16          // time for SOF transfer
                                        - 16 * 16;         // time for EOF transfer
                    LogTrace(Uart.output, Uart.byteCnt, (sof_time * 4), (eof_time * 4), NULL, true);
                }
                // And ready to receive another command
                Uart14bReset();
                Demod14bReset();
                reader_is_active = false;<--- reader_is_active is assigned
                expect_tag_answer = true;
            }

            reader_is_active = (Uart.state > STATE_14B_GOT_FALLING_EDGE_OF_SOF);<--- reader_is_active is overwritten<--- reader_is_active is overwritten
        }
```

[image]

Straightforward is to remove first two assignments but maybe the logic should be to do the third assignment only if none of the *if* clause was followed ? In which case the third assignment should be put before the two *if* clauses.

iceman1001, 2021-09-04:
yeah... 14B...  a sad child of code...

In the particular case we can remove the two first assignments since the code isn't in great shape and we just wanna make cppchecker not complain

doegox, 2021-09-05:
ok, done

closed 2021-09-05 by doegox