RfidResearchGroup/proxmark3 #1500: Segmentation Fault when reading DESFire File

ninjas28, 2021-09-08:
**Describe the bug**
I get a segmentation fault when trying to read a file on a Mifare DESFire card (ORCA transit card). This card file can be read using an Android phone card reader app and works for use on the transit system (the file I'm trying to read stores current card balance).

**To Reproduce**
Steps to reproduce the behavior:
1. Place an ORCA card on the HF antenna.
2. run ```hf mfdes read --no-auth --aid f21030 --fid 02```
3. Observe a segmentation fault.

**Expected behavior**
The console outputs the stored bytes in that file.

**Screenshots**

```[usb] pm3 --> hf mfdes lsfiles --aid f21030 --no-auth -v
[=] Key num: 0 Key algo: des Key[8]: 00 00 00 00 00 00 00 00
[=] Secure channel: n/a Command set: niso Communication mode: mac
[=] AID f21030 is selected
[!!] Desfire GetFileISOIDList command error. Result: -20
[!] ISO ID list returned no data
[=] ------------------------------------------ File list -----------------------------------------------------
[+]  ID |ISO ID|     File type       | Mode  | Rights: raw, r w rw ch   | File settings
[+] ------------------------------------------------------------------------------------------------------------
[+]  05 |      |0x01 Backup data     | Plain |e432, free key4 key3 key2 | size: 32 [0x20]
[+]  00 |      |0x01 Backup data     | Plain |e432, free key4 key3 key2 | size: 160 [0xa0]
[+]  0f |      |0x00 Standard data   | MAC   |e432, free key4 key3 key2 | size: 416 [0x1a0]
[+]  02 |      |0x04 Cyclic Record   | Plain |e432, free key4 key3 key2 | record count 10/11 size: 48 [0x30]b
[+]  03 |      |0x04 Cyclic Record   | Plain |e432, free key4 key3 key2 | record count 5/6 size: 48 [0x30]b
[+]  04 |      |0x01 Backup data     | MAC   |e432, free key4 key3 key2 | size: 64 [0x40]
[+]  06 |      |0x01 Backup data     | Plain |e432, free key4 key3 key2 | size: 64 [0x40]
[+]  07 |      |0x01 Backup data     | Plain |e432, free key4 key3 key2 | size: 64 [0x40]
[usb] pm3 --> hf mfdes read --aid f21030 --fid 02 -m plain --no-auth -v
[=] Key num: 0 Key algo: des Key[8]: 00 00 00 00 00 00 00 00
[=] Secure channel: n/a Command set: niso Communication mode: plain
[=] AID f21030 is selected
[=] Got file type: Cyclic Record. Option: record. comm mode: plain
[=] ------------------------------- File 02 data -------------------------------
[=] Record length 48
pm3: line 249:   974 Segmentation fault      $CLIENT "$@"
```

**Desktop (please complete the following information):**
 - OS: Windows 10 20H2
 - hw version:
 ```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.13441-968-gbac58d2a6 2021-09-07 12:29:46
  compiled with MinGW-w64 10.3.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.13441-968-gbac58d2a6 2021-09-07 12:29:32
       os: RRG/Iceman/master/v4.13441-968-gbac58d2a6 2021-09-07 12:29:39
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 53% used )
```
 - hw status
```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 44520
[#]   Available memory........ 42212
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 1035
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 198656
[#]   Transfer Speed PM3 -> Client... 397312 bytes/s
[#] Various
[#]   Max stack usage......... 4120 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 48
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32417 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#]
```
 - data tune
```
[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
[|] 10
[=] ---------- LF Antenna ----------
[+] LF antenna: 27.21 V - 125.00 kHz
[+] LF antenna: 21.91 V - 134.83 kHz
[+] LF optimal: 27.88 V - 126.32 kHz
[+] Approx. Q factor (*): 7.2 by frequency bandwidth measurement
[+] Approx. Q factor (*): 8.1 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 15.05 V - 13.56 MHz
[+] Approx. Q factor (*): 4.4 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
```

doegox, 2021-09-08:
Hmm it might be hard to reproduce without the specific tag.
I see you're under Windows, is it doable to run the proxmark3 client via valgrind ?

merlokk, 2021-09-08:
exec please command with `-av` flags

merlokk, 2021-09-08:
try please the lastest source)

ninjas28, 2021-09-08:
Never heard of valgrind, but if it can run in Windows or WSL I can try it.

Tried running the command again with -av flags on my current version and it seems to be able to read some of the data but still gives a segmentation fault. I will try again later after I rebuild proxmark.
```
[usb] pm3 --> hf mfdes read --aid f21030 --fid 02 --no-auth -av
[=] Key num: 0 Key algo: des Key[8]: 00 00 00 00 00 00 00 00
[=] Secure channel: n/a Command set: niso Communication mode: mac
[+] Setting ISODEP -> inactive
[+] >>>> 90 5A 00 00 03 30 10 F2 00
[+] Setting ISODEP -> inactive
[+] Setting ISODEP -> NFC-A
[+] <<<< 91 00
[=] AID f21030 is selected
[+] >>>> 90 F5 00 00 01 02 00
[+] <<<< 04 00 32 E4 30 00 00 0B 00 00 0A 00 00 91 00
[=] Got file type: Cyclic Record. Option: record. comm mode: plain
[=] ------------------------------- File 02 data -------------------------------
[=] Record length 48
[+] >>>> 90 BB 00 00 07 02 00 00 00 00 00 00 00
[+] <<<< 10 90 00 76 13 2A DD 48 00 02 5A A8 60 12 00 02 8A 0C 83 4F FF F9 00 0D C0 E3 60 0F 52 70 E0 10 01 00 21 98 00 00 00 00 00 00 00 00 00 00 00 00 10 90 00 76 13 2B CE 9F B0 03 38 91 AF
[+] >>>> 90 AF 00 00 00
[+] <<<< B8 60 12 00 00 00 03 FF FE 57 DA 80 0B 42 72 8F FF FF FF F0 10 01 00 21 98 00 00 00 00 00 00 00 00 00 00 00 00 10 90 00 76 13 2B FA AF B0 02 87 08 60 12 00 00 00 07 FF FE 06 AA 91 AF
[+] >>>> 90 AF 00 00 00
[+] <<<< 80 0B 42 83 9F FF FF FF F0 10 01 00 21 98 00 00 00 00 00 00 00 00 00 00 00 00 10 90 00 46 13 77 D4 D8 00 01 CB 78 60 12 00 02 26 0C 84 4B FF FD 00 0D C5 36 10 0F 61 B7 50 10 01 91 AF
[+] >>>> 90 AF 00 00 00
[+] <<<< 00 20 85 00 00 00 00 00 00 00 00 00 00 00 00 10 90 00 76 13 78 AA 98 00 02 55 A8 60 12 00 00 64 0C 80 F3 FF FF 80 0D C5 53 E0 0F 1B 77 20 10 01 00 20 53 00 00 00 00 00 00 00 00 91 AF
[+] >>>> 90 AF 00 00 00
[+] <<<< 00 00 00 00 10 90 00 76 13 80 7A 88 00 02 55 A8 60 12 00 02 8A 0C 80 F3 FF FF 80 0D C5 53 E0 0F 1B 77 F0 10 01 00 1F 0E 00 00 00 00 00 00 00 00 00 00 00 00 10 90 00 46 13 80 E2 91 AF
[+] >>>> 90 AF 00 00 00
[+] <<<< 88 00 01 CA 58 60 12 00 00 00 0C 84 D7 FF FD 00 0D C5 51 90 0F 61 BD 00 10 01 00 1F 0E 00 00 00 00 00 00 00 00 00 00 00 00 10 90 00 76 13 81 47 B8 00 02 5A 28 60 12 00 00 00 0C 91 AF
[+] >>>> 90 AF 00 00 00
[+] <<<< 84 FD FF F9 00 0D C0 BF 40 0F 62 02 60 10 01 00 1F 0E 00 00 00 00 00 00 00 00 00 00 00 00 10 90 00 46 13 8C EB 18 00 01 C8 78 60 12 00 02 26 0C 84 4B FF FD 00 0D C5 30 80 0F 61 91 AF
[+] >>>> 90 AF 00 00 00
[+] <<<< B7 50 10 01 00 1D FB 00 00 00 00 00 00 00 00 00 00 00 00 10 90 00 76 13 8D C6 D8 00 02 55 B8 60 12 00 00 64 0C 80 F1 FF FF 80 0D C0 F3 A0 0F 1B 75 D0 10 01 00 1D C9 00 00 00 00 91 AF
[+] >>>> 90 AF 00 00 00
[+] <<<< 00 00 00 00 00 00 00 00 91 00
pm3: line 249:  1891 Segmentation fault      $CLIENT "$@"
```

ninjas28, 2021-09-08:
It's fixed using the latest commits.

closed 2021-09-08 by merlokk

iceman1001, 2021-09-09:
I believe the DESFire Ev2 and Ev3 has large files support,  to  65kb.
We should ask how large the file is before reading/download it,   and dynamically allocate the needed size.