RfidResearchGroup/proxmark3 #1503: An improper locking bug(e.g., deadlock) on the lock ctx->queue_mutex

ryancaicse, 2021-09-11:
Hi, developers, thank you for your checking. It seems the lock **ctx->queue_mutex** is not released correctly when **ret != QUEUE_EMPTY** in the function **wu_queue_destroy**?

@matrix @doegox

https://github.com/RfidResearchGroup/proxmark3/blob/9da0073e0dd9758cece4bf215654c4550929c971/tools/hitag2crack/crack5opencl/queue.c#L374-L385

matrix, 2021-09-11:
Hi @ryancaicse,  you got an error here?
If you add "pthread_mutex_unlock(&ctx->queue_mutex);" before "return ret;" works?
Thanks :)

matrix, 2021-09-11:
Also the wu_queue_pop() function lock the mutex, so another way could be to move the wu_queue_destroy() lock later. Let me know

ryancaicse, 2021-09-12:
So, there could be two errors? 

The first is **ctx->queue_mutex** is not released before the function returns.
https://github.com/RfidResearchGroup/proxmark3/blob/9da0073e0dd9758cece4bf215654c4550929c971/tools/hitag2crack/crack5opencl/queue.c#L374-L386

The seond is **ctx->queue_mutex** could be reacquired in the function **wu_queue_pop**.
https://github.com/RfidResearchGroup/proxmark3/blob/9da0073e0dd9758cece4bf215654c4550929c971/tools/hitag2crack/crack5opencl/queue.c#L288-L296

We would love to propose a PR to fix them but do know how to fix them can fit in the program semantics. Any suggestions?

matrix, 2021-09-12:
Why two? only one. I wrote you in the last comment what to do right?

matrix, 2021-09-12:
@ryancaicse please try with this patch. Most likely the mutex is not needed in wu_queue_destroy()

```
diff --git a/tools/hitag2crack/crack5opencl/queue.c b/tools/hitag2crack/crack5opencl/queue.c
index bce67520f..39e503bc1 100644
--- a/tools/hitag2crack/crack5opencl/queue.c
+++ b/tools/hitag2crack/crack5opencl/queue.c
@@ -371,7 +371,7 @@ int wu_queue_destroy(wu_queue_ctx_t *ctx) {
     if (!ctx) return ERROR_CTX_NULL;
     if (!ctx->init) return ERROR_CTX_IS_NOT_INIT;
 
-    pthread_mutex_lock(&ctx->queue_mutex);
+    //pthread_mutex_lock(&ctx->queue_mutex);
 
     int ret = -1;
 
@@ -398,7 +398,7 @@ int wu_queue_destroy(wu_queue_ctx_t *ctx) {
     ctx->queue_tail = 0; //NULL;
     ctx->init = 0;
 
-    pthread_mutex_unlock(&ctx->queue_mutex);
+    //pthread_mutex_unlock(&ctx->queue_mutex);
 
     pthread_mutex_destroy(&ctx->queue_mutex);
     pthread_mutexattr_destroy(&ctx->queue_mutex_attr);

```

ryancaicse, 2021-09-12:
@matrix Ok, thanks

iceman1001, 2021-09-14:
@matrix   are you making a PR for this one?

matrix, 2021-09-14:
hi @iceman1001 :) I will send it as soon as possible, thanks!

iceman1001, 2021-09-14:
No rush,  the new release will happen sometime this week if you want to have it in there.   Its not a critical bug.

PR #1510 (2021-09-19): fix crack5opencl queue/threads.c
fix #1503
closed 2021-09-19 by iceman1001