RfidResearchGroup/proxmark3 #1514: proxmark dumpin gin the wrong location and reading in the wrong location

cvanh, 2021-10-03:
**Describe the bug**
issue when running pm3 in terminal and then dumping.

when dumping a card in a shell opened with the pm3 command it dumps it to ~ but when it tries to read it it tries to find in in the root directory and the sub directory /client resulting in a error wich says cannot find x

**To Reproduce**
Steps to reproduce the behavior:
0. when installing run `make install`

0. run `pm3` in the home directory
1. run `hf mf autpwn` (yes im a idiot)
2. wait for it to finish 
3. run `hf mf restore {the name of the file that was dumped}`

**Expected behavior**
it dumps the files in 1 location and reads it there.

**Screenshots**
dont think this really fits

**Desktop (please complete the following information):**
 - OS: linux mint 20.2 kernel:  5.11.0-37-generic
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version:
 ` [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.14434-4-g85f8f2b65 2021-09-18 22:38:14
  compiled with GCC 9.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.13441-881-gce4eed2ac 2021-08-25 00:40:37
       os: RRG/Iceman/master/v4.13441-881-gce4eed2ac 2021-08-25 00:40:56
  compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 60% used )
`
 - hw status:
 `[#] Memory
[#]   BigBuf_size............. 42376
[#]   Available memory........ 42376
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD5697C30973E6F23
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 298496
[#]   Transfer Speed PM3 -> Client... 596992 bytes/s
[#] Various
[#]   Max stack usage......... 4096 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31284 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[#] 
`
 - data tune:
 `[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 71,04 V - 125,00 kHz
[+] LF antenna: 35,20 V - 134,83 kHz
[+] LF optimal: 71,54 V - 126,32 kHz
[+] Approx. Q factor (*): 12,1 by frequency bandwidth measurement
[+] Approx. Q factor (*): 12,5 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 47,20 V - 13.56 MHz
[+] Approx. Q factor (*): 8,2 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134,83 kHz, 95 (red) is 125,00 kHz.

`

**Additional context**
if you have any question regarding this please contact me on github or on discord `أكبر ذيب#7092`

cvanh, 2021-10-11:
i installed it with `make install` so i can open the proxmark client anywhere with the command `pm3`

iceman1001, 2022-02-05:
closing because of inactivity.  
there is already a issue about having user to specify where the default dump paths should be.

closed 2022-02-05 by iceman1001