RfidResearchGroup/proxmark3 #1533: Possible races issue

ryancaicse, 2021-11-19:
Hi, developers, thank you for your checking.
For this #1510, this PR is supposed to fix an improper locking bug (missing releases at **Line 385**) due to the unreleased lock by removing the lock `ctx->queue_mutex` entirely.

https://github.com/RfidResearchGroup/proxmark3/blob/9da0073e0dd9758cece4bf215654c4550929c971/tools/hitag2crack/crack5opencl/queue.c#L374-L401

But, after thinking for a while, would it better to move the lock **at Line 374** to Line **387**, becasue the lock `ctx->queue_mutex` should guard these operations to prevent multiple races.

https://github.com/RfidResearchGroup/proxmark3/blob/9da0073e0dd9758cece4bf215654c4550929c971/tools/hitag2crack/crack5opencl/queue.c#L388-L399

Notes: The method `wu_queue_pop` is already guarded by the lock `ctx->queue_mutex`.

iceman1001, 2021-11-20:
@matrix 
this one is in your domain.

matrix, 2021-11-20:
Hi @ryancaicse, please try to attach a patch with the changes you would like to make. As soon as I have time I will do a review.

Thanks

ryancaicse, 2021-11-21:
I created a PR. Thank you for your checking.

closed 2021-11-27 by ryancaicse