RfidResearchGroup/proxmark3 #1539: HF_Mattyrun keys issue

hexan86, 2021-12-01:
**Describe the bug**
HF_MATTYRUN working fine only with MFC 1K with key FFFFFFFFFFFF

**To Reproduce**
Steps to reproduce the behavior:
1. Add "A Key" to mattyrun source
2. Compile + Flash
3. Put MFC 1K on Proxmark RDV2
4. Standalone mode

**Expected behavior**
Find the key of the tag, getting it from the source (where is correct), instead it fails.
Standalone TAG 01 with A A4A4A4A4A4A4 and B FFFFFFFFFFFF:  https://pastebin.com/RGdrYuwN
Standalone TAG 02 with only FFFFFFFFFFFF keys (A & B): https://pastebin.com/ECUdTqSk

Tag 02 with PM3 client: 
hf mf autopwn keys: https://pastebin.com/mWNAZzSA
here the key is found with Nested attack, but I've inserted the key in the source code of mattyrun

if I access from pm3 client to Tag 01, using "hf mf rdsc -s 0 -k A4A4A4A4A4A4" it works and shows content
hf mf rdsc -s 0 -k FFFFFFFFFFFF = Shows nothing
hf mf rdsc -s '1-15' -k FFFFFFFFFFFF = Shows content
hf mf rdsc -s '1-15' -k A4A4A4A4A4A4 = Shows nothing

So, apparently, the command "hf mf rdbl --blk 0 -k A4A4A4A4A4A4"  is getting access to the MFC 1K, reading the content. At this point it's not clear to me if I'm missing something, ig I have to edit hf_mattyrun.c and change "uint8_t blockNo = 3;" to zero, or something else and then recompile it.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - hw version: https://pastebin.com/ALAYm44a
 - hw status: https://pastebin.com/4NjsbAd5
 - data tune: https://pastebin.com/bcjZteHS

iceman1001, 2021-12-01:
like the output says,  that standalone mode doesn't support the nested attack,  which is needed to recover all keys.
As you see with the autopwn command it does.

closed 2021-12-01 by iceman1001

hexan86, 2021-12-01:
> like the output says,  that standalone mode doesn't support the nested attack,  which is needed to recover all keys.
> As you see with the autopwn command it does.
> 
> 

But having added the Tag Key in the source code of mattyrun, shouldn't use that instead of attempting a Nested attack? This is what I'm asking.

iceman1001, 2021-12-03:
Ok,  I see.  You added your key to the source code, you compiled w/o errors,  you flashed w/o errors and it used the key for the right keytype and you have this error?    Do I understand the problem correct?

hexan86, 2021-12-03:
> Ok, I see. You added your key to the source code, you compiled w/o errors, you flashed w/o errors and it used the key for the right keytype and you have this error? Do I understand the problem correct?

Exactly. But now I have deleted the folder, downloaded the repo again, done all over again and now it works correctly.

iceman1001, 2021-12-04:
Good,  its resolved by itself.