RfidResearchGroup/proxmark3 #1540: /opt/homebrew/bin/pm3: line 249: 53337 Segmentation fault: 11  $CLIENT "$@"

AjayShanker-geek, 2021-12-06:
**Describe the bug**
run `hf mf autopwn` or  'hf mf hardnested --blk 0 -a -k a0a1a2a3a4a5 --tblk 7 --ta -w -s'

```
 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 10 threads and no SIMD core                 |                 |
       0 |       0 | Brute force benchmark: 984 million (2^29.9) keys/s      | 140737488355328 |    2d
       2 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    2d
       4 |     112 | Apply bit flip properties                               |     68702724096 |   70s
       5 |     224 | Apply bit flip properties                               |     11340033024 |   12s
       6 |     335 | Apply bit flip properties                               |      2272691712 |    2s
       7 |     444 | Apply bit flip properties                               |      2272691712 |    2s
       8 |     556 | Apply bit flip properties                               |      1248303104 |    1s
       9 |     667 | Apply bit flip properties                               |      1248303104 |    1s
      10 |     778 | Apply bit flip properties                               |      1248303104 |    1s
      10 |     888 | Apply bit flip properties                               |      1248303104 |    1s
      11 |     998 | Apply bit flip properties                               |      1248303104 |    1s
      12 |    1106 | Apply bit flip properties                               |      1248303104 |    1s
      13 |    1217 | Apply bit flip properties                               |      1248303104 |    1s
      14 |    1325 | Apply bit flip properties                               |      1248303104 |    1s
      15 |    1435 | Apply bit flip properties                               |      1248303104 |    1s
/opt/homebrew/bin/pm3: line 249: 53337 Segmentation fault: 11  $CLIENT "$@"
```

**Desktop (please complete the following information):**
 - OS: macOS 12.0.1 21A559 arm64

 - hw version

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.14434 2021-09-18 21:44:55
  compiled with Clang/LLVM Apple LLVM 13.0.0 (clang-1300.0.29.3) OS:OSX ARCH:aarch64

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.14434 2021-09-18 21:44:55
       os: RRG/Iceman/master/v4.14434 2021-09-18 21:44:55
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 53% used )

```

 - hw status

```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 44520
[#]   Available memory........ 42212
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 293888
[#]   Transfer Speed PM3 -> Client... 587776 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 12
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32270 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)

```
 - data tune

```
[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 42.75 V - 125.00 kHz
[+] LF antenna: 39.73 V - 134.83 kHz
[+] LF optimal: 51.83 V - 129.03 kHz
[+] Approx. Q factor (*): 10.0 by frequency bandwidth measurement
[+] Approx. Q factor (*): 15.1 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 29.28 V - 13.56 MHz
[+] Approx. Q factor (*): 8.5 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

[!] ⚠️  You appear to be on a MacOS device without XQuartz.
[!] ⚠️  You may need to install XQuartz (https://www.xquartz.org/) to make the plot work.

```

iceman1001, 2021-12-06:
Are you able to run it with `gdb`  and get a more detailed log output?

AjayShanker-geek, 2021-12-08:
I am actually using an M1 MacBook Pro device. Is this the cause of the error, since an arm64 native build of arm-none-eabi-gcc is still not available?

iceman1001, 2021-12-08:
dunno,  might be.  Without proper debugging output its hard to tell

AjayShanker-geek, 2021-12-11:
Hi, I am having issues running and installing `gdb` on M1 Mac, installing via `brew`. The workaround for now is using UTM on my M1 Mac.

closed 2021-12-11 by AjayShanker-geek

iceman1001, 2021-12-11:
Not quite solved,  good that you found a way around it.