RfidResearchGroup/proxmark3 #1541: [RELEASE v4.14550] Checklist

iceman1001, 2021-12-12:
# Checklist

- [ ] CHANGELOG.md: add title: `## [releasename][YYYY-MM-DD]`
- [ ] `make style`
- [ ] `make miscchecks`
- [ ] `make clean; make client CC=clang CXX=clang++ LD=clang++` on recent Debian or Ubuntu
- [ ] `mymanualchecks.sh`
- [ ] `mycppcheck.sh` no alarming warning?
- [ ] `mystandalone_makes.sh` check that the script contains all standalone modes then compile all standalone modes (linux only)
- [ ] `experimental_lib` compilation & tests
- [ ] `experimental_client_with_swig` compilation & tests
- [ ] GitHub Actions - green across the board ( MacOS, Ubuntu, Windows)
- [ ] [Appveyor](https://ci.appveyor.com/project/RfidResearchGroup/proxmark3/history) green (PS)

# OS compilation and tests

```bash
#!/usr/bin/env bash

make clean && make -j PLATFORM=PM3GENERIC PLATFORM_EXTRAS= && tools/pm3_tests.sh --long || exit 1
make clean && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS= && tools/pm3_tests.sh --long || exit 1
make clean && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && tools/pm3_tests.sh --long || exit 1
make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && sudo make install PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && ( cd /tmp; proxmark3 -c 'data load -f lf_EM4x05.pm3;lf search -1'|grep 'Valid FDX-B ID found' ) && sudo make uninstall || exit 1

( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3GENERIC PLATFORM_EXTRAS= && ../../tools/pm3_tests.sh --clientbin $(pwd)/proxmark3 client ) || exit 1
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4  PLATFORM_EXTRAS= && ../../tools/pm3_tests.sh --clientbin $(pwd)/proxmark3 client ) || exit 1
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && ../../tools/pm3_tests.sh --clientbin $(pwd)/proxmark3 client ) || exit 1

# Hitag2crack, optionally with --long and --gpu ...
make hitag2crack/clean && make hitag2crack && tools/pm3_tests.sh hitag2crack || exit 1
```

- [ ] RPI Zero
- [ ] Jetson Nano
- [ ] WSL
- [ ] PSv3.10
- [ ] Archlinux
- [ ] Kali
- [ ] Debian Stable
- [ ] Debian Testing
- [ ] Ubuntu21
- [ ] ParrotOS
- [ ] Fedora
- [ ] OpenSuse Leap
- [ ] OpenSuse Tumbleweed
- [ ] OSX
- [ ] Android
- [ ] Termux

# creating release

- [ ] `make release RELEASE_NAME="ice awesome"`
  - last line of output,  gives you next command to run.
  - Sample:  `git push && git push origin v4.14550`
- [ ] CHANGELOG.md: edit title to add version info: `## [releasename.4.14550][YYYY-MM-DD]`

## Step Github releases

- [ ] Go to Github releases,  create release based on the new created tag and publish

## Step Homebrew updates

- [ ] update homebrew repo, file `proxmark3.rb`
  - with a SHA256 sum of the file `v4.14550.tar.gz`
  - with updated list of standalone modes

## Step package maintains

- [ ] make a list of new standalone modes, so when we alert package maintainers they have a sporting chance of adding them

doegox, 2021-12-13:
currently broken in Proxspace

iceman1001, 2021-12-13:
fixed

doegox, 2021-12-15:
Ran compilations and tests from script above on all Docker images, except for Hitag2crack tests.
If you have a setup for GPU, please go ahead.

Gator96100, 2021-12-16:
I am aware that the precompiled builds don’t bundle the tools. This was done so save space on my Google Drive, now that the builds are hosted on aws I could add them.

The hitag2 tools do need some porting to work on mingw, however if there is a need to run them on Windows Cygwin might work. I didn’t test it, but Cygwin might be able to compile them on Windows without any changes.

s00se, 2021-12-26:
Termux builds fine

linuxgemini, 2021-12-30:
Builds fine on macOS 12.1 with MacPorts with following Portfile locked to the commit 85def31a8d678a4bf3ff45746d754a38dcc128e9

```tcl
# -*- coding: utf-8; mode: tcl; tab-width: 4; indent-tabs-mode: nil; c-basic-offset: 4 -*- vim: fenc=utf-8:ft=tcl:et:sw=4:ts=4:sts=4

PortSystem          1.0
PortGroup           github 1.0
PortGroup           qt5 1.0

name                proxmark3-iceman
version             4.14610

github.setup        RfidResearchGroup proxmark3 85def31a8d678a4bf3ff45746d754a38dcc128e9
github.tarball_from archive
checksums           sha256  b2f03298571b9be1a2d3898d0e9f115aa18eab5cae8a82ac2e405e364b721d74 \
                    rmd160  fc2488ffb2382ecf87f1cdeccb0e5db184249c0f \
                    size    24758655

revision            0
license             GPL-2
categories          science comms
platforms           darwin

description         RRG/Iceman version of proxmark3 RFID Instrument Client/FW

long_description    ${description}. By default, the repository is built for PM3RDV4. \
                    To build for older hardware releases, please install the package \
                    with the \"pm3generic\" variant.

maintainers         {@linuxgemini linuxgemini.space:proxmark3packaging} openmaintainer

depends_lib-append  port:readline \
                    port:jansson \
                    port:lua52 \
                    port:python39 \
                    port:bzip2

depends_build-append \
                    port:arm-none-eabi-gcc \
                    port:arm-none-eabi-binutils

use_configure       no

build.env-append    USE_BREW=0 \
                    USE_MACPORTS=1 \
                    PREFIX=${prefix} \
                    MACPORTS_PREFIX=${prefix}

build.args-append   CC=${configure.cc} \
                    CXX=${configure.cxx} \
                    CPP=${configure.cpp} \
                    LD=${configure.cxx}

variant pm3generic description {Build firmware for PM3GENERIC instead of PM3RDV4} {
    build.args-append   PLATFORM=PM3GENERIC
    destroot.env-append PLATFORM=PM3GENERIC
}

# buildsystem quirk: "make install" calls "make all" as well
destroot.args       {*}${build.args}
destroot.env        {*}${build.env}
```

iceman1001, 2021-12-30:
@linuxgemini   are you building both RDV4 / GENERIC?

linuxgemini, 2021-12-30:
> @linuxgemini are you building both RDV4 / GENERIC?

built both RDV4 and PM3GENERIC

tested client for PM3GENERIC

[image]

[image]

iceman1001, 2022-01-09:
Alrighty then,    
if y'all could do a pull latest and compile on all your different platforms,  and test running the stuff you usually do, 
It would be appreciated.

doegox, 2022-01-09:
# Checklist

- [ ] CHANGELOG.md: add title: `## [Frostbit] [YYYY-MM-DD]`
- [ ] `make style`
- [ ] `make miscchecks`
- [ ] `make clean; make client CC=clang CXX=clang++ LD=clang++` on recent Debian or Ubuntu
- [ ] `mymanualchecks.sh`
- [ ] `mycppcheck.sh` no alarming warning?
- [ ] `tools/build_all_firmwares.sh` check that the script contains all standalone modes then compile all standalone modes (linux only)
- [ ] `experimental_lib` compilation & tests
- [ ] `experimental_client_with_swig` compilation & tests
- [ ] GitHub Actions - green across the board ( MacOS, Ubuntu, Windows)

# OS compilation and tests

```bash
#!/usr/bin/env bash

make clean && make -j PLATFORM=PM3GENERIC PLATFORM_EXTRAS= && tools/pm3_tests.sh --long || exit 1
make clean && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS= && tools/pm3_tests.sh --long || exit 1
make clean && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && tools/pm3_tests.sh --long || exit 1
make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && sudo make install PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && ( cd /tmp; proxmark3 -c 'data load -f lf_EM4x05.pm3;lf search -1'|grep 'Valid FDX-B ID found' ) && sudo make uninstall || exit 1

( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3GENERIC PLATFORM_EXTRAS= && cp -a ../*scripts ../*libs . && ../../tools/pm3_tests.sh --clientbin $(pwd)/proxmark3 client ) || exit 1
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4  PLATFORM_EXTRAS= && cp -a ../*scripts ../*libs . && ../../tools/pm3_tests.sh --clientbin $(pwd)/proxmark3 client ) || exit 1
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && cp -a ../*scripts ../*libs . && ../../tools/pm3_tests.sh --clientbin $(pwd)/proxmark3 client ) || exit 1

# Hitag2crack, optionally with --long and --gpu ...
make hitag2crack/clean && make hitag2crack && tools/pm3_tests.sh hitag2crack || exit 1
```

- [ ] RPI Zero
- [ ] Jetson Nano
- [ ] WSL
- [ ] PSv3.10
- [ ] Archlinux
- [ ] Kali
- [ ] Debian Stable
- [ ] Debian Testing
- [ ] Ubuntu 21.04
- [ ] ParrotOS
- [ ] Fedora
- [ ] OpenSuse Leap
- [ ] OpenSuse Tumbleweed
- [ ] OSX
- [ ] Android
- [ ] Termux
- [ ] `make hitag2crack/clean && make hitag2crack && tools/pm3_tests.sh hitag2crack`

# creating release

- [ ] `make release RELEASE_NAME="Frostbit"`
  - last line of output,  gives you next command to run.
  - Sample:  `git push && git push origin v4.14810`
- [ ] CHANGELOG.md: edit title to add version info: `## [Frostbit.4.14810][YYYY-MM-DD]`

## Step Github releases

- [ ] Go to Github releases,  create release based on the new created tag and publish

## Step Homebrew updates

- [ ] update homebrew repo, file `proxmark3.rb`
  - with a SHA256 sum of the file `v4.14810.tar.gz`
  - with updated list of standalone modes

## Step package maintains

- [ ] make a list of new standalone modes, so when we alert package maintainers they have a sporting chance of adding them

linuxgemini, 2022-01-11:
Testing on macOS (x86) with MacPorts:

  - **Base Commit**: 105d5334dbe90d7f03db8e9487a1b7725a7e2246
  - **System**:
     ```
     macOS 12.1 21C52 x86_64
     Xcode 13.2.1 13C100
    ```
  - **MacPorts Version**: 2.7.1
  - **Portfile**:
    ```tcl
    # -*- coding: utf-8; mode: tcl; tab-width: 4; indent-tabs-mode: nil; c-basic-offset: 4 -*- vim: fenc=utf-8:ft=tcl:et:sw=4:ts=4:sts=4
    
    PortSystem          1.0
    PortGroup           github 1.0
    PortGroup           qt5 1.0
    
    name                proxmark3-iceman
    version             4.147820
    
    github.setup        RfidResearchGroup proxmark3 105d5334dbe90d7f03db8e9487a1b7725a7e2246
    github.tarball_from archive
    checksums           sha256  84ce9a65387fa0cd2c19e6f7c6e7d1054a13fb196de6d34bf7dea9d1c39e6fcd \
                        rmd160  d92707cb10812190f404ebbe551abe38414e9852 \
                        size    24762270
    
    revision            0
    license             GPL-2
    categories          science comms
    platforms           darwin
    
    description         RRG/Iceman version of proxmark3 RFID Instrument Client/FW
    
    long_description    ${description}. By default, the repository is built for PM3RDV4. \
                        To build for older hardware releases, please install the package \
                        with the \"pm3generic\" variant.

    maintainers         {@linuxgemini linuxgemini.space:proxmark3packaging} openmaintainer
    
    depends_lib-append  port:readline \
                        port:jansson \
                        port:lua52 \
                        port:python39 \
                        port:bzip2

    depends_build-append \
                        port:arm-none-eabi-gcc \
                        port:arm-none-eabi-binutils

    use_configure       no
    
    build.env-append    USE_BREW=0 \
                        USE_MACPORTS=1 \
                        PREFIX=${prefix} \
                        MACPORTS_PREFIX=${prefix}
    
    build.args-append   CC=${configure.cc} \
                        CXX=${configure.cxx} \
                        CPP=${configure.cpp} \
                        LD=${configure.cxx}

    variant pm3generic description {Build firmware for PM3GENERIC instead of PM3RDV4} {
        build.args-append   PLATFORM=PM3GENERIC
        destroot.env-append PLATFORM=PM3GENERIC
    }
    
    # buildsystem quirk: "make install" calls "make all" as well
    destroot.args       {*}${build.args}
    destroot.env        {*}${build.env}
    ```
  - `pm3` run: 
    [image]
  - `hf mf auto` and (after deleting files of `hf-mf*`) `hf mf staticnested` run:
    [image]

iceman1001, 2022-01-11:
Nice, but first,
The license has now been updated to `GPL3+`  and its unified across the repository,  except for scripts and dependencies. 

the all zeros on the static nested seems to have worked or where trying to say it failed?   The wrong response length message usually is a indication that you should have more space between tag and reader

linuxgemini, 2022-01-11:
> Nice, but first, The license has now been updated to `GPL3+` and its unified across the repository, except for scripts and dependencies.
>
> the all zeros on the static nested seems to have worked or [...]

Alright will do that

iceman1001, 2022-01-11:
Thank you everyone, 
This release is now live.

closed 2022-01-11 by iceman1001

wh201906, 2023-03-10:
@iceman1001 Hi. I think the title should be `v4.14831` rather than `v4.147831`. Is this a typo?

doegox, 2023-03-10:
indeed, thanks for spotting it @wh201906 !

iceman1001, 2023-03-11:
My bad,   Thanks for pointing it out!