RfidResearchGroup/proxmark3 #1555: [idea]  merge ALTRL branch into master.

iceman1001, 2022-01-11:
After looking at the new up-to-date readline/linenoise branch,   I think it should be merged into master.
It simplifies source code parts and gives an alternative to those systems who doesn't have readline installed.

doegox, 2022-01-11:
Just to record key points of our discussion:
* it appears linenoise-ng contains ConvertUTF.cpp which is under some redis-only license, which lead to various discussions, e.g.  https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=864729#26
* investigate https://github.com/rain-1/linenoise-mob which seems better maintained and which has removed the ConvertUTF.cpp

iceman1001, 2022-01-11:
messy messy,   
the question I have is how many of our Proxmark3 users who doesn't have Readline-lib and have linenoise-lib instead.
Or are we going to default to _linenoise-lib_  ?

doegox, 2022-01-11:
Well, "have linenoise-lib" is not much an issue as we would embed it in the project.
So far, I'm not satisfied with it enough to throw away readline. E.g. no filename completion, we would need to add it ourselves in the callback.
And linenoise-mob doesn't provide exactly the same user experience as linenoise-ng.
So I'll bring the API change, just to be a bit more independent from readline and get a more readable code, and for linenoise I don't know yet, maybe just a script to fetch it if someone wants to use it rather than readline.

iceman1001, 2022-01-12:
having readline takes precedence,   then falling back to linenoise  would still give a better experience than falling back to nothing.

The API change is nice,  make the source much cleaner and readable.

doegox, 2022-01-29:
Done.
I changed a bit the logic so now if someone wants linenoise, there is the get_linenoise.sh and a local copy of the zip.
So maintainers can just ignore or discard these 2 files if linenoise-ng license is an issue for them

closed 2022-01-29 by doegox

iceman1001, 2022-01-29:
Super-b!

I almost posted here this morning asking about it.  haha,   now we can remove some branches :)