RfidResearchGroup/proxmark3 #1581: Test failure on armhf

sbrun, 2022-02-03:
**Describe the bug**
The tests fail when we build proxmark3 version 4.14831 on armhf for Kali Linux.

**To Reproduce**
When we build the package, the tests are run after the build. The following commands are run:
```
./tools/build_all_firmwares.sh
echo 'PLATFORM=PM3RDV4' > Makefile.platform
echo 'PLATFORM_EXTRAS=BTADDON' >> Makefile.platform
make all
make check
```

**Screenshots**
```
[...]
Testing scripts:
script run cmdscript                    [ OK ] ✔️
script run luascript                    [ OK ] ✔️
check Python support                    [ OK ] ✔️
script run pyscript                     [ FAIL ] ❌
Execution trace:

------------------------------------------------------------

Tests [ FAIL ] ❌
```
I tried to get more details about the failure with gdb:
```
(gdb) run -c 'script run parity.py 10 1234'
Starting program: /build/proxmark3/client/proxmark3 -c 'script run parity.py 10 1234'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[=] Session log /tmp/.proxmark3/logs/log_20220202.txt
[+] loaded from JSON file /tmp/.proxmark3/preferences.json
[+] execute command from commandline: script run parity.py 10 1234

[=] Running in OFFLINE mode. Check "proxmark3 -h" if it's not what you want.

[offline|script] pm3 --> script run parity.py 10 1234
[+] executing python /build/proxmark3/client/pyscripts/parity.py
[+] args '10 1234'

Program received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/arm/armv6t2/strlen.S:126
126	../sysdeps/arm/armv6t2/strlen.S: No such file or directory.

(gdb) bt
#0  strlen () at ../sysdeps/arm/armv6t2/strlen.S:126
#1  0xf6b7bd06 in __vfprintf_internal (s=s@entry=0xfffece50, 
    format=format@entry=0x60ad2c "Could open file \033[33m%s\033[0m", ap=..., ap@entry=..., 
    mode_flags=mode_flags@entry=0) at vfprintf-internal.c:1647
#2  0xf6b873a8 in __vsnprintf_internal (string=string@entry=0xfffecfec "Could open file \033[33m", 
    maxlen=<optimized out>, maxlen@entry=2048, 
    format=format@entry=0x60ad2c "Could open file \033[33m%s\033[0m", args=..., 
    mode_flags=mode_flags@entry=0) at vsnprintf.c:114
#3  0xf6b873f8 in ___vsnprintf (string=string@entry=0xfffecfec "Could open file \033[33m", 
    maxlen=maxlen@entry=2048, format=format@entry=0x60ad2c "Could open file \033[33m%s\033[0m", 
    args=..., args@entry=...) at vsnprintf.c:124
#4  0x0050aeb8 in PrintAndLogEx (level=level@entry=ERR, 
    fmt=0x60ad2c "Could open file \033[33m%s\033[0m") at src/ui.c:262
#5  0x004cb352 in CmdScriptRun (Cmd=<optimized out>) at src/cmdscript.c:438
#6  0x004ca788 in CmdsParse (Commands=0x67dbf8 <CommandTable>, Cmd=<optimized out>)
    at src/cmdparser.c:312
#7  0x004ca788 in CmdsParse (Commands=0x67d794 <CommandTable>, 
    Cmd=Cmd@entry=0xc4b788 "script run parity.py 10 1234") at src/cmdparser.c:312
#8  0x004c975a in CommandReceived (Cmd=Cmd@entry=0xc4b788 "script run parity.py 10 1234")
    at src/cmdmain.c:355
#9  0x00506fbc in main_loop (script_cmds_file=script_cmds_file@entry=0x0, 
    script_cmd=0xfffefe0c "USER=kalibuild", 
    script_cmd@entry=0xfffefdef "script run parity.py 10 1234", 
    stayInCommandLoop=stayInCommandLoop@entry=false) at src/proxmark3.c:470
#10 0x0042e442 in main (argc=3, argv=0xfffefca4) at src/proxmark3.c:1139
```

**Desktop :**
 - OS: Kali Linux
 - hw version
 ```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  RRG/Iceman/master/v4.14831 2022-01-11 19:17:19
  compiled with............. GCC 11.2.0
  platform.................. Linux / arm
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present
```

 - hw status (I'm not sure this relevant for this case)
 ```
[offline] pm3 --> hw status
[!] ⚠️  This command is not available in this mode
```

 - data tune
```
[offline] pm3 --> data tune
[!] ⚠️  This command is not available in this mode
```

**Additional context**
It's probably an out of bounds access because the problem disappeared if we modified the code to add an other variable on the stack.
```
--- a/client/src/cmdscript.c
+++ b/client/src/cmdscript.c
@@ -399,6 +399,7 @@ static int CmdScriptRun(const char *Cmd) {
         PrintAndLogEx(SUCCESS, "executing python " _YELLOW_("%s"), script_path);
         PrintAndLogEx(SUCCESS, "args " _YELLOW_("'%s'"), arguments);

+       int length = strlen(script_path);
         wchar_t *program = Py_DecodeLocale(filename, NULL);
         if (program == NULL) {
             PrintAndLogEx(ERR, "could not decode " _YELLOW_("%s"), filename);
@@ -433,6 +434,7 @@ static int CmdScriptRun(const char *Cmd) {
         // setup search paths.
         set_python_paths();

+      printf("length = %d\n", length);
         FILE *f = fopen(script_path, "r");
         if (f == NULL) {
             PrintAndLogEx(ERR, "Could open file " _YELLOW_("%s"), script_path);
```

doegox, 2022-02-04:
Ouch, won´t be easy to debug...
I couldn't find a Docker with Kali for armhf, so I tried to reproduce with a Debian testing armhf (arm32v7) Docker image and Proxmark3 v4.14831:
```
docker run --rm --privileged multiarch/qemu-user-static:register --reset
cp -a /usr/bin/qemu-arm-static .
docker build -t "pm3-debian-bookworm-armhf:1.0" .
```
with a Dockerfile starting with
```
FROM arm32v7/debian:bookworm-slim
ADD qemu-arm-static /usr/bin
```
```
make client
client/proxmark3 -c 'script run parity.py 10 1234'
```
and.. I get a segfault \o/
That's cool, at least now we can reproduce the bug on our machine as well.

doegox, 2022-02-04:
Ok I found it.
Indeed out-of-bound access on array `py_args`
Fixed in https://github.com/RfidResearchGroup/proxmark3/commit/aadca614142fd3778d31034deb6feccb971f98f8
Feel free to add it as local Debian patch till next release.
Thanks!

iceman1001, 2022-02-04:
nice catch,   its getting harder to keep up the function testing on all different OS/host hardware setups we encounter.

sbrun, 2022-02-04:
Thank you very much for the really quick answer and for the patch! I will add it as local Debian patch.

closed 2022-02-04 by doegox

iceman1001, 2022-02-04:
Lets see if we can do a faster release this time,