RfidResearchGroup/proxmark3 #1582: hf 14a list - end of frame time is same as start for long frames

ah01, 2022-02-03:
**Describe the bug**

I was sniffing some MFD communication. When I print communication, file reading shows for ~100 B long file wrong end time. It is same as start of frame time (so duration is 0). It doesn't matter if `hf 14a list` or `hf mfd list` is used.

**To Reproduce**

```
hf 14a sniff -r
hf mfd list
```

Here last frame is from **7718640** to **7718640** so duration is **0**, which seems to be wrong.

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
…   
    6458044 |    6462812 | Rdr |e0  80  31  73                                                           |  ok | RATS
    6466432 |    6475712 | Tag |06  75  77  81  02  80  02  f0                                           |  ok |
    6489932 |    6503852 | Rdr |02  90  5a  00  00  03  4e  32  53  00  1d  5a                           |  ok | SELECT APPLICATION
    6524160 |    6530048 | Tag |02  91  00  29  10                                                       |     |
    6544588 |    6557420 | Rdr |02  90  71  00  00  02  02  00  00  bb  97                               |  ok | AUTH EV2 First
    6617472 |    6641728 | Tag |03  c1  73  7d  b8  79  df  e9  fa  e2  b6  94  e2  65  df  51  b3  91   |     |
            |            |     |af  45  77                                                               |  ok |
    6661132 |    6708460 | Rdr |02  90  af  00  00  20  c8  41  b8  45  a4  88  c9  21  90  3c  d1  1d   |     |
            |            |     |0a  c5  5b  fa  81  98  27  25  93  5c  cc  05  f0  11  9b  0e  ca  78   |     |
            |            |     |41  88  00  d8  2e                                                       |  ok | AUTH FRAME / NEXT FRAME
    6812368 |    6855056 | Tag |02  b1  30  90  ec  30  72  73  72  23  06  4d  57  19  fe  c2  4c  ff   |     |
            |            |     |02  d7  ae  78  ff  b3  4c  de  c6  94  3f  5f  f7  23  c9  91  00  2c   |     |
            |            |     |f9                                                                       |  ok |
    6887852 |    6907532 | Rdr |02  90  51  00  00  08  69  0e  4f  20  90  2e  76  c7  00  f6  78       |  ok |
    6988000 |    7021472 | Tag |03  ad  41  2b  af  e9  cc  43  a9  13  f8  28  7b  7b  71  c5  31  8f   |     |
            |            |     |55  8f  ad  8e  12  90  18  91  00  17  8b                               |  ok |
    7043804 |    7055484 | Rdr |02  90  77  00  00  01  01  00  55  f7                                   |  ok | AUTH EV2 Non First
    7090448 |    7114704 | Tag |02  24  28  30  46  b5  88  37  24  b8  b2  96  70  d2  24  cd  29  91   |     |
            |            |     |af  5a  6c                                                               |  ok |
    7134348 |    7181676 | Rdr |02  90  af  00  00  20  55  81  17  2d  e6  84  fd  a2  58  f7  e1  01   |     |
            |            |     |24  b4  21  ff  aa  1d  9e  1f  5b  67  da  7f  b2  18  ed  bb  f2  36   |     |
            |            |     |26  06  00  03  7b                                                       |  ok | AUTH FRAME / NEXT FRAME
    7276480 |    7300800 | Tag |03  bc  e0  9f  aa  96  2b  41  de  39  f8  79  31  85  14  6e  dd  91   |     |
            |            |     |00  bf  ad                                                               |  ok |
    7334044 |    7361788 | Rdr |02  90  ad  00  00  0f  03  00  00  00  02  00  00  83  60  bd  f8  c1   |     |
            |            |     |76  39  35  00  23  2b                                                   |  ok |
    7455952 |    7489424 | Tag |02  a5  18  c5  e1  33  c4  5f  fd  4b  aa  73  58  bd  b9  18  2b  06   |     |
            |            |     |0c  ea  4c  08  24  df  5b  91  00  d8  84                               |  ok |
    7511868 |    7539612 | Rdr |02  90  ad  00  00  0f  03  02  00  00  66  00  00  97  cf  81  05  a9   |     |
            |            |     |ef  6b  d0  00  e9  11                                                   |  ok |
    7718640 |    7718640 | Tag |03  00  d9  b7  76  ad  cd  91  af  4e  6a  74  4b  e2  29  7a  b9  db   |     |
            |            |     |4e  07  f1  02  0c  41  80  13  d1  cc  49  e8  49  71  0a  1e  d2  cd   |     |
            |            |     |1b  bd  7d  52  b4  fb  bc  94  b4  a5  c8  c4  71  7d  ee  f6  45  a6   |     |
            |            |     |4b  79  68  71  bf  a0  3e  d2  cb  19  01  7d  2d  bb  90  74  70  99   |     |
            |            |     |a8  eb  2a  3a  c7  3f  ff  57  af  10  20  89  9f  42  9b  65  ea  88   |     |
            |            |     |cb  d0  bf  8d  60  68  59  58  f3  c4  cb  fa  cd  46  38  82  11  07   |     |
            |            |     |58  da  09  67  0d  4c  45  e4  da  a2  38  01  52  91  00  4f  5a       |  ok |
…
```

**Desktop (please complete the following information):**
```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  RRG/Iceman/master/v4.14831-176-gdf083bffb 2022-02-03 18:36:45
  compiled with............. MinGW-w64 10.3.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.14831-176-gdf083bffb 2022-02-03 18:36:28
       os: RRG/Iceman/master/v4.14831-176-gdf083bffb 2022-02-03 18:36:36
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 59% used )
```

iceman1001, 2022-02-03:
Nice sniff,  got a trace file for it?

Is it the last entry of the trace that shows same end time as start time?

ah01, 2022-02-03:
> Is it the last entry of the trace that shows same end time as start time?

No. I have posted only short section. But only this large frame has this problem.

> Nice sniff, got a trace file for it?

I did not save it. So here is new one, but same card and reader and same problem.

[mfdtrace.zip](https://github.com/RfidResearchGroup/proxmark3/files/7998431/mfdtrace.zip)

iceman1001, 2022-02-04:
I would say this occures because we only use two bytes for storing duration of the packages.

Long frames has a long duration
https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/BigBuf.c#L259-L266

if you were to uncomment these lines,  and run with debug level ..  (but that might interup with pm3 comms w tag..)
you would get the message too long..

ah01, 2022-02-04:
Yes. It's there.

```
[usb] pm3 --> hf 14a sniff -c

[#] Starting to sniff. Press PM3 Button to stop.
[#] Error in LogTrace: duration too long for 16 bits encoding: 0x00023300   start: 0x0a89df67 end: 0x0a8c1267
[#] Error in LogTrace: duration too long for 16 bits encoding: 0x000232c0   start: 0x0abf2537 end: 0x0ac157f7
[#] Error in LogTrace: duration too long for 16 bits encoding: 0x00023300   start: 0x0af48627 end: 0x0af6b927
[#] trace len = 2982
```

Is it possible to increase duration size? I guess it is not that simple as to just change type in `tracelog hdr_t` struct, right?

iceman1001, 2022-02-05:
I believe the idea was to save space...     Limited amount of ram available for logging.
if we make it one byte larger,  we loose one byte per entry.  200-400 entries,   maybe not the biggest loss of ram.
I don't think the large frames where there when the implementation came to life.

Haven't looked at the impl,  but there will be some checks also needed to adapt for 3 bytes frames..

iceman1001, 2022-02-05:
if we change it however,  we loose backwards compability...

iceman1001, 2025-06-08:
I think if we could get it to be more specific,  like ISO14443A-4 logging,  a div4 method would work just fine.

Side note:
We could even get a bigger logging using the spiffs but then not all devices has it.