RfidResearchGroup/proxmark3 #1583: hf mfd info - will verify signature only once after client starts

ah01, 2022-02-03:
**Describe the bug**

I have MFD EV2 (in factory default). When I start client and run `hf mfd info` everything is OK. When I run it **second time**, section _Tag Signature_ will fail with message `Card doesn't support GetSignature cmd`. Some other following information are missing as well, but signature is first.

It seems to be some problem with client. `hw reset` or physical reconnect of device will not solve the issue. I need to terminate client and start over.

I have tested several builds for windows from last ~3 months and build master in WSL with same result. One build that I have and that works for me is from August 2021. But this one is before @merlokk have rewritten most of mfd, right(?)

**To Reproduce**

Steps to reproduce the behavior:
1. Start client
2. Run `hf mfd info`
3. Result is OK, except error `error DESFIRESendApdu ...`, but that is visible only in full client debug mode.
2. Run `hf mfd info` again
4. Signature, free memory, ... fails.

**First run:**

```
[usb] pm3 --> hf mfd info

[=] ---------------------------------- Tag Information ----------------------------------
[+]               UID: 04 0F 28 CA 65 70 80
[+]      Batch number: CF 2D D4 51 90
[+]   Production date: week 38 / 2020

[=] --- Hardware Information
[=]    raw: 04010112001805
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 12.0 ( DESFire EV2 )
[=]   Storage size: 0x18 ( 4096 bytes )
[=]       Protocol: 0x05 ( ISO 14443-2, 14443-3 )

[=] --- Software Information
[=]    raw: 04010102011805
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 2.1
[=]   Storage size: 0x18 ( 4096 bytes )
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )

[=] --------------------------------- Card capabilities ---------------------------------

[=] --- Tag Signature
[=]  IC signature public key name: DESFire EV2
[=] IC signature public key value: 04B304DC4C615F5326FE9383DDEC9AA8
[=]                              : 92DF3A57FA7FFB3276192BC0EAA252ED
[=]                              : 45A865E3B093A3D0DCE5BE29E92F1392
[=]                              : CE7DE321E3E5C52B3A
[=]     Elliptic curve parameters: NID_secp224r1
[=]              TAG IC Signature: 6F69E4CF824EBFEC2EEEC136A7226D42
[=]                              : E982F622B8C997CF2FAE14803F0AEEB4
[=]                              : 5B2AE2CDA1338318A262905E74FF6D88
[=]                              : B2ED44768C406763
[+]        Signature verification: successful
[#] error DESFIRESendApdu Current authentication status does not allow the requested command
[#] error DESFIRESendApdu Current authentication status does not allow the requested command
[#] error DESFIRESendApdu Current authentication status does not allow the requested command
[+] ------------------------------------ PICC level -------------------------------------
[+] Applications count: 0 free memory 5120 bytes
[+] PICC level auth commands: auth: YES auth iso: YES auth aes: NO auth ev2: NO auth iso native: YES auth lrp: NO
[+] PICC level rights:
[+] [1...] CMK Configuration changeable   : YES
[+] [.1..] CMK required for create/delete : NO
[+] [..1.] Directory list access with CMK : NO
[+] [...1] CMK is changeable              : YES
[+]
[+] Key: 2TDEA
[+] key count: 1
[+] PICC key 0 version: 0 (0x00)

[=] --- Free memory
[+]    Available free memory on card         : 5120 bytes

[=] Standalone DESFire
```

**Second run:**

```
[usb] pm3 --> hf mfd info

[=] ---------------------------------- Tag Information ----------------------------------
[+]               UID: 04 0F 28 CA 65 70 80
[+]      Batch number: CF 2D D4 51 90
[+]   Production date: week 38 / 2020

[=] --- Hardware Information
[=]    raw: 04010112001805
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 12.0 ( DESFire EV2 )
[=]   Storage size: 0x18 ( 4096 bytes )
[=]       Protocol: 0x05 ( ISO 14443-2, 14443-3 )

[=] --- Software Information
[=]    raw: 04010102011805
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 2.1
[=]   Storage size: 0x18 ( 4096 bytes )
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )

[=] --------------------------------- Card capabilities ---------------------------------

[=] --- Tag Signature
[#] error DESFIRESendRaw
[!] --- Card doesn't support GetSignature cmd
[#] error DESFIRESendRaw
[#] error DESFIRESendRaw
[#] error DESFIRESendRaw
[#] error DESFIRESendApdu Current authentication status does not allow the requested command
[#] error DESFIRESendApdu Current authentication status does not allow the requested command
[#] error DESFIRESendApdu Current authentication status does not allow the requested command
[+] ------------------------------------ PICC level -------------------------------------
[+] Applications count: 0 free memory n/a
[+] PICC level auth commands: auth: YES auth iso: YES auth aes: NO auth ev2: NO auth iso native: YES auth lrp: NO

[=] --- Free memory
[+]    Card doesn't support 'free mem' cmd

[=] Standalone DESFire
```

**Desktop (please complete the following information):**

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  RRG/Iceman/master/v4.14831-176-gdf083bffb 2022-02-03 18:36:45
  compiled with............. MinGW-w64 10.3.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.14831-176-gdf083bffb 2022-02-03 18:36:28
       os: RRG/Iceman/master/v4.14831-176-gdf083bffb 2022-02-03 18:36:36
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 59% used )

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 42488
[#]   Available memory........ 42488
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 163
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD5697C3097747623
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 501ms
[#]   Bytes transferred.............. 237568
[#]   Transfer Speed PM3 -> Client... 474187 bytes/s
[#] Various
[#]   Max stack usage......... 4120 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 39
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30754 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare.................. 1094 keys
[#]   T55x7................... 116 keys
[#]   iClass.................. 9 keys
[#]
```

merlokk, 2022-02-03:
strange. all is ok for me(

try to change please this line
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/src/cmdhfmfdes.c#L592
from
SetAPDULogging(false);
to
SetAPDULogging(true);

and execute again

ah01, 2022-02-03:
> SetAPDULogging(true);

Funny, I started to dig deeper and this was first thing I did 😀 But it was not enough. I also had to disable silentMode on ExchangeRAW14a (last parameter in desfirecore.c line 508) and uncomment macro `COMMS_DEBUG` and `COMMS_DEBUG_RAW`.

Here is log of first and second run
- [mfd-info_first.txt](https://github.com/RfidResearchGroup/proxmark3/files/7998906/mfd-info_first.txt)
- [mfd-info_second.txt](https://github.com/RfidResearchGroup/proxmark3/files/7998907/mfd-info_second.txt)

After I compare first and second run I found this:

[image]

The value `0B` on line 67 of diff come from https://github.com/RfidResearchGroup/proxmark3/blob/master/client/src/cmdhf14a.c#L814 and variable `responseNum`. 

This local variable is _static_, so at least it explains why it works only first time and then I need to restart client 😉

This is where I end now. I have no idea what is purpose of `responseNum`. Will this help you, or should I dig deeper.

iceman1001, 2022-02-04:
somewhere there is a call to exchangeraw14a,  where the activatefield changed during these months.   Where the static var should be set to zero (resetted)

merlokk, 2022-02-04:
I just fixed it several months ago( maybe not in this function...
Its iso14443 counter and it should be cleared after the iso anticollision-select sequence

merlokk, 2022-02-04:
I try to fix...

iceman1001, 2022-02-04:
I push a fix,   @ah01   if you pull latest and test?

closed 2022-02-04 by iceman1001

ah01, 2022-02-04:
I works fine now. I can run `hf mfd info` several times without any issue. 
Great work, thx 👍

commit 8076e8ec (2022-02-04): fix #1583