RfidResearchGroup/proxmark3 #1597: Cannot upgrade smartcard firmware (Unknown command 0x0142)

cactuschibre, 2022-02-11:
**Describe the bug**

I would like to use the Smartcard reader but cannot upgrade the firmware :

```bash
[usb] pm3 --> smart upgrade -f sim011.bin
[=] -------------------------------------------------------------------
[!] ⚠️  WARNING - sim module firmware upgrade
[!] ⚠️  A dangerous command, do wrong and you could brick the sim module
[=] -------------------------------------------------------------------

[=] firmware file       sim011.bin
[=] Checking integrity  sim011.sha512.txt
[+] loaded 733 bytes from binary file sim011.bin
[+] loaded 141 bytes from binary file sim011.sha512.txt
[=] Don't turn off your PM3!
[+] Sim module firmware uploading to PM3...
[#] unknown command:: 0x0142
[!] ⚠️  timeout while waiting for reply.
```

**To Reproduce**

Steps to reproduce the behavior:
1. Have the Proxmark3 up-to-date (Following https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md for Kali).
2. Get into Proxmark3 menu (./pm3) (With or without the Smartcard reader plugged in ; without antennas).
3. Do not forget to check the SHA512 sum (Read https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/2_Configuration-and-Verification.md#verify-sim-module-firmware-version)
3. Do `smart upgrade -f sim011.bin`
4. See error

**Expected behavior**

```bash
[+] Sim module firmware uploading to PM3...
 🕑 733 bytes sent
[+] Sim module firmware updating...
[#] FW 0000
[#] FW 0080
[#] FW 0100
[#] FW 0180
[#] FW 0200
[#] FW 0280
[+] Sim module firmware upgrade successful    
```

**Desktop (please complete the following information):**

 - OS: Kali Linux
 
 - hw version

```bash
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  RRG/Iceman/master/v4.14831-173-g468d2ac2b 2022-02-03 10:21:20
  compiled with............. GCC 10.2.1 20210110
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.14831-162-gc2c57d956 2022-02-02 16:29:28
       os: RRG/Iceman/master/v4.14831-162-gc2c57d956 2022-02-02 16:29:41
  compiled with GCC 10.3.1 20210621 (release)

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 47% used )
```

 - hw status

```bash
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 45156
[#]   Available memory........ 45156
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 507ms
[#]   Bytes transferred.............. 28672
[#]   Transfer Speed PM3 -> Client... 56552 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30167 Hz
[#] Installed StandAlone Mode
[#]   No standalone mode present
[#] 

```

 - data tune

```bash
[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[!] ⚠️  LF antenna is UNUSABLE
[=] ---------- HF Antenna ----------
[!] ⚠️  HF antenna is UNUSABLE

(*) Q factor must be measured without tag on the antenna

[-] ⛔ Not showing LF tuning graph since all values is zero.
```

iceman1001, 2022-02-11:
You have flashed PM3 Generic firmware..    Only a RDV4 has a sim card module,

cactuschibre, 2022-02-14:
Ok thanks @iceman1001. I have just seen the `Makefile.platform` with the PM3 Generic firmware configuration. It is a collaborative project for our team, it was a forgotten test I think. I will try without this and I'll let you know!

closed 2022-02-15 by cactuschibre

iceman1001, 2022-02-15:
no problem,