RfidResearchGroup/proxmark3 #1600: Compile error on opensuse 15.2

nitraiolo, 2022-02-14:
Since commit #ce74a25d064639e6961aeb2ef3be221eba366ece ("use macros instead" from @iceman1001) the code doesn't compile out of the box on opensuse 15.2 anymore with following error:
`
Version info:      RRG/Iceman/master/v4.14831-277-g6da5c9d0f
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  LF HITAG EM4x50 EM4x70 ZX8211 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN

[*] MAKE client/clean

Version info:      RRG/Iceman/master/v4.14831-277-g6da5c9d0f
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 4.8.7 in /usr/lib64)
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.6 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  gcc (SUSE Linux) 7.5.0

PLATFORM Linux

... cut ...

brute_key.c: In function ‘main’:
brute_key.c:154:33: error: expected ‘)’ before ‘PRIu64’
         printf("\btimestamp: %" PRIu64 "\nkey: ", timestamp);
                                 ^~~~~~
brute_key.c:154:30: error: spurious trailing ‘%’ in format [-Werror=format=]
         printf("\btimestamp: %" PRIu64 "\nkey: ", timestamp);
                              ^
cc1: all warnings being treated as errors
make[1]: *** [../../Makefile.host:95: obj/brute_key.o] Error 1
make: *** [Makefile:146: mfd_aes_brute/all] Error 

`

To compile it I had to remove "PRIu64" and use the "%lu", as in the previous version.

Thanks

doegox, 2022-02-14:
thank you, please test latest commit

nitraiolo, 2022-02-14:
Now it compiles well.

[*] MAKE mfd_aes_brute/all
[-] CC brute_key.c
[-] CC util_posix.c
[=] LD brute_key
[-] CC mfd_aes_brute.c
[=] LD mfd_aes_brute

Thanks @doegox !

closed 2022-02-14 by iceman1001