RfidResearchGroup/proxmark3 #1604: command fail on changing iso15 magic card

tandisbh, 2022-02-16:
Hello, 
I am using proxmark3 for changing uid of nxp sli magic cards.
I have changed a nxp sli card once, but I am not able to change uid changeable sticker tag bought form this link:
(https://m.alibaba.com/product/1600167810376/1600167810376.html)

I get this error while changing uid using { script run hf_15_magic -u E004013344556677 } command 

new UID | E004013344556677
Using backdoor Magic tag function
[!] command failed
[!] command failed
[!] command failed
[!] command failed

What is the problem?

iceman1001, 2022-02-16:
dunno,  you would need to debug the script and verify that the card works with that script.

tandisbh, 2022-02-17:
Thank you for your reply,

I either used this command : { hf 15 csetuid -u E0040150A837FF70 },
but it fails:

[+] reverse input UID 70 FF 37 A8 50 01 04 E0
[=] getting current card details...

[+]  UID: E0 04 01 50 A8 37 FF 67
[+] TYPE: NXP(Philips); IC SL2 ICS20/ICS21(SLI) ICS2002/ICS2102(SLIX) ICS2602(SLIX2)
[=] updating tag uid...
[=] getting updated card details...

[+]  UID: E0 04 01 50 A8 37 FF 67
[+] TYPE: NXP(Philips); IC SL2 ICS20/ICS21(SLI) ICS2002/ICS2102(SLIX) ICS2602(SLIX2)
[-] setting new UID (failed)

Would you please tell me how should I debug it?

iceman1001, 2022-02-19:
I recommend you contact the seller and ask how you modify the uid on the sticker.

iceman1001, 2022-12-05:
Ping!
Did you get any feedback from seller?

tandisbh, 2022-12-07:
Hi!
Yes, unfortunately the seller told me the magic tag could not work with
PROXMARK3.
Thanks for your email.

iceman1001, 2022-12-08:
That sounds like BS answer from the seller in order to sell you "their" software...    I heard it before.

tandisbh, 2022-12-10:
Exactly!
At last I sent them my UIDs, and they produced tags with those UIDs for me!

closed 2022-12-10 by iceman1001

mayjack0312, 2023-12-25:
> Exactly! At last I sent them my UIDs, and they produced tags with those UIDs for me!
> […](#)

May I ask where you obtained these cards? What are their prices?

c5e3, 2024-04-17:
looks like you bought this card with 64 blocks: https://aliexpress.com/item/1005005298673526.html
i am currently wondering, how to find out the 'set uid'-command without buying the expensive device from the seller.
did you have any luck in the meantime?

c5e3, 2024-04-22:
can confirm, it works!

iceman1001, 2024-04-22:
@s00se   I pushed a addition.
If y'all can test it out that would be nice.

`hf 15 csetuid -h`   for instructions

c5e3, 2024-04-23:
can confirm; the `--v2` command works!