RfidResearchGroup/proxmark3 #1608: hf mf hardnested doesn't work in the new version?

SaneBow, 2022-02-24:
I installed the latest stable of this fork by  `brew install --with-generic proxmark3` and then found out that `hardnested` always fails with `⛔ No match for the First_Byte_Sum (119), is the card a genuine MFC Ev1?`, and there a lot of `AcquireEncryptedNonces: Auth1 error` during the process.

However, the version installed with `brew install proxmark/proxmark3/proxmark3` works fine without any Auth1 error. 

I am using MacOS 12.2.1 on an M1 MacBook.

Failed version information:
```
 [ CLIENT ]
  RRG/Iceman/master/v4.14831 2022-01-11 19:17:19
  compiled with............. Clang/LLVM Apple LLVM 13.0.0 (clang-1300.0.29.30)
  platform.................. OSX / aarch64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.14831 2022-01-11 19:17:19
       os: RRG/Iceman/master/v4.14831 2022-01-11 19:17:19
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 53% used )
```

Success version:
```
 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-3347-gb89436564 2021-03-18 17:34:19
  compiled with Clang/LLVM Apple LLVM 12.0.0 (clang-1200.0.32.29) OS:OSX ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-3347-gb89436564 2021-03-18 17:37:14
       os: RRG/Iceman/master/v4.9237-3347-gb89436564 2021-03-18 17:37:29
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 53% used )
```

iceman1001, 2022-02-24:
Feel free to find the bug and make a PR

Doridian, 2022-03-27:
Had the same issue with the current Homebrew version, pulled latest git master and compiled myself and now its fine.
So I assume whatever bug happened here probably got fixed already.

closed 2022-03-27 by iceman1001