RfidResearchGroup/proxmark3 #1609: [!!] No port found

elijahlimas123, 2022-02-24:
***Things to try before submitting bug report***
read the [troubleshooting guide](/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to RRG/Iceman repo.

-

**Describe the bug**
running Proxmark3 Easy on a MacBook air. the MacBook will only detect the Proxmark3 mark when the Proxmark is in boot loader mode, and even then it is hit or miss. the command "pm3 --list" returns "[!!] No port found"

**To Reproduce**
Steps to reproduce the behavior:
hold proxmark3 button as you plug it in to MacBook Air to enter boot loader mode.
make sure it is detected with system_profiler SPUSBDataType 
once detected use command "pm3-flash-all"
it returns what is shown in the screenshot.

**Expected behavior**
I expected to run the "pm3-flash-all" command and have the Proxmark3 Easy be detected at /dev/tty.usbmodemiceman1

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: macOS Big Sur version 11.6.4
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
Add any other context about the problem here.

iceman1001, 2022-02-24:
... which device do you have vs which compiled version of the firmware you flashed?

elijahlimas123, 2022-02-24:
I have the Proxmark3 Easy from dangerous things I ordered it a week ago (beyond that I don't know how to tell which device I have), I flashed Release v4.14831 - Frostbit
thanks for the quick response

iceman1001, 2022-02-24:
My guess you didn't follow instructions and didn't compile a firmware/client for your device.
That would be a user-issue and not a issue for this repo.  Ask questions in the Pm3 forum or the discord.

closed 2022-02-24 by iceman1001