RfidResearchGroup/proxmark3 #1611: #680121b - Build failed

JohnDoeIsKIng, 2022-02-25:
Dear community,

the current repo fails with:
`[-] CC src/cmdlfio.c
src/cmdlfio.c: In function ‘demodIOProx’:
src/cmdlfio.c:145:44: error: ‘__builtin___snprintf_chk’ output may be truncated before the last format character [-Werror=format-truncation=]
         snprintf(crc_str, sizeof(crc_str), "( " _RED_("fail") " ) 0x%02X != 0x%02X", crc, calccrc);
                                            ^~~~
src/cmdlfio.c:145:83: note: format string is defined here
         snprintf(crc_str, sizeof(crc_str), "( " _RED_("fail") " ) 0x%02X != 0x%02X", crc, calccrc);
                                                                                   ^
In file included from /usr/include/stdio.h:862:0,
                 from src/cmdlfio.c:21:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 31 and 37 bytes into a destination of size 36
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:901: recipe for target 'obj/cmdlfio.o' failed
make[1]: *** [obj/cmdlfio.o] Error 1
Makefile:157: recipe for target 'client/all' failed
make: *** [client/all] Error 2
`

Any hints to change this ?
Best regards

JD.

iceman1001, 2022-02-25:
Thanks, try pulling latest and see if its been fixed with latest commit

JohnDoeIsKIng, 2022-02-27:
@Iceman: Perfetto, it builds.
Thanx.

closed 2022-02-27 by JohnDoeIsKIng