RfidResearchGroup/proxmark3 #1612: PM3 RDV4 (and Easy) USB device not recognized on an up-to-date recent Arch Linux

maxieds, 2022-02-26:
I bought both a PM3 Easy and then a Proxgrind (Iceman collection) RDV4 device. I have been having a hard time getting this device recognized on my recent, updated Arch Linux box. I had read on the WIKI pages that the developers could not offer support for non-endorsed devices, so I bought the second pricier RDV4 device. It may in retrospect have not worked for the same set of USB problems. 

This issue is about installation on a factory-fresh RDV4 device. It is two-fold insomuch as I am going to report my problems getting things working on two different OS's. I hope this bug report makes it easier for others to use and helps the developers pin down weird corner cases with the install process. 

N.b., my new device did not come with any instructions even telling me that it needed to be flashed first (this requires holding the button down on power-up, which is not the first thing I would have tried without warning). The WIKI pages on this site are also somewhat scattered and hard to navigate for someone new to using this device. As a hacker myself, I  completely **GET** the front page warning:

>Third rule of proxmarking learning is hard and takes time

I know systems programming pretty well myself. But even so, this is kind of an issue (IMO) that the developers and manufacturers should start working on. For a $350 device purchase, the documentation needs to be better. With that stated, I will quit my b-tching and get to the point. 

### MacOS installation side tracking (on Big Sur)

I was able to install the Homebrew tap version of the most recent (HEAD) ``proxmark3`` software on my Mac. Then forced the unflashed device to get its firmare by running: 
```bash
pm3-flash-all --force
```
Without the ``--force`` option, the command failed (again on MacOS, Big Sur) reporting:
```bash
 pm3-flash-all 
[=] Session log /Users/mschmidt34/.proxmark3/logs/log_20220226.txt
[+] loaded from JSON file /Users/mschmidt34/.proxmark3/preferences.json
[+] About to use the following files:
[+]    /usr/local/Cellar/proxmark3/HEAD-0d5b786/bin/../share/proxmark3/firmware/bootrom.elf
[+]    /usr/local/Cellar/proxmark3/HEAD-0d5b786/bin/../share/proxmark3/firmware/fullimage.elf
[+] Loading ELF file /usr/local/Cellar/proxmark3/HEAD-0d5b786/bin/../share/proxmark3/firmware/bootrom.elf
[+] ELF file version RRG/Iceman/master/v4.14831-362-g0d5b7864a 2022-02-26 02:30:40 
[!] ⚠️  Unable to check version_information
[=] Make sure to flash a correct and up-to-date version
[=] You can force flashing this firmware by using the option '--force'
[!!] 🚨 Aborted on error

[=] Have a nice day!
```

### Completely non-functional factory-new device on Arch Linux

I was able to build the software directly by cloning this repo and running: ``make clean && make all``. Then I tried to run ``make accessrights``. The second command failed because I have no ``bluetooth`` group on the system. Though manually running (after the automated script failed with an error) the command to add my local user to ``uucp``. I personally loathe Bluetooth and intend on using the PM3 only over USB. This is ERROR-1. 

The second issue on Arch is getting the newly arrived, factory-fresh RDV4 device to even get recognized over USB. The ``lsusb`` command reports nothing when you plug it in (with or without holding down the button to make it enter flash mode). Running ``sudo dmesg | grep -i usb`` similarly reports nothing useful, i.e., the device doesn't get recognized on Arch. 

After flashing the device successfully on the Mac machine, I plugged it back in over USB to my laptop running Arch. Now we are finally in business! The ``dmesg`` output reports that the RDV device is there with serial number (``iceman``, ha) at ``/dev/ttyACM0``. 

Can someone please work on fixing this for us PM3 newbies? If I hadn't had the spare Mac box, I wouldn't have been able to flash this new device. I refuse to use Windows (since the days of NT). Y'all are hackers too, Arch is the new Gentoo, and the support is not there. Happy penguin!

maxieds, 2022-02-27:
The idea was to report a few glitches in the install process I had trying to flash a factory-new (fresh out of the box from the manufacturer that stocked my PM3) device. I tried to flash the blank device first on Arch Linux. Then after repeated failures, I tried it on MacOS with the Homebrew tap, which did finally work after supplying the ``--force`` option. There is no mention of how to fix the problem I am having in the local WIKI instructions, nor anywhere I can find by searching other sites.

On Arch Linux, the blank device (before the flash on Mac) does not get recognized as a valid USB device. The kernel rejects it plugging the new PM3 in both with and without holding down the button to enter bootloader mode. After flashing with a Mac, the PM3 DOES get recognized by the Arch box at the expected ``/dev/ttyACM0`` handler. Trying to re-flash the device on Arch with ``pm3-flash-all`` after it is running a valid firmware fails. 

I conclude that there is a valid issue running the code in this repo on Arch Linux that needs to get fixed. Do you need more information like which version of the kernel I have running? This is still a problem because the ``pm3`` software I have on the linux box warns me that the MacOS version of the ARM firmware does not match the source at the time the client was compiled. But I cannot flash the PM3 with the version I compiled myself on linux because of the USB recognition problem I am experiencing. I have not spent much time with the source to this repo. It seems to me that this is another legitimate bug that needs to get addressed either by figuring out what Arch's kernel expects the device to spew out before the system will call it a TTY, or possibly (depending on how you coded this all up) within the flashing program.

There is also the problem where ``make accessrights`` fails if the system does not have a group named ``bluetooth``, which my system does not. That should be easy to fix from whatever script you call in the ``Makefile``. More minor of a problem, but still a bug nonetheless. 

N.b., I purchased the PM3 (finally) to finalize the DESFire project support for the Chameleon Mini firmware over on the ``emsec`` page.

iceman1001, 2022-02-27:
If you think there are glitches in the installation instructions for Arch-linux found on this repo,    feel free to improve them.

Arch-linux - setup of env related docs.
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/ModemManager-Must-Be-Discarded.md

I added an passage about `--force`   https://github.com/RfidResearchGroup/proxmark3/commit/df88335cf1f7302f3be72cd31589a9fc70427f37  in this guide  https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md

and here
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md#pm3-flash-stops-and-warns-about-up-to-date-firmware-images

> `I conclude that there is a valid issue running the code in this repo on Arch Linux that needs to get fixed. Do you need more information like which version of the kernel I have running? This is [...]

The bug report template give a base for which information is needed to start looking into it.
But otherwise,   you will need to provide,   what you steps you did,  which instructions guide you follower,  which step it failed on,  the output related to the command steps you did. 

General information about your dev-env,   if did you compile yourself (which guide did you follow)  or did you download a package?   If compiled yourself, 

* compiler version
* hostOS
* specific pm3 device model 

if downloaded a package,  which, how?
there are specific repositories which has instructions for how to use them. like homebrew  https://github.com/RfidResearchGroup/homebrew-proxmark3

I recommend you make individual GH issues about specific bugs,  like the `make accessrights` failing,  in order to make it easy to follow fixes and discussions.

iceman1001, 2022-02-27:
There is a bug in mkversion.sh on OSX,  leading to bug in version_pm3.c ,  leading to the firmware warning message.

iceman1001, 2022-02-27:
the Homebrew bug should be resolved now.

doegox, 2022-02-27:
Hi, I agree with Iceman, you mention valuable issues but it's very hard for us to go through all your description and cherry-pick what should be fixed.
So far:
* the firmware mismatch issue on OSX you got: it was indeed 2 bugs specific to OSX. Should be fixed now (including a new dependency to coreutils)
* arch make accessrights: indeed on some platforms ppl may not have installed the bluetooth stack and don't need it, so now it's fixed and doesn't issue an error if bluetooth group doesn't exist (both Arch and other linux distros)
* doc: a number of fixes have been done, probably still some to do. Note that the main doc entry point in the main README, not the Wiki

Still remains the problem you mentioned in the title of this issue. Honestly that's the very first time I hear it and other ppl have been using Arch with pm3 before you. So we'll need more info or try to discuss with Arch users on the Discord to understand why you have that behavior on your setup.

maxieds, 2022-03-01:
@iceman1001 @doegox 
I will create a separate issue for the ``make accessrights`` problem on Arch. 

Pardon my bad attitude above. I have had a **really** bad week. It has nothing to do with your software. 

The support and community on discord were very helpful. We seem to have concluded that the problem I was having about Arch recognition with the USB device is due to a faulty micro-USB cable. I ordered new cables and will follow up if problems persist on discord. You can close out this issue. 

Thanks again. **--MDS**

closed 2022-03-02 by doegox