RfidResearchGroup/proxmark3 #1615: Problem running `make accessrights` on Arch Linux

maxieds, 2022-03-01:
**Describe the bug**
There is an error running `make accessrights` on recent Arch Linux. The version of Arch Linux I am running is ``5.16.111-arch1-1`` on ``x_86 GNU/Linux``. 

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the ``proxmark3`` repository
2. Build the source by running ``make clean && make all``
3. Attempt to give the local user access rights by running ``make accessrights``
4. The second command in (3) fails because there is no ``bluetooth`` group on the system. This failing command is ``sudo usermod -aG bluetooth maxieds``: ``usermod: group 'bluetooth' does not exist``
5. Running the first command ``sudo usermod -aG uucp maxieds`` succeeds by itself when run manually by the user

**Additional context**
My suggestion is to modify the script called by invoking ``make accessrights`` so that the operation succeeds to add the local user to the ``uucp`` group, even when there is no ``bluetooth`` group on the default Arch system.

maxieds, 2022-03-02:
@doegox
I ran ``git pull && make accessrights`` on the Arch system. The second command still produces an error:
```bash
make: *** [Makefile:268: accessrights] Error 2
```
A resolution to this problem should be that there is no error when running the command when the group ``bluetooth`` does not exist on the system?

doegox, 2022-03-02:
yes that was the intent.
the error you get is actually because Arch doesn't define USER.
This should be fixed now.

doegox, 2022-03-03:
@maxieds is the problem solved?

maxieds, 2022-03-27:
@doegox 
This has been one of the last things on my mind. I will try to remember to run ``git pull`` and check it on Arch later.

closed 2022-03-27 by iceman1001