RfidResearchGroup/proxmark3 #1616: Errors flashing the PM3 Easy devices with a generic build (MacOS)

maxieds, 2022-03-02:
**Describe the bug**
There are errors flashing the PM3 Easy line of devices using ``pm3-flash-all`` on MacOSX. The generic build of the firmware is installed by running
```bash
brew reinstall --with-generic proxmark3
```
where the Homebrew package is, as usual, the tap provided by this repo (Icemann/RFIDRes). Then when I plug in the Easy device over USB, and run the flash command, the following is the full output I get:
```bash
$ pm3-flash-all
[=] Session log /Users/localusernameredacted/.proxmark3/logs/log_20220302.txt
[+] loaded from JSON file /Users/localusernameredacted/.proxmark3/preferences.json
[+] About to use the following files:
[+]    /usr/local/Cellar/proxmark3/HEAD-ab52131/bin/../share/proxmark3/firmware/bootrom.elf
[+]    /usr/local/Cellar/proxmark3/HEAD-ab52131/bin/../share/proxmark3/firmware/fullimage.elf
[+] Loading ELF file /usr/local/Cellar/proxmark3/HEAD-ab52131/bin/../share/proxmark3/firmware/bootrom.elf
[+] ELF file version Iceman/master/v4.14831-404-gab5213126 2022-03-02 05:49:24 ea44b0c23

[+] Loading ELF file /usr/local/Cellar/proxmark3/HEAD-ab52131/bin/../share/proxmark3/firmware/fullimage.elf
[+] ELF file version Iceman/master/v4.14831-404-gab5213126 2022-03-02 05:49:38 ea44b0c23

[+] Waiting for Proxmark3 to appear on /dev/tty.usbmodem14601
 🕑  59 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Waiting for Proxmark3 to appear on /dev/tty.usbmodem14601
 🕓  59 found
[!!] 🚨 ====================== OBS ! ===========================================
[!!] 🚨 Note: Your bootloader does not understand the new CMD_BL_VERSION command
[!!] 🚨 It is recommended that you first update your bootloader alone,
[!!] 🚨 reboot the Proxmark3 then only update the main firmware

[!!] 🚨 ------------- Follow these steps -------------------

[!!] 🚨  1)   ./pm3-flash-bootrom
[!!] 🚨  2)   ./pm3-flash-all
[!!] 🚨  3)   ./pm3

[=] ---------------------------------------------------

[=] Available memory on this board: UNKNOWN

[!!] 🚨 ====================== OBS ! ======================================
[!!] 🚨 Note: Your bootloader does not understand the new CHIP_INFO command
[=] Permitted flash range: 0x00100000-0x00140000
[!!] 🚨 ====================== OBS ! ========================================
[!!] 🚨 Note: Your bootloader does not understand the new START_FLASH command
[+] Loading usable ELF segments:
[+]    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1: V 0x00200000 P 0x00100200 (0x00000d1c->0x00000d1c) [R X] @0x298

[+] Loading usable ELF segments:
[+]    1: V 0x00102000 P 0x00102000 (0x0003ff3c->0x0003ff3c) [R X] @0xb8
[!!] 🚨 Error: PHDR is not contained in Flash
[!!] 🚨 Firmware is probably too big for your device
[!!] 🚨 See README.md for information on compiling for platforms with 256KB of flash memory
[!] ⚠️  The flashing procedure failed, follow the suggested steps!
[+] All done

[=] Have a nice day!
```
Note that I am able to build a RDV4 firmware image from source and flash it to the other device using the corresponding command. 

**Additional context**
Other users on Discord have noticed the error. No apparent fix is known for the bug. 

Here is more information about the device:
```bash
$ lsusb | ggrep -i proxmark
Bus 020 Device 011: ID 9ac4:4b8f 9ac4 proxmark3 

$ ioreg -w0 -rc IOUSBHostInterface -l
+-o IOUSBHostInterface@0  <class IOUSBHostInterface, id 0x100001d1c, registered, matched, active, busy 0 (4 ms), retain 11>
  | {
  |   "USBSpeed" = 1
  |   "iInterface" = 0
  |   "IOServiceLegacyMatchingRegistryID" = 4294974750
  |   "bInterfaceProtocol" = 1
  |   "bAlternateSetting" = 0
  |   "idProduct" = 19343
  |   "bcdDevice" = 1
  |   "USB Product Name" = "proxmark3"
  |   "locationID" = 341835776
  |   "bInterfaceClass" = 2
  |   "bInterfaceSubClass" = 2
  |   "IOCFPlugInTypes" = {"2d9786c6-9ef3-11d4-ad51-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   "USBPortType" = 0
  |   "bConfigurationValue" = 1
  |   "bInterfaceNumber" = 0
  |   "USB Vendor Name" = "proxmark.org"
  |   "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
  |   "idVendor" = 39620
  |   "bNumEndpoints" = 1
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "IOClassNameOverride" = "IOUSBInterface"
  | }
  | 
  +-o AppleUSBACMControl  <class AppleUSBACMControl, id 0x100001d21, registered, matched, active, busy 0 (0 ms), retain 7>
      {
        "IOProbeScore" = 60000
        "CFBundleIdentifier" = "com.apple.driver.usb.cdc.acm"
        "IOProviderClass" = "IOUSBHostInterface"
        "IOClass" = "AppleUSBACMControl"
        "bInterfaceClass" = 2
        "CFBundleIdentifierKernel" = "com.apple.driver.usb.cdc.acm"
        "IOMatchCategory" = "IODefaultMatchCategory"
        "InterruptPipeReturn" = 0
        "bInterfaceSubClass" = 2
        "bInterfaceProtocol" = 1
      }

+-o IOUSBHostInterface@1  <class IOUSBHostInterface, id 0x100001d1d, registered, matched, active, busy 0 (240 ms), retain 9>
  | {
  |   "USBSpeed" = 1
  |   "iInterface" = 0
  |   "IOServiceLegacyMatchingRegistryID" = 4294974752
  |   "bInterfaceProtocol" = 0
  |   "bAlternateSetting" = 0
  |   "idProduct" = 19343
  |   "bcdDevice" = 1
  |   "USB Product Name" = "proxmark3"
  |   "locationID" = 341835776
  |   "Product Name" = "proxmark3"
  |   "bInterfaceClass" = 10
  |   "bInterfaceSubClass" = 0
  |   "IOCFPlugInTypes" = {"2d9786c6-9ef3-11d4-ad51-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   "USBPortType" = 0
  |   "bConfigurationValue" = 1
  |   "bInterfaceNumber" = 1
  |   "USB Vendor Name" = "proxmark.org"
  |   "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
  |   "idVendor" = 39620
  |   "IODEXTMatchCount" = 1
  |   "bNumEndpoints" = 2
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "IOClassNameOverride" = "IOUSBInterface"
  | }
  | 
  +-o AppleUSBACMData  <class AppleUSBACMData, id 0x100001d24, registered, matched, active, busy 0 (0 ms), retain 7>
    | {
    |   "IOClass" = "AppleUSBACMData"
    |   "CFBundleIdentifier" = "com.apple.driver.usb.cdc.acm"
    |   "IOProviderClass" = "IOUSBHostInterface"
    |   "IOTTYBaseName" = "usbmodem"
    |   "idProduct" = 19343
    |   "IOProbeScore" = 49998
    |   "bInterfaceSubClass" = 0
    |   "HiddenPort" = Yes
    |   "IOMatchCategory" = "IODefaultMatchCategory"
    |   "IOMatchDefer" = Yes
    |   "idVendor" = 39620
    |   "CFBundleIdentifierKernel" = "com.apple.driver.usb.cdc.acm"
    |   "IOTTYSuffix" = "14601"
    |   "bInterfaceClass" = 10
    | }
    | 
    +-o IOSerialBSDClient  <class IOSerialBSDClient, id 0x100001d26, registered, matched, active, busy 0 (0 ms), retain 5>
        {
          "IOClass" = "IOSerialBSDClient"
          "CFBundleIdentifier" = "com.apple.iokit.IOSerialFamily"
          "IOProviderClass" = "IOSerialStreamSync"
          "IOTTYBaseName" = "usbmodem"
          "IOSerialBSDClientType" = "IORS232SerialStream"
          "IOProbeScore" = 1000
          "IOResourceMatch" = "IOBSD"
          "IOCalloutDevice" = "/dev/cu.usbmodem14601"
          "IOMatchCategory" = "IODefaultMatchCategory"
          "IOTTYDevice" = "usbmodem14601"
          "IODialinDevice" = "/dev/tty.usbmodem14601"
          "IOPersonalityPublisher" = "com.apple.iokit.IOSerialFamily"
          "CFBundleIdentifierKernel" = "com.apple.iokit.IOSerialFamily"
          "IOTTYSuffix" = "14601"
        }
```

maxieds, 2022-03-02:
Note that the version of the PM3 Easy device I have does have 512M available.

doegox, 2022-03-02:
ok so you've trouble flashing the pm3 easy on OSX?

Did you use `--with-generic` as mentioned at https://github.com/RfidResearchGroup/homebrew-proxmark3 ?

Did you try the steps described in the error you pasted ? 

```
!!] 🚨  1)   ./pm3-flash-bootrom
[!!] 🚨  2)   ./pm3-flash-all
[!!] 🚨  3)   ./pm3
```

secit-pl, 2022-03-07:
In some cases you need to hold the PM3 button all the time while flashing firmware (that was in my case).

maxieds, 2022-03-26:
@secit-pl 
Thank you! Thank you! Thank you! Holding the button down while running
```
pm3-flash-bootrom && pm3-flash-all
```
Fixed the problem. :)

@doegox Perhaps we should consider creating a WIKI or docs page just for PM3 Easy users?

closed 2022-03-26 by iceman1001