RfidResearchGroup/proxmark3 #1640: Uart errors in latest build

quentynt, 2022-03-27:
in recent build ( as of todays date) i now get constant UART errors in any action - build was with out error, flash was without error. Re cloning and rebuilding / flashing results in the same error on Linux mint ( ubuntu). This is the same laptop / lunux build i have used for a whie without errors 

Uname -a 

Linux elite840 5.4.0-105-generic #119-Ubuntu SMP Mon Mar 7 18:49:24 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

HW version

[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-500-g7fdcad853 2022-03-27 21:24:28 1e1882337
  compiled with............. GCC 9.4.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: Iceman/master/v4.14831-500-g7fdcad853 2022-03-27 21:24:57 1e1882337
       os: Iceman/master/v4.14831-500-g7fdcad853 2022-03-27 21:25:13 1e1882337
  compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 64% used )

HW status 

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 38808
[#]   Available memory........ 38808
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode....................
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD567A882A78EC626
[#] Smart card module (ISO 7816)
[#]   version................. v2.06
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 283136
[#]   Transfer Speed PM3 -> Client... 566272 bytes/s
[#] Various
[#]   Max stack usage......... 4096 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30483 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#] 

HW tune 

[usb] pm3 --> hw tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕕  12
[=] ---------- LF Antenna ----------
[!] ⚠️  LF antenna is UNUSABLE
[=] ---------- HF Antenna ----------
[!] ⚠️  HF antenna is UNUSABLE

(*) Q factor must be measured without tag on the antenna

[-] ⛔ Not showing LF tuning graph since all values is zero.

**Additional context**

[usb] pm3 --> hf search
 🕗  Searching for ThinFilm tag...[!] ⚠️  timeout while waiting for reply.
 🕛  Searching for ISO14443-A tag...[!!] 🚨 UART:: write time-out
 🕑  Searching for ISO15693 tag...[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out
 🕓  Searching for iCLASS / PicoPass tag...[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out

[+] Valid iCLASS tag / PicoPass tag found

 🕕  Searching for LEGIC tag...[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out
 🕗  Searching for Topaz tag...[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out
 🕙  Searching for ISO14443-B tag...[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out
 🕛  Searching for FeliCa tag...[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out

[+] Valid ISO 18092 / FeliCa tag found

[!!] 🚨 UART:: write time-out                 
[!!] 🚨 UART:: write time-out

iceman1001, 2022-03-28:
Did you flash the device?  Since your output states you don't have a fpga image loaded...

iceman1001, 2022-03-28:
the fpga_ring buffer seem to be the problem,   I let me adapt it.

commit d5205e5a (2022-03-28): fix #1640 the btaddon compilation leaves the bigbuff malloc around 38kb.  By lessen the fpga ring buffer to be 30kb instead leaves some room for a smaller bigbuff