RfidResearchGroup/proxmark3 #1648: HF14a - apdufind: EXC_BAD_ACCESS / bus error

uhei, 2022-03-31:
**Describe the bug**
Running `hf 14a apdufind` crashes the `proxmark3` client.

**To Reproduce**
Steps to reproduce the behavior:
1. Put 14a tag on proxmark
2. Run `hf 14a apdufind`
3. `proxmark3` client crash with error `Bus error: 10`

**Expected behavior**
Output of enumerating APDUs

**Screenshots**
```
deckard:proxmark3 uhei$ lldb -- ./proxmark3 /dev/tty.usbmodemiceman1 
(lldb) target create "./proxmark3"
Current executable set to 'proxmark3' (x86_64).
(lldb) settings set -- target.run-args  "/dev/tty.usbmodemiceman1"
(lldb) run
Process 17992 launched: '/usr/local/bin/proxmark3' (x86_64)
[=] Session log /Users/uhei/.proxmark3/logs/log_20220331.txt
[+] loaded from JSON file /Users/uhei/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC

  ██████╗ ███╗   ███╗█████╗ 
  ██╔══██╗████╗ ████║╚═══██╗
  ██████╔╝██╔████╔██║ ████╔╝
  ██╔═══╝ ██║╚██╔╝██║ ╚══██╗
  ██║     ██║ ╚═╝ ██║█████╔╝ 
  ╚═╝     ╚═╝     ╚═╝╚════╝     [ Iceman ❄️ ]

2022-03-31 11:47:49.943131+0200 proxmark3[17992:2374999] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=17992
2022-03-31 11:47:49.943253+0200 proxmark3[17992:2374999] SecTaskCopyDebugDescription: proxmark3[17992]/0#-1 LF=0

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  RRG/Iceman/master/v4.14434-400-gfc13b6b20 2022-01-12 10:07:27
  compiled with............. Clang/LLVM Apple LLVM 13.0.0 (clang-1300.0.29.30)
  platform.................. OSX / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: Iceman/master/v4.14831-518-g8ccda0ea9 2022-03-30 21:51:12 no sha256
       os: Iceman/master/v4.14831-518-g8ccda0ea9 2022-03-30 21:51:36 no sha256
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 65% used )

[usb] pm3 --> hf 14a apdufind
Process 17992 stopped
* thread #7, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x70000480eff8)
    frame #0: 0x00007ff8101c7fa7 libsystem_pthread.dylib`___chkstk_darwin + 55
libsystem_pthread.dylib`___chkstk_darwin:
->  0x7ff8101c7fa7 <+55>: testq  %rcx, -0x8(%rcx)
    0x7ff8101c7fab <+59>: cmpq   $0x1000, %rax             ; imm = 0x1000 
    0x7ff8101c7fb1 <+65>: jb     0x7ff8101c7fcd            ; <+93>
    0x7ff8101c7fb3 <+67>: pushq  %rax
Target 0: (proxmark3) stopped.
(lldb) 
```

**Desktop:**
 - OS: macos 12.3
 - hw version => see above
 ```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 38808
[#]   Available memory........ 38808
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD567A882A792A926
[#] Smart card module (ISO 7816)
[#]   version................. v2.06
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 292864
[#]   Transfer Speed PM3 -> Client... 585728 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30658 Hz
[#] Installed StandAlone Mode
[#]   HF Legic Prime simulation mode - a.k.a LEGICSIM
[#] Flash memory dictionary loaded
[#] 
```
 ```
[usb] pm3 --> data tune
[#] DEBUG: (setClockGrid) clear settings
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕐  20[#] Got Waiting Time eXtension request 1500 ms
 🕛   9[#] Got Waiting Time eXtension request 1500 ms

[=] ---------- LF Antenna ----------
[+] LF antenna: 70,47 V - 125,00 kHz
[+] LF antenna: 31,92 V - 134,83 kHz
[+] LF optimal: 70,47 V - 125,00 kHz
[+] Approx. Q factor (*): 12,0 by frequency bandwidth measurement
[+] Approx. Q factor (*): 12,3 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 43,89 V - 13.56 MHz
[+] Approx. Q factor (*): 7,7 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134,83 kHz, 95 (red) is 125,00 kHz.
```

iceman1001, 2022-03-31:
hm... is that the latest client you are running?  It doesn't look like it... or is that ascii version is different on osx..

uhei, 2022-03-31:
Ah sorry, output of try with older client. This issue also happens with the latest client:
```
deckard:proxmark3 uhei$ client/proxmark3 /dev/tty.usbmodemiceman1 
[=] Session log /Users/uhei/.proxmark3/logs/log_20220331.txt
[+] loaded from JSON file /Users/uhei/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ❄️ ]

  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 Kb ( 65% used )

    Client.... Iceman/master/v4.14831-518-g8ccda0ea9 2022-03-31 22:15:49
    Bootrom... Iceman/master/v4.14831-518-g8ccda0ea9 2022-03-30 21:51:12 
    OS........ Iceman/master/v4.14831-518-g8ccda0ea9 2022-03-30 21:51:36 
    Target.... RDV4

[usb] pm3 --> hf 14a apdufind
Bus error: 10
```

iceman1001, 2022-03-31:
I can't replicate your bug.  Neither with a card nor with-out a card on the antenna and with same compilation params.

uhei, 2022-04-01:
Yes, after disabling BTADDON and removing standalone mode, `hf 14a apdu` runs without crashing.

Sorry, for opening the issue.

closed 2022-04-01 by uhei

uhei, 2022-04-01:
When opening a shell via ssh at the system where proxmark is connected running `hf 14a apdufind` works.
However `proxmark3` client crashes when running on the native shell:
```
deckard:client uhei$ pwd
/Users/uhei/Downloads/nfc-zeugs/proxmark3/client
deckard:client uhei$ ./proxmark3 /dev/tty.usbmodemiceman1 
[=] Session log /Users/uhei/.proxmark3/logs/log_20220401.txt
[+] loaded from JSON file /Users/uhei/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ❄️ ]

  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 Kb ( 64% used )

    Client.... Iceman/master/v4.14831-523-gabcc06125 2022-04-01 13:16:26
    Bootrom... Iceman/master/v4.14831-523-gabcc06125 2022-04-01 10:52:22 
    OS........ Iceman/master/v4.14831-523-gabcc06125 2022-04-01 10:52:42 
    Target.... RDV4

[usb] pm3 --> hf 14a apdufind
Bus error: 10
deckard:client uhei$ ssh localhost
(uhei@localhost) Password:
Last login: Fri Apr  1 17:51:11 2022 from ::1

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
deckard:~ uhei$ cd Downloads/nfc-zeugs/proxmark3/client/
deckard:client uhei$ ./proxmark3 /dev/tty.usbmodemiceman1 
[=] Session log /Users/uhei/.proxmark3/logs/log_20220401.txt
[+] loaded from JSON file /Users/uhei/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ❄️ ]

  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 Kb ( 64% used )

    Client.... Iceman/master/v4.14831-523-gabcc06125 2022-04-01 13:16:26
    Bootrom... Iceman/master/v4.14831-523-gabcc06125 2022-04-01 10:52:22 
    OS........ Iceman/master/v4.14831-523-gabcc06125 2022-04-01 10:52:42 
    Target.... RDV4

[usb] pm3 --> hf 14a apdufind
[=] Sending a test APDU (select file command) to check if the tag is responding to APDU
[=] Press <Enter> to exit

[+] Starting the APDU finder [ CLA 00 INS 00 P1 00 P2 00 ]
[=] Got response for APDU "00220000": 6B00 (Wrong parameter(s) P1-P2)
^C
```

I've already tried to sort out env variables. No luck so far.

iceman1001, 2022-04-01:
something is going on.   What is your metal?

uhei, 2022-04-01:
Running on an iMac 2015 with a 'Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz'.

Any hints how I can debug this issue?

iceman1001, 2022-04-02:
did you install all dependencies?  what has changed in your OS since you first started to see this behavior?
and I have no idea why.