RfidResearchGroup/proxmark3 #1652: ISO 14443B simulation does not work

jacopo-j, 2022-04-04:
**Describe the bug**
The `hf 14b sim` feature does not seem to work properly: readers do not detect the simulated tag.

**To Reproduce**
1. Run `hf 14b sim -u 11AA33BB`

**Expected behavior**
External readers detect the presence of a tag.

**Desktop (please complete the following information):**
 - OS: macOS

 **hw version**

```

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-530-g8236de119-dirty-unclean 2022-04-04 00:23:11 917abd9ba
  compiled with............. Clang/LLVM Apple LLVM 13.1.6 (clang-1316.0.21.2)
  platform.................. OSX / aarch64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.14831-530-g8236de119-dirty-unclean 2022-04-03 15:24:18 e49e4ed9a
       os: Iceman/master/v4.14831-530-g8236de119-dirty-unclean 2022-04-04 00:23:13 917abd9ba
  compiled with GCC 10.3.1 20210824 (release)

 [ FPGA ]
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 58% used )
```

 **hw status**

```
[#] Memory
[#]   BigBuf_size............. 42784
[#]   Available memory........ 42784
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 300032
[#]   Transfer Speed PM3 -> Client... 600064 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32120 Hz
[#] Installed StandAlone Mode
[#]  HF 14B SNIFF,  a ISO14443b sniffer
```

 **data tune**

```
[=] ---------- HF Antenna ----------
[+] HF antenna: 31,08 V - 13.56 MHz
[+] Approx. Q factor (*): 9,0 by peak voltage measurement
[+] HF antenna is OK
```

**Additional context**
- ISO 14443A emulation works properly
- Traces show that the emulated tag correctly responds to the reader WUPB commands, however, the reader does not seem to "hear" the tag. This suggests it may be an encoding/modulation problem. The same happens with multiple different readers.

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |          0 | Rdr |05  00  08  39  73                                                       |  ok | WUPB
          0 |          0 | Tag |50  11  aa  33  bb  20  38  19  22  00  21  85  7e  59                   |  ok |
          0 |          0 | Rdr |05  00  08  39  73                                                       |  ok | WUPB
          0 |          0 | Tag |50  11  aa  33  bb  20  38  19  22  00  21  85  7e  59                   |  ok |
          0 |          0 | Rdr |05  00  08  39  73                                                       |  ok | WUPB
          0 |          0 | Tag |50  11  aa  33  bb  20  38  19  22  00  21  85  7e  59                   |  ok |
          0 |          0 | Rdr |05  00  08  39  73                                                       |  ok | WUPB
          0 |          0 | Tag |50  11  aa  33  bb  20  38  19  22  00  21  85  7e  59                   |  ok |
          0 |          0 | Rdr |05  00  08  39  73                                                       |  ok | WUPB
          0 |          0 | Tag |50  11  aa  33  bb  20  38  19  22  00  21  85  7e  59                   |  ok |
          0 |          0 | Rdr |05  00  08  39  73                                                       |  ok | WUPB
          0 |          0 | Tag |50  11  aa  33  bb  20  38  19  22  00  21  85  7e  59                   |  ok |
          0 |          0 | Rdr |05  00  08  39  73                                                       |  ok | WUPB
          0 |          0 | Tag |50  11  aa  33  bb  20  38  19  22  00  21  85  7e  59                   |  ok |
          0 |          0 | Rdr |05  00  08  39  73                                                       |  ok | WUPB
          0 |          0 | Tag |50  11  aa  33  bb  20  38  19  22  00  21  85  7e  59                   |  ok |
```

codecat007, 2022-04-13:
I tested on Raspberry Pi 3B, ISO 14443B simulation also does not work. It seems there is a problem with this feature.

iceman1001, 2022-04-13:
`hf 14b`  commands would need some more love.  Feel free to contribute!

jacopo-j, 2022-05-11:
@valtoo16 that is interesting, can you share a repository or a gist with the complete code?

iceman1001, 2022-05-11:
@valtoo16   If you have a fix for 14b,  you are welcome to make a PR

iceman1001, 2023-10-13:
@valtoo16      your code?

AkechiShiro, 2023-11-28:
Hi I'd like to help work on this feature, but I have never contributed to this project, what would be a good way to start making progress for this feature ?

iceman1001, 2023-11-28:
There are no implementation of a 14B  protocol to start with.  
So you would need to first to that,  then you would debug with reader/sniffer

iceman1001, 2023-11-28:
I love to see you contribute!

iceman1001, 2023-11-28:
you need 14B data sheets,  then you can look at `armsrc/iso14443b.c`  to start with...

iceman1001, 2024-01-26:
`hf 14b commands` have gotten some serious love and works better now. 

Feel free to improve the simulation.

AkechiShiro, 2024-01-27:
Thanks a lot for letting us know here, I'll try and test it at some point (when I have some free time), I'll report on it when I do and try to see if I we can work on documenting what works and what doesn't, I guess.

michi-jung, 2024-06-11:
Dumping the precomputed ATQB response frame from ISO14443-B emulation:

```
[#] ff ff ff ff ff 00 00 00 00 00 ff 00 00 0f 0f 0f  1111111111 0000000000 11 0 00001010 1
[#] 0f 00 0f 00 0f 00 f0 f0 f0 ff 0f f0 0f f0 0f 0f  0 10001000 1 0 01010101 1 0 11001100 1 0 1
[#] f0 ff f0 ff 00 00 00 f0 0f 00 00 ff f0 0f 0f 00  1011101 1 0 00000100 1 0 00011100 1 0 100
[#] ff 00 0f 00 f0 00 f0 0f 00 00 00 00 0f 0f 00 00  11000 1 0 01000100 1 0 00000000 1 0 10000
[#] f0 0f 0f 0f 00 00 ff 00 ff ff ff 0f 0f 00 ff 0f  100 1 0 10100001 1 0 01111110 1 0 1001101
[#] 0f 00 00 00 00 00                                0 1 0000000000

1111111111 Unmodulated subcarrier (10 ETUs)
0000000000 SoS t_PICC,S,1 (10 ETUs)
11         SoS t_PICC,S,2 (2 ETUs)
0          start bit
00001010   x50
1          stop bit
0          start bit
10001000   x11
1
0
01010101   xAA
1
0
11001100   x33
1
0
11011101   xbb
1
0
00000100   x20
1
0
00011100   x38
1
0
10011000   x19
1
0
01000100   x22
1
0
00000000   x00
1
0
10000100   x21
1
0
10100001   x85
1
0
01111110   x7e
1
0
10011010   x59
1
0000000000 EoS (10 ETU)
```

Which correlates to the proxmark3 trace:
```
          0 |      65535 | Rdr |05  00  08  39  73                                                       |     | 
          0 |      65535 | Tag |50! 11! aa! 33! bb! 20  38  19  22! 00! 21! 85  7e! 59!                  |     | 
```

However, on the scope the bit coding is reversed, the TR0 MIN is not respected, and there is a garbage '1' bit at the beginning:

[image]

iceman1001, 2024-06-12:
@michi-jung    That is awesome!     

Now if @valtoo16  give access to his simulation setup we can even take 14b sim to the next level

closed 2024-06-12 by iceman1001