RfidResearchGroup/proxmark3 #1656: Mifare Classic 1K  Magic Gen 1a card bricked after writing with `hf mf cload

uspilot, 2022-04-08:
**Describe the bug**
Mifare Classic 1K Magic Gen 1a card bricked after writing with `hf mf cload`
**To Reproduce**
Steps to reproduce the behavior:
1. Place working Magic Gen1 card on PM3 device:  `pm3 --> hf search`
2. Write card: `pm3 --> hf mf cload -f hf-mf-2F64EF3E-dump-1`
3. No error
4. Check card: `pm3 --> hf search`
5. Can't read card

**Expected behavior**
Card must be working

**Screenshots**
`[usb] pm3 --> hf search`
 🕐  Searching for ISO14443-A tag...          
[+]  UID: 1A B6 F8 16 
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Magic capabilities : Gen 1a
[+] Prng detection: weak
[#] Auth error

[+] Valid ISO 14443-A tag found

**[usb] pm3 --> hf mf cload -f hf-mf-2F64EF3E-dump-1**
[+] loaded 1024 bytes from binary file hf-mf-2F64EF3E-dump-1
[=] Copying to magic gen1a card
[=] .................................................................

[+] Card loaded 64 blocks from file
[=] Done!

`[usb] pm3 --> hf search`
 🕕  Searching for ISO14443-A tag...[#] BCC0 incorrect, got 0x00, expected 0x33
[#] Aborting
[!] ⚠️  No known/supported 13.56 MHz tags found

**Desktop (please complete the following information):**
 - OS: macOS Catalina Version: 10.15.6 (19G2021)
`[usb] pm3 --> hw version`

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  RRG/Iceman/master/v4.14831-2-g4e78034a6 2022-01-11 22:33:23
  compiled with............. Clang/LLVM Apple LLVM 12.0.0 (clang-1200.0.32.29)
  platform.................. OSX / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.14831-2-g4e78034a6 2022-01-11 22:34:09
       os: RRG/Iceman/master/v4.14831-2-g4e78034a6 2022-01-11 22:34:30
  compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30
[... 127 lines cut ...]
[=]  40 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  41 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  42 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  43 | FF FF FF FF FF FF FF 07 80 00 FF FF FF FF FF FF | ................
[=]  44 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  45 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  46 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  47 | FF FF FF FF FF FF FF 07 80 00 FF FF FF FF FF FF | ................
[=]  48 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  49 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  50 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  51 | FF FF FF FF FF FF FF 07 80 00 FF FF FF FF FF FF | ................
[=]  52 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  53 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  54 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  55 | FF FF FF FF FF FF FF 07 80 00 FF FF FF FF FF FF | ................
[=]  56 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  57 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  58 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  59 | FF FF FF FF FF FF FF 07 80 00 FF FF FF FF FF FF | ................
[=]  60 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  61 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  62 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  63 | FF FF FF FF FF FF FF 07 80 00 FF FF FF FF FF FF | ................
[=] ----+-------------------------------------------------+-----------------

mwalker33, 2022-04-08:
A few questions.
1. Given that its a magic gen1a card, is it really bricked, or just not working.  
I made a gen1a card with that data and got an error on normal commands, but could write a valid block 0 back and the card still works.

2. There seems to be something wrong with the block 0 data.
While I have only had a very quick look... the error I got was 
```
[usb] pm3 --> hf 14a info
[#] BCC0 incorrect, got 0x00, expected 0x33
[#] Aborting
```
So I change the 5th byte in block 0 to 0x33
```
original : hf mf csetblk --blk 0 -d 3400070000D182BB5300000000000700
updated  : hf mf csetblk --blk 0 -d 3400070033D182BB5300000000000700
```
Which then allows the info to work
```
[usb] pm3 --> hf 14a info

[+]  UID: 34 00 07 00
[+] ATQA: BB 82
[+]  SAK: D1 [2]
[+] Possible types:
[+]    MIFARE Plus 4K in SL2
[=] proprietary non iso14443-4 card found, RATS not supported
```
But there could be more not 100% correct with that block 0.
Was the original card a 4 byte or 7 Byte UID ?

uspilot, 2022-04-08:
```
[usb] pm3 --> hf 14a info

[+]  UID: 2F 64 EF 3E 
[+] ATQA: 00 44
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K CL2
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Prng detection: hard
[#] Auth error
```

Dump has been generated with `hf mf autopwn` command

```
[usb] pm3 --> hf mf autopwn

[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 42 keys from hardcoded default array
[=] running strategy 1
[=] Chunk 0.6s | found 32/32 keys (42)
[+] target sector   0 key type A -- found valid key [ A0A1A2A3A4A5 ] (used for nested / hardnested attack)
[+] target sector   0 key type B -- found valid key [ B0B1B2B3B4B5 ]
[+] target sector   1 key type A -- found valid key [ A0A1A2A3A4A5 ]
[+] target sector   1 key type B -- found valid key [ B0B1B2B3B4B5 ]
[+] target sector   2 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   3 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   3 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   4 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   4 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   5 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   5 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   6 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   6 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   7 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   7 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   8 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   8 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   9 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   9 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  10 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  10 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  11 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  11 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  12 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  12 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  13 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  13 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  14 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  14 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  15 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  15 key type B -- found valid key [ FFFFFFFFFFFF ]

[+] found keys:

[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | a0a1a2a3a4a5   | D | b0b1b2b3b4b5   | D |
[+] | 001 | a0a1a2a3a4a5   | D | b0b1b2b3b4b5   | D |
[+] | 002 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 003 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 004 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 005 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 006 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 007 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 008 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 009 | ffffffffffff   | D | ffffffffffff   | D |
[... 102 lines cut ...]
```
[usb] pm3 --> hf mf rdsc -k a0a1a2a3a4a5 -s 0 -v

[=]   # | sector 00 / 0x00                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   0 | 34 00 07 00 00 D1 82 BB 53 00 00 00 00 00 07 00 | 4.......S.......
[=]   1 | 01 00 F4 71 09 67 64 15 07 13 00 1B 30 2E 02 15 | ...q.gd.....0... 
[=]   2 | 00 DD B7 42 88 8D D4 01 A2 B1 25 A4 F7 4A 6F B0 | ...B......%..Jo. 
[=]   3 | 00 00 00 00 00 00 FF 07 80 69 B0 B1 B2 B3 B4 B5 | .........i......

[=] ----------------------- Sector trailer decoder -----------------------
[=] key A........ 000000000000
[=] acr.......... FF0780
[=] user / gpb... 69
[=] key B........ B0B1B2B3B4B5

[=]   # | Access rights
[=] ----+-----------------------------------------------------------------
[=]   0 | read AB; write AB; increment AB; decrement transfer restore AB
[=]   1 | read AB; write AB; increment AB; decrement transfer restore AB
[=]   2 | read AB; write AB; increment AB; decrement transfer restore AB
[=]   3 | write A by A; read/write ACCESS by A; read/write B by A
[=] ----------------------------------------------------------------------

```

mwalker33, 2022-04-08:
Can I get you to post the sector 0 of the original.

So far it seems that the restore (cload) worked as expected, but the dump has an incorrect block 0.  (I want to test with 100% the same block 0, as the original)

Thanks

mwalker33, 2022-04-08:
Note: I can make it happen.    For testing, I updated my client the the very latest (as at now)
I setup a card based on your dump from above (with a working block 0, that may not match yours)

```
hf mf autopwn
```
Block 0 is what was on my test card
```
[usb] pm3 --> hf mf view -f hf-mf-2F64EF3E-dump-2
[+] loaded 1024 bytes from binary file hf-mf-2F64EF3E-dump-2

[=] ----+-------------------------------------------------+-----------------
[=] blk | data                                            | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   0 | 2F 64 EF 3E 9A D1 82 BB 53 00 00 00 00 00 07 00 | /d.>....S.......
[=]   1 | 01 00 F4 71 09 67 64 15 07 13 00 1B 30 2E 02 15 | ...q.gd.....0...
```

Restore to a gen1a card
```
[usb] pm3 --> hf mf cload -f hf-mf-2F64EF3E-dump-2.bin
[+] loaded 1024 bytes from binary file hf-mf-2F64EF3E-dump-2.bin
[=] Copying to magic gen1a card
[=] .................................................................

[+] Card loaded 64 blocks from file
[=] Done!
```

hf search finds the correct card (based on my block 0)
```
[usb] pm3 --> hf mf cload -f hf-mf-2F64EF3E-dump-2.bin
[+] loaded 1024 bytes from binary file hf-mf-2F64EF3E-dump-2.bin
[=] Copying to magic gen1a card
[=] .................................................................

[+] Card loaded 64 blocks from file
[=] Done!
```

So unless there is some weirdness in your original card, I cant see why it would not work.
Lets see what happens when I get your original block 0.
Thanks

uspilot, 2022-04-08:
This is ORIGINAL card
It looks very strange, but it have no idea why is block 0 don't read correct 
```
[usb] pm3 --> hf mf rdsc -k a0a1a2a3a4a5 -s 0 -v

[=]   # | sector 00 / 0x00                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   0 | 34 00 07 00 00 D1 82 BB 53 00 00 00 00 00 07 00 | 4.......S.......
[=]   1 | 01 00 F4 71 09 67 64 15 07 13 00 1B 30 2E 02 15 | ...q.gd.....0... 
[=]   2 | 00 DD B7 42 88 8D D4 01 A2 B1 25 A4 F7 4A 6F B0 | ...B......%..Jo. 
[=]   3 | 00 00 00 00 00 00 FF 07 80 69 B0 B1 B2 B3 B4 B5 | .........i......

[=] ----------------------- Sector trailer decoder -----------------------
[=] key A........ 000000000000
[=] acr.......... FF0780
[=] user / gpb... 69
[=] key B........ B0B1B2B3B4B5

[=]   # | Access rights
[=] ----+-----------------------------------------------------------------
[=]   0 | read AB; write AB; increment AB; decrement transfer restore AB
[=]   1 | read AB; write AB; increment AB; decrement transfer restore AB
[=]   2 | read AB; write AB; increment AB; decrement transfer restore AB
[=]   3 | write A by A; read/write ACCESS by A; read/write B by A
[=] ----------------------------------------------------------------------
```
```
[usb] pm3 --> hf 14a info

[+]  UID: 2F 64 EF 3E 
[+] ATQA: 00 44
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K CL2
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Prng detection: hard
[#] Auth error

```

mwalker33, 2022-04-08:
Thanks.
So now we can say that the proxmark dump does contain what it read and the cload did restore what was in the dump.

As to why block 0 does not match the UID seen, no idea.  I am wondering if its protected somehow (started to get outside of my mifare classic skill set), just a guess.

Maybe someone else has some idea.

uspilot, 2022-04-08:
Yes, it seems like some kind of protection because I have the same readout with Chinese firmware on PM3 Easy.

iceman1001, 2022-04-09:
Closing, since its no issue in the code base.   I suggest you take it to discord and continue in the _mifare_ channel

closed 2022-04-09 by iceman1001