RfidResearchGroup/proxmark3 #1661: getSamplesEx() doesn't work

wh201906, 2022-04-09:
**Describe the bug**  
1. The `hf sniff` result will be affected by `lf config`. 
  The `getSamplesEx()` function in `client/src/cmddata.c` will ALWAYS use the `bits_per_sample` option set in `lf config`, but this function is also called when running `hf sniff`. So the `getSamplesEx()` will use `bits_per_sample` when unpacking the hf sniff data.
2. The `getSamplesEx()` doesn't work when bits_per_sample is not 8.
  If I set BPS to 1~7, the data of `lf sniff` is weird. Also, when I run `lf sniff -s 1000 -v` with different BPS, the numbers of samples are not the same.
  The `getSamplesEx()` function uses the length in bytes as the number of samples, which is wrong when BPS is not 8.

**To Reproduce**
Steps to reproduce the behavior:
For the first bug:
1. Compile the client with verbose
2. run `lf config -b 6`(set BPS to any value except 8)
3. run `hf sniff`, the output will show this bug

For the second bug:
1. run `lf config -b 6`(set BPS to any value except 8)
2. run `lf sniff -s 1000 -v`
3. run `data plot`. You will see the number of samples is not 1000, and the data is weird.

**Expected behavior**
For the first bug:
The `lf config` will not affect `hf sniff`
For the second bug:
`lf sniff` should work with any BPS

**Screenshots**
For the first bug:
```
[usb] pm3 --> hf sniff
[#] Skipping first 0 sample pairs, Skipping 0 triggers
[=] HF sniff (44360 samples)
[?] Use `data hpf` to remove offset
[?] Use `data plot` to view
[?] Use `data save` to save
[=] Reading 44359 bytes from device memory
[+] Data fetched
[=] Samples @ 6 bits/smpl, decimation 1:1 
[=] Unpacking...
[=] Unpacked 44359 samples
[=] Done.
[usb] pm3 --> lf config

[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 6
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 10
[#]   [s] samples to skip..... 0 
[#] 
```

For the second bug:

```
[usb] pm3 --> lf sniff -s 1000 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 6
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 1000 out of 0 seen samples at 6 bits/sample
[=] Reading 750 bytes from device memory
[+] Data fetched
[=] Samples @ 6 bits/smpl, decimation 1:1 
[=] Unpacking...
[=] Unpacked 750 samples
```
**Additional context**
Test on commit aec3ff64e21c9c1abe2c86b3f61ee7942370ba4e

wh201906, 2022-04-09:
Some .pm3 files with different BPS option.
[data.zip](https://github.com/RfidResearchGroup/proxmark3/files/8456781/data.zip)

iceman1001, 2022-04-09:
Regarding 1)
`LF CONFIG` should not impact any HF commmands.
If it does,  then something is wrong.

Regarding 2)   
Make separate issues,  one per bug. 
I saw the question got answered in discord.
You took 1000 samples of 8bits,  which got compressed down,  leading to the number of samples available on client side.
Just like it is intended.   No bug.

wh201906, 2022-04-09:
Ok
I will separate this issue into two.

wh201906, 2022-04-09:
I've checked the code. The getSamplesEx() is called after running "hf sniff"(line 351 in client/src/cmdhf.c) 
and it does uses the bps argument in the following getByte()(line 1783 in client/src/cmddata.c)
I don't know why the result seems fine.

iceman1001, 2022-04-09:
arm/hfsnoop.c  doesn't use  the device side lf config settings.   

So,  a bit more complex logic bug.

1.  cmdhf.c / CmdHFSniff   downloads full bytes by calling getSamplesEx.
2.  cmddata.c / getSamplesEx   reads  response.oldarg[0]  which is set to 1  in  arm/appmain.c  when done.
3.  that triggers it to applying lf config on a case (hf sniff) where it shall not.

Try pulling latest,  now `hf sniff`  should never trigger the LF config settings part

closed 2022-04-09 by iceman1001

wh201906, 2022-04-09:
I'll try it tomorrow. Thanks!

commit 159d5ffd (2022-04-09): fix #1661  -  hf sniff shall ignore lf config settings