RfidResearchGroup/proxmark3 #1662: The "getSamplesEx()" doesn't work when bits_per_sample is not 8

wh201906, 2022-04-09:
**Describe the bug**  
If I set BPS to 1~7, the data of `lf sniff` is weird. Also, when I run `lf sniff -s 1000 -v` with different BPS, the numbers of samples are not the same.
The `getSamplesEx()` function uses the length in bytes as the number of samples, which is wrong when BPS is not 8.

**To Reproduce**
Steps to reproduce the behavior:
1. run `lf config -b 6`(set BPS to any value except 8)
2. run `lf sniff -s 1000 -v`
3. run `data plot`. You will see the number of samples is not 1000, and the data is weird.

**Expected behavior**
`lf sniff` should work with any BPS

**Screenshots**

```
[usb] pm3 --> lf sniff -s 1000 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 6
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 1000 out of 0 seen samples at 6 bits/sample
[=] Reading 750 bytes from device memory
[+] Data fetched
[=] Samples @ 6 bits/smpl, decimation 1:1 
[=] Unpacking...
[=] Unpacked 750 samples
```
**Additional context**
Test on commit aec3ff64e21c9c1abe2c86b3f61ee7942370ba4e

wh201906, 2022-04-09:
Some .pm3 files with different BPS option.
[data.zip](https://github.com/RfidResearchGroup/proxmark3/files/8456781/data.zip)

iceman1001, 2022-04-09:
I saw the question got answered in discord.
You took 1000 samples of 8bits on device side, which got compressed down, leading to the number of samples available on client side.
Just like it is intended. No bug.

wh201906, 2022-04-09:
For you reply in #1661
> Regarding 2) 
> Make separate issues, one per bug. 
> I saw the question got answered in discord. 
> You took 1000 samples of 8bits, which got compressed down, leading to the number of samples available on client side. 
> Just like it is intended. No bug.

I've checked the code and I guess the smaller the BPS is, the less space per sample will take so the hardware can save more samples, right?

wh201906, 2022-04-09:
Or maybe I just don't know what the exact meaning of -s option in lf sniff is, or wrongly treat BPS as resolution per sample.
If so, could you please tell me how to use them?

wh201906, 2022-04-09:
Also, when bps is not 8, the PM3 hardware might reboot after running `lf sniff` more than once. I only test it on an PM3 easy so I'm not sure is this a software problem or hardware problem.

iceman1001, 2022-04-09:
Lf sniff 
` -s, --samples <dec>            number of samples to collect`
Means number of samples to collect on device side.    The `ll config --bps`  means it will compress the collected samples on device side,  and you are seeing the compressed samples,   which is less than the amount you collected.
That compression is destructive.   So on client side you will only see 750 samples,  if you use 

```
lf config -b 8
lf sniff -s 1000
data plot
```
You will take 1000 samples,  compress it,  and you will view 750 samples in plot window.  

Did that make sense?

iceman1001, 2022-04-09:
And yes,  there seem to be a bug in lf sniff when running it several times...

wh201906, 2022-04-09:
How about make the compression not destructive?
When bps=8, the data in the bigbuf is like 
[sample1 0:7][sample2 8:15][sample3 16:23]...
When bps=6, the data in the bigbuf is like 
[sample1 0:5][sample2 6:11][sample3 12:17]...

If we unpack the data of bps=6 like this
2'b00[sample1 0:5]2'b00[sample2 8:15]2'b00[sample3 16:23]...
Then the plotted data will be more handy

iceman1001, 2022-04-09:
if you want to rewrite a non-destructive compression and apply it to all use-cases on deviceside,  be my guest.

The reason for the destructive compression is because we got very limited amount of ram on device side.
The big buffer is about 40kb of the 64kb.   So wanna waste precious ram to get another compression,  and  not to mention it takes time to execute...

wh201906, 2022-04-09:
I've read the compressing code on the device side. It keeps the x MSB(x=bps).
If we can construct every sample on the client side with the original x MSB and (8-x) LSB, the plotted data will be more useful

iceman1001, 2022-04-09:
That unpacking should already be done in the cmddata.c / getsampleex  function...

wh201906, 2022-04-09:
But the unpacking process is weird. That's why I got the weird .pm3 file above.
My expression on English is bad, so I might not express my thoughts clearly. All I want to do(or request for some help) is to change the getSamplesEx() function so the plotter can draw similar waveforms whatever the bps is.

iceman1001, 2022-04-09:
You are making a LF SNIFF. It's raw signal data. There is no text data.  
We center that signal data from  0-255 ->  127 - -127 for it look good in the plot window.

HF SNIFF is also signal data.   No text.  

When you `HF 14A SNIFF`   you get decoded signal data in form of the ISO14443-a protocol,  all viewable with the trace commands.   Here it differs.

wh201906, 2022-04-09:
My expression is ambiguous. Sorry about that. My "text data" or "ASCII data" refers to the integer array look like "-127,0,127,0,...", rather than the decoded data.

For example,
If I sampled with bps=4, Then I got 
0, 1, 0, 2, 15, 15, 3, 0(8 samples)
The data in the bigbuf is
0x10 0x20 0xFF 0x03(4 bytes)
The data I want to plot is
0, 1, 0, 2, 15, 15, 3, 0(8 dots)
or
-8, -7, -8, -6, 7, 7, -5, -8(8 dots, -8 offset)
However, the unpacking process gives me
-111, -95, 128, -124(4dots, -127 as offset, meanless, inconvenient to convert it back to raw data)

wh201906, 2022-04-09:
> You will take 1000 samples, compress it, and you will view 750 samples in plot window.
> Because we are throwing away 2bits / byte.

Yes, we lost 2 LSB when storing them on device. But I think we should keep every sample when unpacking them, just setting the 2 LSB to zero, then keep the 6 MSB unchanged, then we can add an offset -127

wh201906, 2022-04-10:
I just found something unexpected again. Hope it helps.
if I set bps=8 then run `lf sniff -s 19876 -v` for 3 times, the bytes I get from device memory is always 19876
```
[usb|script] pm3 --> lf sniff -s 19876 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 19876 out of 0 seen samples at 8 bits/sample
[=] Reading 19876 bytes from device memory
[+] Data fetched
[=] Samples @ 8 bits/smpl, decimation 1:1 

[usb|script] pm3 --> lf sniff -s 19876 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 19876 out of 0 seen samples at 8 bits/sample
[=] Reading 19876 bytes from device memory
[+] Data fetched
[=] Samples @ 8 bits/smpl, decimation 1:1 

[usb|script] pm3 --> lf sniff -s 19876 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 19876 out of 0 seen samples at 8 bits/sample
[=] Reading 19876 bytes from device memory
[+] Data fetched
[=] Samples @ 8 bits/smpl, decimation 1:1 
```
if I set bps=7 then run `lf sniff -s 19876 -v` for 3 times, the bytes I get from device memory increases from 17391 to 34783, then it crashes.
```
[usb|script] pm3 --> lf sniff -s 19876 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 7
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 19876 out of 0 seen samples at 7 bits/sample
[=] Reading 17391 bytes from device memory
[+] Data fetched
[=] Samples @ 7 bits/smpl, decimation 1:1 
[=] Unpacking...
[=] Unpacked 17391 samples

[usb|script] pm3 --> lf sniff -s 19876 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 7
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 19876 out of 0 seen samples at 7 bits/sample
[=] Reading 34783 bytes from device memory
[+] Data fetched
[=] Samples @ 7 bits/smpl, decimation 1:1 
[=] Unpacking...
[=] Unpacked 34783 samples

[usb|script] pm3 --> lf sniff -s 19876 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 7
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[!] (lf_read) command execution time out

[!] Communicating with Proxmark3 device failed
```

wh201906, 2022-04-10:
> And yes, there seem to be a bug in lf sniff when running it several times...

I guess it is caused by an overflow...

iceman1001, 2022-04-10:
Found that samples bug,  it was related to a bad initialization of the data stream buffer.  Pull latest and test

iceman1001, 2022-04-10:
the unpacking starts with msb,  and skips all lsb that wasn't included.  so I still have problems with understanding what you are up to.
The signal files is meant to be used with the proxmark client,   nothing else...

wh201906, 2022-04-10:
> Found that samples bug, it was related to a bad initialization of the data stream buffer. Pull latest and test

@iceman1001 the bug still exists...
```
[usb] pm3 --> lf config -t 10 -b 8

[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 10
[#]   [s] samples to skip..... 0 
[#] 
[usb] pm3 --> lf sniff -s 1000 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 10
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 1000 out of 0 seen samples at 8 bits/sample
[=] Reading 1000 bytes from device memory
[+] Data fetched
[=] Samples @ 8 bits/smpl, decimation 1:1 
[usb] pm3 --> lf sniff -s 1000 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 10
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 1000 out of 0 seen samples at 8 bits/sample
[=] Reading 1000 bytes from device memory
[+] Data fetched
[=] Samples @ 8 bits/smpl, decimation 1:1 
[usb] pm3 --> lf config -t 10 -b 6

[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 6
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 10
[#]   [s] samples to skip..... 0 
[#] 
[usb] pm3 --> lf sniff -s 1000 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 6
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 10
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 1000 out of 0 seen samples at 6 bits/sample
[=] Reading 1750 bytes from device memory
[+] Data fetched
[=] Samples @ 6 bits/smpl, decimation 1:1 
[=] Unpacking...
[=] Unpacked 1750 samples
[usb] pm3 --> lf sniff -s 1000 -v
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 6
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 10
[#]   [s] samples to skip..... 0 
[#] 
[#] Done, saved 1000 out of 0 seen samples at 6 bits/sample
[=] Reading 2500 bytes from device memory
[+] Data fetched
[=] Samples @ 6 bits/smpl, decimation 1:1 
[=] Unpacking...
[=] Unpacked 2500 samples
[usb] pm3 --> 

```

iceman1001, 2022-04-10:
did you flash?

wh201906, 2022-04-10:
Yes

iceman1001, 2022-04-10:
it works for me,  can't replicate it since the fix.

wh201906, 2022-04-10:
> it works for me, can't replicate it since the fix.

I did't use the latest firmware when testing. Now it works.

wh201906, 2022-04-10:
Since the overflow issue has been fixed. I think I can close it.
I might make a PR or a new issue when I fully understand how the bps option works.

closed 2022-04-10 by wh201906