RfidResearchGroup/proxmark3 #1667: PM3 will not read any cards anymore

rr775, 2022-04-29:
***Things to try before submitting bug report***
read the [troubleshooting guide](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to Iceman repo.

-

**Describe the bug**
PM3 fails to initialize and parse data on a card, below example has no card anywhere near the device for both lf and hf searches, both fail with a card either way. Operating latest version and cannot seem to get a solution to this through the other issue requests I've found online. Worked fine untouched up until about last week, since then I have reflashed the hardware 5 or 6 times, both macOS and windows, and have found no solution to my issue. 

**To Reproduce**
Steps to reproduce the behavior:
Run lf search or hf search

**Expected behavior**
A clear and concise description of what you expected to happen.
Code should detect a card, but will fail every time

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Macos
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-555-gad79e8433 2022-04-28 22:34:22 f194e98b6
  compiled with............. Clang/LLVM Apple LLVM 13.1.6 (clang-1316.0.21.2.3)
  platform.................. OSX / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
 - hw status
 [usb] pm3 --> hw status
[!!] 🚨 UART:: write time-out
[!] ⚠️  Status command timeout. Communication speed test timed out
 - data tune
[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕗  20[!!] 🚨 UART:: write time-out
 🕓   0
[!] ⚠️  No response from Proxmark3. Aborting...

[usb] pm3 --> 

**Additional context**
Add any other context about the problem here.

[=] Session log /Users/rr/.proxmark3/logs/log_20220429.txt
[+] loaded from JSON file /Users/rr/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ❄️ ]

  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 Kb ( 64% used )

    Client.... Iceman/master/v4.14831-555-gad79e8433 2022-04-28 22:34:22
    Bootrom... Iceman/master/v4.14831-555-gad79e8433 2022-04-28 22:33:29 
    OS........ Iceman/master/v4.14831-555-gad79e8433 2022-04-28 22:33:51 
    Target.... RDV4

[usb] pm3 --> hw ver

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-555-gad79e8433 2022-04-28 22:34:22 f194e98b6
  compiled with............. Clang/LLVM Apple LLVM 13.1.6 (clang-1316.0.21.2.3)
  platform.................. OSX / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.14831-555-gad79e8433 2022-04-28 22:33:29 f194e98b6
       os: Iceman/master/v4.14831-555-gad79e8433 2022-04-28 22:33:51 f194e98b6
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 64% used )

[usb] pm3 --> lf sea
[=] You can cancel this operation by pressing the pm3 button
[!] ⚠️  (lf_read) command execution time out
[-] ⛔ Data in Graphbuffer was too small.
[usb] pm3 --> hf sea
 🕑  Searching for ThinFilm tag...[!] ⚠️  timeout while waiting for reply.
 🕙  Searching for iCLASS / PicoPass tag...   
[+] Valid iCLASS tag / PicoPass tag found

 🕕  Searching for FeliCa tag...              
[+] Valid ISO 18092 / FeliCa tag found

[usb] pm3 -->

iceman1001, 2022-04-29:
Which proxmark3 device do you have?

rr775, 2022-04-29:
RDV4

iceman1001, 2022-04-30:
the fw/client looks ok,  

falls back to usb port,   usb cable,  modem manager,   and if you are on a virtual machine..

rr775, 2022-04-30:
I've tried it on two computers, Mac and Windows, reflashed both scenarios. One was USBC to Micro, one USB to Micro, so troubleshooted with the port, cable, and essentially VM was eliminated too. I'll try a different cable on each, maybe I'm that unlucky

iceman1001, 2022-04-30:
since you followed the installation guides,   and the troubleshooting guide and swapped usb cable, there is not very much more comes to mind.
You can try asking questions in the discord server or pm3 forum.

closed 2022-05-08 by iceman1001