RfidResearchGroup/proxmark3 #1679: Ubuntu 20 arm Pi 4 - error in make clean && make -j

etothex23, 2022-06-06:
I tried to install the proxmark3 in Ubuntu for Pi . I get this error in the make clean && make -j
Here is the teminal output:
-] CC src/emv/emv_pk.c
src/cmdhfmfu.c: In function ‘CmdHF14AMfuEView’:
src/cmdhfmfu.c:2072:97: error: array subscript ‘mfu_dump_t[0]’ is partly outside array bounds of ‘unsigned char[1020]’ [-Werror=array-bounds]
 2072 |     PrintAndLogEx(INFO, "Max data page... " _YELLOW_("%d") " ( " _YELLOW_("%d") " bytes )", card->pages - 1, card->pages * 4);
      |                                                                                             ~~~~^~~~~~~
src/cmdhfmfu.c:4111:21: note: referencing an object of size 1020 allocated by ‘calloc’
 4111 |     uint8_t *dump = calloc(bytes, sizeof(uint8_t));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdhfmfu.c:2068:84: error: array subscript ‘mfu_dump_t[0]’ is partly outside array bounds of ‘unsigned char[1020]’ [-Werror=array-bounds]
 2068 |         PrintAndLogEx(INFO, "Counter %d... %s", i, sprint_hex(card->counter_tearing[i], 3));
      |                                                               ~~~~~~~~~~~~~~~~~~~~~^~~
src/cmdhfmfu.c:4111:21: note: referencing an object of size 1020 allocated by ‘calloc’
 4111 |     uint8_t *dump = calloc(bytes, sizeof(uint8_t));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[-] CC src/emv/emv_pki.c

Not much else happens but after the command finishes you can't run pm3, ./pm3. I saw something about a fedora vesion missing a c lib or something so i was thinking it could be a missing lib or a mismatch in the function between x86 vs arm

iceman1001, 2022-06-06:
Pull latest and try to compile again

iceman1001, 2022-06-06:
hm,   I compile it on ubuntu and no problem.
In the beginning of the compilation you see some information,   cut and paste it here

etothex23, 2022-06-06:
This part:
==================================================================
Version info:      Iceman/master/v4.14831-620-gad64fdfeb
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ZX8211 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/clean
===================================================================
Version info:      Iceman/master/v4.14831-620-gad64fdfeb
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.15.3 in /usr/lib/aarch64-linux-gnu)
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.10 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
===================================================================
===================================================================
PLATFORM Linux
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean
[*] MAKE mf_nonce_brute/clean
[*] MAKE mfd_aes_brute/clean

===================================================================
Version info:      Iceman/master/v4.14831-620-gad64fdfeb
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ZX8211 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================

iceman1001, 2022-06-06:
If you put the test inside tripple back ticks,   it will format better.

```
like this
```

etothex23, 2022-06-06:
```
s===================================================================
s
Version info:      Iceman/master/v4.14831-620-gad64fdfeb
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ZX8211 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================

[*] MAKE client/clean

===================================================================

Version info:      Iceman/master/v4.14831-620-gad64fdfeb
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.15.3 in /usr/lib/aarch64-linux-gnu)
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.10 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
===================================================================
PLATFORM Linux
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean
[*] MAKE mf_nonce_brute/clean
[*] MAKE mfd_aes_brute/clean

===================================================================
Version info:      Iceman/master/v4.14831-620-gad64fdfeb
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ZX8211 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
*] MAKE client/all
[*] MAKE bootrom/all
[*] MAKE fpga_compress/all
[*] MAKE mfkey/all
[*] MAKE nonce2key/all
[*] MAKE mf_nonce_brute/all
[*] MAKE mfd_aes_brute/all
[-] CC nonce2key.c
[-] CC fpga_compress.c
[-] CC mfkey32.c
[-] CC ../../common/lz4/lz4hc.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC ../../common/lz4/lz4.c
[-] CC ../../common/crapto1/crapto1.c
compiler version:  arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
[=] GEN version_pm3.c
[-] CC ../../common/bucketsort.c
[-] CC ../../common/bucketsort.c
[-] CC mfkey32v2.c
[-] CC ram-reset.s
[-] CC mfkey64.c
[-] CC flash-reset.s
[-] CC ../common_arm/usb_cdc.c
[-] CC ../common_arm/clocks.c
[-] CC brute_key.c
[-] CC mf_nonce_brute.c
[-] CC util_posix.c
[-] CC bootrom.c
[-] CC randoms.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC mfd_aes_brute.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC mfd_multi_brute.c
[-] CC ../../common/bucketsort.c
[-] CC iso14443crc.c
[-] CC sleep.c
[-] CC util_posix.c
[-] CC mf_trace_brute.c
===================================================================
-] CC version_pm3.c
Version info:      Iceman/master/v4.14831-620-gad64fdfeb
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.15.3 in /usr/lib/aarch64-linux-gnu)
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.10 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
===================================================================

```

etothex23, 2022-06-06:
sorry about the terrible cut and paste job on the first

iceman1001, 2022-06-06:
GCC 11.2  and you are on a RDV4 model.

iceman1001, 2022-06-06:
on ARM...  hm,  can't test..  But the error message should be different now.
The calloc is larger.

etothex23, 2022-06-06:
i hadn't seen the error before. I have it installd on a few machines but i think this is the first time i put it on arm. I'll check the gcc. Yes the prox3 rdv4

etothex23, 2022-06-06:
Error:
```
src/cmdhfmfu.c: In function ‘CmdHF14AMfuEView’:
src/cmdhfmfu.c:2072:97: error: array subscript ‘mfu_dump_t[0]’ is partly outside array bounds of ‘unsigned char[1076]’ [-Werror=array-bounds]
 2072 |     PrintAndLogEx(INFO, "Max data page... " _YELLOW_("%d") " ( " _YELLOW_("%d") " bytes )", card->pages - 1, card->pages * 4);
      |                                                                                             ~~~~^~~~~~~
src/cmdhfmfu.c:4111:21: note: referencing an object of size 1076 allocated by ‘calloc’
 4111 |     uint8_t *dump = calloc(bytes, sizeof(uint8_t));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdhfmfu.c:2068:84: error: array subscript ‘mfu_dump_t[0]’ is partly outside array bounds of ‘unsigned char[1076]’ [-Werror=array-bounds]
 2068 |         PrintAndLogEx(INFO, "Counter %d... %s", i, sprint_hex(card->counter_tearing[i], 3));
      |                                                               ~~~~~~~~~~~~~~~~~~~~~^~~
src/cmdhfmfu.c:4111:21: note: referencing an object of size 1076 allocated by ‘calloc’
 4111 |     uint8_t *dump = calloc(bytes, sizeof(uint8_t));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

iceman1001, 2022-06-06:
Could be related to GCC and this.. 

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=98503

etothex23, 2022-06-08:
Thanks! I changed compilers:

```
sudo apt -y install gcc-10 g++-10
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-10 10
sudo update-alternatives --config gcc
sudo update-alternatives --config g++
make clean && make -j
```
no errors

iceman1001, 2022-06-08:
cool,
one of these days,  GCC11.2  on ARM will not have false positives when compiling.  
I am closing this one.

etothex23, 2022-06-08:
thanks for the help. I was unaware of the gcc arm issue.