RfidResearchGroup/proxmark3 #1681: pm3: line 246: 90818 Killed                  $CLIENT "$@" with autopwn or hardnested on MF 1K

nico0481, 2022-06-09:
Getting this type of error when trying to use "hf mf autopwn" on a mifare 1K

/usr/local/bin/pm3: line 246: 90818 Killed                  $CLIENT "$@"
pm3 stops and exits
Also happens when trying to use "hf mf hardnested"

-

**Describe the bug**
A clear and concise description of what the bug is.

Setup:
   [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 Kb ( 65% used )

    Client.... Iceman/master/v4.14831-621-g171133803 2022-06-08 22:51:59
    Bootrom... Iceman/master/v4.14831-621-g171133803 2022-06-08 22:51:14 
    OS........ Iceman/master/v4.14831-621-g171133803 2022-06-08 22:51:41 
    Target.... RDV4

*******************
[usb] pm3 --> hf search
 🕕  Searching for ISO14443-A tag...[#] Card didn't answer to CL1 select all
[!] ⚠️  No known/supported 13.56 MHz tags found
[usb] pm3 --> hf search
 🕐  Searching for ISO14443-A tag...          
[+]  UID: 44 96 1D E0 
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Prng detection: hard
[=] 
[=] --- Tag Signature
[=]  IC signature public key name: NXP Mifare Classic MFC1C14_x
[=] IC signature public key value: 044F6D3F294DEA5737F0F46FFEE88A356EED95695DD7E0C27A591E6F6F65962BAF
[=]     Elliptic curve parameters: NID_secp128r1
[=]              TAG IC Signature: 41D67B0225BD9D8454C58EF47BFE4A32273FB5862BF1FBF37740693142002481
[+]        Signature verification: successful
[?] Hint: try `hf mf` commands

**************************************
[usb] pm3 --> hf mf autopwn 
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 42 keys from hardcoded default array
[=] running strategy 1
[=] Chunk 1.1s | found 30/32 keys (42)
[=] running strategy 2
[=] Chunk 0.9s | found 30/32 keys (42)
[+] target sector   0 key type A -- found valid key [ A0A1A2A3A4A5 ] (used for nested / hardnested attack)
[+] target sector   1 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   1 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   3 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   3 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   4 key type A -- found valid key [ A0A1A2A3A4A5 ]
[+] target sector   5 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   5 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   6 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   6 key type B -- found valid key [ FFFFFFFFFFFF ]
.
.
.
.
.
.
.
.

[+] target sector  15 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  15 key type B -- found valid key [ FFFFFFFFFFFF ]
[=] Hardnested attack starting...
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]          |         |                                                         | Expected to brute force
[=]  Time    | #nonces | Activity                                                | #states         | time 
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]        0 |       0 | Start using 4 threads and AVX2 SIMD core                |                 |
[=]        0 |       0 | Brute force benchmark: 351 million (2^28.4) keys/s      | 140737488355328 |    5d
[=]       11 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    5d
/usr/local/bin/pm3: line 246: 90818 Killed                  $CLIENT "$@"

Thanks
Regards

iceman1001, 2022-06-09:
it likes your computer has no RAM left.

Hardnested needs 1.6gb RAM.   What are you running on?

nico0481, 2022-06-09:
Hello, 
Indeed...only 4Gb
Too many running process, too many open tabs...
Reboot and then ok
Feeling really stupid...

Thank

closed 2022-06-09 by nico0481