RfidResearchGroup/proxmark3 #1684: Unable to detect EM 4x50 tags.

T0py, 2022-06-09:
**Describe the bug**
I am no longer able to read a known (and functional) EM 4x50 tag after upgrading from an old firmware. 

**To Reproduce**
Steps to reproduce the behaviour:
Fresh install of MacOS Monterey on a 2015 MBP
1. Download and install xCode 13.4 (extract .xip and move app into applications folder)
2. Download and install Command Line Tools for xCode 13.4
3. Install ‘brew’
`  /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)" `
4. `brew install git readline qt5 pkgconfig coreutils openssl`
5. `brew install recode`
6. `brew install astyle`
7. `brew install RfidResearchGroup/proxmark3/arm-none-eabi-gcc`
8. `git clone https://github.com/RfidResearchGroup/proxmark3.git`
9. `cd proxmark3`
10. `cp Makefile.platform.sample Makefile.platform`
11. Edit Makefile.platform, uncomment the line #PLATFORM_EXTRAS=BTADDON by removing the `#`. Note, I tried compilation with and without this line uncommented.
13. `make clean; make -j`
14. `./pm3-flash-fullimage`

In order to flash the full image, button needs to be held down as device is plugged into PC. 

Once connected, I run the following commands for diagnostics:

```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 38832
[#]   Available memory........ 38832
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD5690C23DF8D402A
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 294400
[... 227 lines cut ...]
[=]  25 | 81 82 82 81 | 81 41 41 81 | user data
[=]  26 | 82 82 82 81 | 81 41 41 41 | user data
[=]  27 | 82 82 81 81 | 81 81 41 41 | user data
[=]  28 | 81 81 80 80 | 01 01 81 81 | user data
[=]  29 | 81 81 81 81 | 81 81 81 81 | user data
[=]  30 | 81 80 80 81 | 81 01 01 81 | user data
[=]  31 | 80 80 81 81 | 81 81 01 01 | user data
[=]  32 | 81 81 82 81 | 81 41 81 81 | device serial number (read only)
[=]  33 | 81 81 81 81 | 81 81 81 81 | device identification (read only)
[=] ----+-------------+-------------+--------------------
[=] Using UID as filename
[+] saved 34 blocks to text file lf-4x50-81818281-dump.eml
[+] saved 136 bytes to binary file lf-4x50-81818281-dump.bin
```
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Compiled on Mac OS Monterey with only the above listed software being installed. Formatted and installed from scratch. 

**Additional context**
I have also compiled, flashed and tried the following firmwares:
Release v4.9237
Release v4.13441
Current release

iceman1001, 2022-06-11:
As a side note,  I notice you haven't followed the installation guide for your RDV4.   

lets see,  on discord you said you tried without the btaddon.  
However now I get the feeling you are using latest repo with the rfid app.   That will not work if you are.
The rfid app needs its own older firmware version.

Please verify running with usb cable against latest source here,  that 4x50 doesn't work.

T0py, 2022-06-11:
I can confirm that all of the tests I have performed so far are via the USB cable against the latest source, Release v4.9237 and Release v4.13441 and  the current release v4.14831.

What part of the installation guide did I not follow?  These are the instructions I followed:

https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md#homebrew-mac-os-x-developer-installation

With regards to the BT addon, that appears to be an issue with me not setting up the client correctly, not the compilation/building of the firmware and mac client.

T0py, 2022-06-11:
Just adding a picture of the EM tag I'm using:
[image]

As well as images of the readers this tag works with (two of which are clearly Indala/ASP readers). 
[image]
[image]
[image]

I can confirm that I've been able to use those readers since this issue has come up.

Also of note, 
1) I repeatedly got the same em4x50 dumps/info output previously. 
2) I am not seeing anything in the data plot to indicate the tag is energizing.
3) The tag isn't being detected as Indala either.

I've included the output of:
```
data plot
lf read
data save -f xxxxxx
```
Here: 
[image]

And this is the saved data
[data-read-1.pm3.zip](https://github.com/RfidResearchGroup/proxmark3/files/8884239/data-read-1.pm3.zip)

I can confirm that the PM is reading T55xx cards reliably
[image]

And I've also tried (just for fun) switching the Q and F values on the antenna and re-running those tests.
In case it's magically at 134kHz or something. No change.
I'm also no longer compiling with the BT addon.
I've got two RDV4's with me. Both are functioning the same way. So I don't believe it's a device/hardware specific issue either.
Nor does it seem to be exclusive to Ubuntu/Mac OS as I've compiled and flashed from both.
I've also tried using different USB cables in case there was something wrong. 

Let me know if there's anything else I can do to test. I'm working on getting a second tag to test with, as well as multiple blank EM4x50 tags.

iceman1001, 2022-06-11:
I see in the guide that the programmer part of the guide doesn't have the same steps in the end.   
[image]

Returning to your issue at hand,  it's odd.  The users @tharexde who wrote the EM4x50 impl tested latest w/o issues.
Looking at your dumps from the 2020 fw,  they look not correct.   EM4x50 functions has improved since,  so it might has been a false positive.  Indala LF does PSK...   What does `lf indala reader`   say?   And your lf read shows no signal at all.  Which would be an indication its a bad coupling (given the small token)  or not LF or it needs a awake command before...

T0py, 2022-06-11:
Oh, okay. Yeah, I didn't know to run `[usb] pm3 --> script run init_rdv4` or to do a 'smart card module upgrade'. I've not done either before.

I had v3.10 of the smart card module. I've now upgraded it to v3.11. 

```
[#] Smart card module (ISO 7816)
[#]   version.................v3.11
```

running `lf indala reader` 
returns `signal looks like noise` 
As a response.

The token is small, but like, i've written/read to smaller implants before.  The reader pictures I posted above are all definitely LF readers. They _do_ support some non-standard frequencies though I believe. But again, I'm not sure how I managed to capture repeatable dumps using the older firmware if it was a false positive? That's what's confusing here to me.

iceman1001, 2022-06-11:
Can you do a `lf sniff`  ?

T0py, 2022-06-20:
I purchased some em4x50 and em410x tags online and these all work fine. 

[image]

So this would suggest that this issue has been a false positive from the start with the older software incorrectly identifying the tag as em 4x50. 

Just an update for anyone who's interested...

I did some more research and managed to get my hands on two other tags from this system.
The small keyfob one provided no new/additional information. But the vehicle transponder...

[image]

[image]

A bit of googling later leads me to this:

https://www.digikey.ca/en/products/detail/texas-instruments/RI-TRP-R9UR-30/1768347#product-details-substitutes

I'll close this issue now as the issue is not with the EM4x50 functionality, but with reading this particular type of tag. I'm still not sure how this tag was detected as EM4x50 repeatedly before and how it now wont even apparently energise (and yeah, I've tried using the 134khz mode as well). But I'll take this to discord for further discussion.

Thanks for your help.

closed 2022-06-20 by T0py