RfidResearchGroup/proxmark3 #1699: hf_mfcsim dump upload bug

jphuangjr, 2022-06-17:
**Describe the bug**
After loading the .bin dump file for mifare 1k classic into the flash memory, when the user starts standalone mode, the wrong UUID is generated. 

**To Reproduce**
Steps to reproduce the behavior:
1. Load .bin dump file. (eg. `mem spiffs upload -s  hf-mf-86C6D462-dump.bin -d hf_mfcsim_dump_01.bin` )
2. Open standalone mode (see pic 1)
3. Emulate and view results on a reader (reader is showing wrong uuid as well)

**Expected behavior**
Expected uuid to be correct, in the example case to be 86C6D462. I ran the emulation mode after loading the bin dump and read it on my reader and it was correct so the file is not the problem.

**Screenshots**
Pic. 1: 
[image]

**Desktop (please complete the following information):**
 - OS: MacOs
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - [ ARM ]
  bootrom: RRG/Iceman/master/v4.14831 2022-01-11 19:17:19 
       os: Iceman/master/v4.14831-660-g977d92320-dirty-unclean 2022-06-15 18:21:50 d18fb406d
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 65% used )

 - hw status
 [#] Memory
[#]   BigBuf_size............. 38812
[#]   Available memory........ 38812
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD5697C3097897E23
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 240640
[#]   Transfer Speed PM3 -> Client... 481280 bytes/s
[#] Various
[#]   Max stack usage......... 4824 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 12
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 33212 Hz
[#] Installed StandAlone Mode
[#]   HF Mifare Classic simulation mode - a.k.a MFCSIM
[#] Flash memory dictionary loaded
[#] 

 - data tune

**Additional context**
Add any other context about the problem here.

iceman1001, 2022-08-16:
That is because the standalone mode is calling the simulation fct with a NULL reference.

I think I have a solution for it and will push it when I got time to verify it.

iceman1001, 2022-08-17:
its fixed.

closed 2022-08-17 by iceman1001