RfidResearchGroup/proxmark3 #1707: Build failure on Termux

swg0101, 2022-07-04:
I was trying to build on the latest master commit but it's failing on Termux/Clang when trying to build the client:

```
[*] MAKE client/all
===================================================================
Version info:      Iceman/master/v4.14831-722-g56eaf3431
Client platform:   Linux
GUI support:       QT not found, disabled
native BT support: Bluez not found, disabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.10 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  clang version 14.0.6
===================================================================
[-] CC src/cmdhf14b.c
[-] CC src/cmdhfemrtd.c
[-] CC src/cmdhffelica.c
[-] CC src/cmdhffido.c
[-] CC src/cmdhfgallagher.c
[-] CC src/cmdhfksx6924.c
[-] CC src/cmdhfcipurse.c
[-] CC src/cmdhficlass.c
[-] CC src/cmdhflegic.c
src/cmdhf14b.c:443:27: error: implicit conversion from enumeration type 'enum ISO14B_COMMAND' to different enumeration type 'iso14b_type_t' (aka 'enum ISO14B_TYPE') [-Werror,-Wenum-conversion]
            *found_type = ISO14B_SELECT_SR;
                        ~ ^~~~~~~~~~~~~~~~
1 error generated.
make[1]: *** [Makefile:909: obj/cmdhf14b.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:158: client/all] Error 2
make: Leaving directory '/data/data/com.termux/files/home/proxmark3'
```

Seems like perhaps a cast might be needed?

merlokk, 2022-07-04:
in macos too(

iceman1001, 2022-07-04:
should be fix with latest commit  https://github.com/RfidResearchGroup/proxmark3/commit/e98675f8e599fdc4c80cb9b7444529cf326c9c1e

closed 2022-07-04 by iceman1001

swg0101, 2022-07-04:
Works now, thanks!