RfidResearchGroup/proxmark3 #1708: Crash when dumping EF_DG2 on Termux

swg0101, 2022-07-04:
I was playing around with the new emrtd features and it looks like it works quite well. The only thing that seems a bit odd was when I tried dumping the contents through Termux it crashes with the following. The client on the PC doesn't seem to do this even though when it's running the same image that's compiled from Android. Any thoughts why this may happen?

```
[usb] pm3 --> hf em dump -n xxxxxxxx d xxxxxx -e xxxxxx
[!!] 🚨 Failed to select 011C
[=] Couldn't dump EF_CardAccess, card does not support PACE
[?] This is expected behavior for cards without PACE, and isn't something to be worried about
[=] Authentication is enforced. Will attempt external authentication.
[=] External authentication with BAC successful.
[=] ..
[=] Read EF_COM, len: 24
[+] saved 24 bytes to binary file ./EF_COM.BIN
[=] ..
[=] Read EF_DG1 , len 93
[+] saved 93 bytes to binary file ./EF_DG1.BIN
[=] .................................
[=] .................................
[=] .................................
[=] .................................
[=] ..............
[=] Read EF_DG2 , len 16550
FORTIFY: vsnprintf: size 18446744073709551614 > SSIZE_MAX
Aborted
```

iceman1001, 2022-07-04:
looks like the 16550 is overflowing a counter...

can you do a  `gdb`  from it?

iceman1001, 2022-07-04:
pull latest,    and see if that fixes this problem?

swg0101, 2022-07-04:
Nope, same issue - will try a gdb

swg0101, 2022-07-04:
Here's the bt:
```
[!!] 🚨 Failed to select 011C
[=] Couldn't dump EF_CardAccess, card does not support PACE
[?] This is expected behavior for cards without PACE, and isn't something to be worried about
[=] Authentication is enforced. Will attempt external authentication.
[=] External authentication with BAC successful.
[=] ..
[=] Read EF_COM, len: 24
[+] saved 24 bytes to binary file ./EF_COM-2.BIN
[=] ..
[=] Read EF_DG1 , len 93
[+] saved 93 bytes to binary file ./EF_DG1-2.BIN
[=] .................................
[=] .................................
[=] .................................
[=] .................................
[=] ..............
[=] Read EF_DG2 , len 16550
FORTIFY: vsnprintf: size 18446744073709551614 > SSIZE_MAX

Thread 1 "proxmark3" received signal SIGABRT, Aborted.
0x0000007ff30a16a8 in abort () from /apex/com.android.runtime/lib64/bionic/libc.so
(gdb) bt
#0  0x0000007ff30a16a8 in abort () from /apex/com.android.runtime/lib64/bionic/libc.so
#1  0x0000007ff30a33e8 in __fortify_fatal(char const*, ...) () from /apex/com.android.runtime/lib64/bionic/libc.so
#2  0x0000007ff30ff150 in snprintf () from /apex/com.android.runtime/lib64/bionic/libc.so
#3  0x0000005555834c3c in sprint_hex_inrow ()
#4  0x0000005555729370 in emrtd_dump_file ()
#5  0x000000555572919c in dumpHF_EMRTD ()
#6  0x0000005555731860 in CmdHFeMRTDDump ()
#7  0x00000055557d8534 in CmdsParse ()
#8  0x00000055557d8534 in CmdsParse ()
#9  0x00000055557d8534 in CmdsParse ()
#10 0x000000555582c44c in main_loop ()
#11 0x000000555582cbdc in main ()
(gdb)
```

iceman1001, 2022-07-04:
It has to be with how much the read file returned, before at line 833

Turn on debug mode also for the client,  and do the same.

`data setdebugmode -2`

swg0101, 2022-07-04:
```
.
[=] Read EF_DG2 , len 16550
[#] Contents (may be incomplete over 2k chars)
[#] ------------------------------------------
FORTIFY: vsnprintf: size 18446744073709551614 > SSIZE_MAX

Thread 1 "proxmark3" received signal SIGABRT, Aborted.
0x0000007ff60e66a8 in abort () from /apex/com.android.runtime/lib64/bionic/libc.so
(gdb) bt
#0  0x0000007ff60e66a8 in abort () from /apex/com.android.runtime/lib64/bionic/libc.so
#1  0x0000007ff60e83e8 in __fortify_fatal(char const*, ...) () from /apex/com.android.runtime/lib64/bionic/libc.so
#2  0x0000007ff6144150 in snprintf () from /apex/com.android.runtime/lib64/bionic/libc.so
#3  0x0000005555834c3c in sprint_hex_inrow ()
#4  0x0000005555729370 in emrtd_dump_file ()
#5  0x000000555572919c in dumpHF_EMRTD ()
#6  0x0000005555731860 in CmdHFeMRTDDump ()
#7  0x00000055557d8534 in CmdsParse ()
#8  0x00000055557d8534 in CmdsParse ()
#9  0x00000055557d8534 in CmdsParse ()
#10 0x000000555582c44c in main_loop ()
#11 0x000000555582cbdc in main ()
(gdb)
```
Looks like line 848

iceman1001, 2022-07-04:
the hex string output buffer is 8193,   and it will limit all your output to that length,  ie  since buffer is 35000, and readlen is 16550,   it would still only print 8193 chars  ...

iceman1001, 2022-07-04:
pull latest,  and see if that helped,
if you on discord it is faster to test and reply

swg0101, 2022-07-04:
Sure, what's the channel?

iceman1001, 2022-07-04:
#proxmark-dev

iceman1001, 2022-07-04:
pull latest and see if that fixed the issue for you?

closed 2022-07-06 by iceman1001