RfidResearchGroup/proxmark3 #1709: Cannot write HID H10301 data to T5577

JC-SoCal, 2022-07-05:
I am trying to write an HID H10301 card to a T5577 card. However, reading the card back, the write is never successful. I have tested this the steps below on two proxmark 3 RDV4s.

**To Reproduce**
Steps to reproduce the behavior:
1. Checked out report 9407be8ea8e1ac6c76a184ab59a0ce13f9d4d461
2. ran ./pm3-flash-bootloader
3. ran ./pm3-flash-all
4. run lf hid clone from cheatsheet: lf hid clone -r 200670012d
5. PM3 says it ran/done
6. ran lf hid reader, no data found
7. copied a card using an X-copy to the t5577 card, data found, pm3 also is able to read data.

**Expected behavior**
Expected the PM3 to write the HID data to the t5577

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]: Linux 5.15.0-kali3-amd64
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune
 [usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-727-g9407be8ea 2022-07-04 22:56:55 9dfd03622
  compiled with............. GCC 11.2.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.14831-727-g9407be8ea 2022-07-04 22:56:34 9dfd03622
       os: Iceman/master/v4.14831-727-g9407be8ea 2022-07-04 22:56:44 9dfd03622
  compiled with GCC 10.3.1 20210621 (release)

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 65% used )

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40904
[#]   Available memory........ 40904
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD567A882A78D7B25
[#] Smart card module (ISO 7816)
[#]   version................. v2.06
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 274432
[#]   Transfer Speed PM3 -> Client... 548864 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 33376 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare.................. 1397 keys
[#]   T55x7................... 124 keys
[#]   iClass.................. 11 keys
[#] 

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40904
[#]   Available memory........ 40904
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD567A882A78D7B25
[#] Smart card module (ISO 7816)
[#]   version................. v2.06
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 274432
[#]   Transfer Speed PM3 -> Client... 548864 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 33376 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare.................. 1397 keys
[#]   T55x7................... 124 keys
[#]   iClass.................. 11 keys
[#] 

**Additional context**
Add any other context about the problem here.

iceman1001, 2022-07-05:
most likely to a bug in the hex_to_buff function in the latest source.   Try some old source,   like 10 commits earlier and see if it solves your issue

JC-SoCal, 2022-07-05:
Thanks,

I did try that including both of these:
87db9e7ec903346ad281aa7489c438a007fda894
0b0cc20dc2caaca386340f572b2b1dfb44d58319

Recompiled both with 
make clean && make -j
sudo make install

Just to be safe, i also reflashed with pm3-flash-all. 

Still does not write.

do you have a commit you can recommend and I'll try it?

closed 2022-07-08 by JC-SoCal

JC-SoCal, 2022-07-08:
More experimentation, I figured it out. To write the card, I had to remove the case, and place the T55 under the antenna just right. if it was off by a few mm, it didn't work.  I'm not sure if this is documented anywhere, but I figured this out by noticing the new antenna that is shipping has a Q switch which takes care of this issue.

iceman1001, 2022-07-08:
... hm,  let me suggest this then.
open the case,  make sure your Q switch is set to 7.
From your initial post, I suggest you follow the guides on how to configure your rdv4 properly.