RfidResearchGroup/proxmark3 #1714: Proxmark seems stuck after `hf emrtd info` on reading passport

ezhevita, 2022-07-08:
**Describe the bug**
Proxmark stops responding to any commands after trying to read Russian international passport using command `hf emrtd info` (and getting timeout during it). Reconnecting (and thus rebooting Proxmark) will bring it back alive.

**To Reproduce**
Steps to reproduce the behavior:
1. Use `hf emrtd info ...` on a Russian international passport

**Expected behavior**
Passport will be successfully read.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 12.4 (21F79) with M1 Pro
 - hw version
```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-763-g11e097c1b 2022-07-08 22:46:20 f50237638
  compiled with............. Clang/LLVM Apple LLVM 13.1.6 (clang-1316.0.21.2.5)
  platform.................. OSX / aarch64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.14831-763-g11e097c1b 2022-07-08 22:46:02 f50237638
       os: Iceman/master/v4.14831-763-g11e097c1b 2022-07-08 22:46:17 f50237638
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ]
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 59% used )
```
 - hw status
```
[#] Memory
[#]   BigBuf_size............. 42492
[#]   Available memory........ 42492
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 293888
[#]   Transfer Speed PM3 -> Client... 587776 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30432 Hz
[#] Installed StandAlone Mode
[#]   HF Mifare sniff/clone - aka MattyRun (Matías A. Ré Medina)
[#]
```
 - data tune
```
[=] ---------- LF Antenna ----------
[+] LF antenna: 30.44 V - 125.00 kHz
[+] LF antenna: 42.45 V - 134.83 kHz
[+] LF optimal: 43.05 V - 133.33 kHz
[+] Approx. Q factor (*): 8.2 by frequency bandwidth measurement
[+] Approx. Q factor (*): 12.5 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 30.94 V - 13.56 MHz
[+] Approx. Q factor (*): 9.0 by peak voltage measurement
[+] HF antenna is OK
```

**Additional context**
Despite being tested on the Chinese copy of Proxmark3 Easy, issue is reproducible on other Easy devices and RDV4. Also NFC chip in the passport I was testing isn't faulty - I tried multiple and they all behave the same.

iceman1001, 2022-07-09:
looks like there is something with russian epassorts and  0x011E  secure file read of EF_COM...

Try enable debug messages,   might get more necessary information out.

```
data setdeb -2
hf emrtd info
```

did you test to call the `hf emrtd info`  with the "-n -d -e"  params set?

ezhevita, 2022-07-09:
Setting `data setdeb -2` does make no difference, output is the same
I was calling with set -n -d -e params, as you can see on the screenshot - without them it will show basic info and stop, since authentication is enforced
Also Proxmark gets stuck even after failed authentication:
[image]

iceman1001, 2022-07-09:
ok, 14B,

Seems to get stuck in the external authentication part where your passport seems to not approve.

And after that your next command fails.  Most likely because of field is still on.

ezhevita, 2022-07-10:
Is there any debug info I can collect to help?

iceman1001, 2022-07-24:
I can confirm that on OSX  the `emrtd` commands get stuck on  `8E08`  apdu,   it  creates it but epassport doesnt answer back.

It fails after detecting  `BrainpoolP384r1`   might be related to the mbedtls library we use. 

I have confirmed that on ubuntu/wsl it work.

realytcracker, 2022-08-30:
client seems to vomit more information on this - `prefs set clientdebug --full`

after the APDU failure, it locks the device, and you need to unplug and replug to get it working again.

doubly-confirmed, same passport works on ubuntu, fails on x86 and m1 macs.

```
[=] ------------------ Basic Info ------------------
[+] Communication standard: ISO/IEC 14443(B)
[+] Authentication........: Enforced
[+] PACE..................: Not available
[+] Authentication result.: Successful
[#] cmd: 0CA4020C80000000
[#] data: 011E800000000000
[#] temp: 8FD4DDB974CD9A2C
[#] do87: 8709018FD4DDB974CD9A2C
[#] m: 0CA4020C800000008709018FD4DDB974CD9A2C
[#] ssc-b: 01D4E9C900000000
[#] ssc-a: 01D4E9C900000001
[#] n: 01D4E9C9000000010CA4020C800000008709018FD4DDB974CD9A2C
[#] cc: 2976C3F480D32E36
[#] do8e: 8E082976C3F480D32E36
[#] lc: 21
[#] data: 8709018FD4DDB974CD9A2C8E082976C3F480D32E36
[+] >>>> 0C A4 02 0C 15 87 09 01 8F D4 DD B9 74 CD 9A 2C 8E 08 29 76 C3 F4 80 D3 2E 36 00
[=] You can cancel this operation by pressing the pm3 button
[!!] 🚨 APDU: reply timeout
[!!] 🚨 Failed to secure select 011E
[!!] 🚨 Failed to read EF_COM.
```

realytcracker, 2022-08-30:
notably, `0C A4 02 0C 15 87 09 01 8F D4 DD B9 74 CD 9A 2C 8E 08 29 76 C3 F4 80 D3 2E 36 00` does not match the correct output.

it should be `0C A4 02 0C` followed by the `do87` value and `do8e` value concatenated, hence the challenge is failing.

iceman1001, 2022-08-30:
it is an odd issue,  working on Ubuntu vs not working on M1...

iceman1001, 2023-01-08:
@realytcracker    I guess you dropped a beat with deadmau5 ?   

I don't think this issue will be resolved until someone starts debugging the mbedtls library calls.    

Close until someone finds more luck?

realytcracker, 2023-01-08:
yes indeed - i lead i very weird existence.if i harvest some time somewhere, i might take a stab at a PR myself to end these shenanigans. feel free to close in the meantime.hacky new year and i hope you are well!

anders0l, 2023-01-09:
same issue for me - stuck on mac os with m1 max and fully work on Ubuntu in UTM virtualisation 

please keep this issue open as maybe someone will dive into

piotrva, 2023-12-24:
Observed even more drastic behavior with Polish passport - both on Linux and Windows machines Proxmark3 hardware reboots after a few seconds following read command...

iceman1001, 2023-12-24:
There is a memory issue,  where several 14B apdu calls eats up the bigbuff and eventually into the stack and the device hangs.

anders0l, 2023-12-24:
> There is a memory issue, where several 14B apdu calls eats up the bigbuff and eventually into the stack and the device hangs.

On my mac, I've got 64gb unified memory and 400GB/s of memory bandwidth, and it's definitely not memory issue. Except there is some configuration memory limit in proxmark itself

piotrva, 2023-12-24:
@anders0l - I think this memory is related to internal memory in Proxmark MCU...

iceman1001, 2023-12-24:
I was unclear.    it is device side related.

piotrva, 2024-01-03:
Do RDV4.01 have bigger internal RAM than others?

iceman1001, 2024-01-03:
no,   it has 64k ram like most MCU from that product line

piotrva, 2024-01-19:
Tested new firmware (Iceman/master/v4.17768-170-gf48d49556). Now with Polish Passport I get:
```
[usb] pm3 --> hf em info -n AB1234567 -d 123456 -e 123456
[=] ..
[=] Authentication is enforced
[=] Switching to external authentication...
[!!] APDU: no APDU response
[!!] Couldn't do external authentication. Did you supply the correct MRZ info?

[=] ------------------ Basic Info ------------------
[+] Communication standard: ISO/IEC 14443(B)
[+] Authentication........: Enforced
[+] PACE..................: Available
[+] Authentication result.: Failed

[=] ----------------- EF_CardAccess ----------------
[+] PACE version..........: 2
[+] PACE algorithm........: ECDH, Generic Mapping, 3DES-CBC-CBC
[+] PACE parameter........: NIST P-256 (secp256r1)
```
So good news is the device is not rebooting.
Bad news - still not reading the data ;)

iceman1001, 2024-01-20:
if you add 
```
data setde -2
hf emrtd info -n AB1234567 -d 123456 -e 123456
trace save -f polish_pp_nAB1234567_d123456_e123456
hf emrtd list
data setde -0
```

One can look and see what is going wrong in the trace if any  and you should have gotten a longer more detailed output

piotrva, 2024-01-20:
Well, it might expose some sensitive data, as this is my actual valid passport...

iceman1001, 2024-01-20:
In that case you are on your own,  I can't not help out. 
But look at the output and see where it stops

iceman1001, 2024-01-23:
Try pulling  latest and flash,      need to test if it is fixed.

piotrva, 2024-01-23:
Hi, again same result
```
[usb] pm3 --> hf em info -n AB1234567 -d 123456 -e 123456
[=] ..
[=] Authentication is enforced
[=] Switching to external authentication...
[!!] APDU: no APDU response
[!!] Couldn't do external authentication. Did you supply the correct MRZ info?

[=] ------------------ Basic Info ------------------
[+] Communication standard: ISO/IEC 14443(B)
[+] Authentication........: Enforced
[+] PACE..................: Available
[+] Authentication result.: Failed

[=] ----------------- EF_CardAccess ----------------
[+] PACE version..........: 2
[+] PACE algorithm........: ECDH, Generic Mapping, 3DES-CBC-CBC
[+] PACE parameter........: NIST P-256 (secp256r1)
```

iceman1001, 2024-01-23:
Well,  either your passport doesn't support BAC,  
```
[!!] Couldn't do external authentication. Did you supply the correct MRZ info?
```
and we don't support PACE.   

The question is first,  do your pm3 client hang?    
the second is that your password is most likely PACE...

piotrva, 2024-01-23:
No, same as I mentioned before - now proxmark does not hang - that is the good news.

Is there any literature I can probably read about the BAC and PACE?

iceman1001, 2024-01-24:
use your google-foo to find about machine readable documents.     Should be public information

iceman1001, 2024-01-24:
Closing ,  since original issue is solved.

closed 2024-01-24 by iceman1001

ezhevita, 2024-02-12:
@iceman1001 I can still reproduce this issue using the latest pm3 client and firmware

iceman1001, 2024-02-12:
Still don't have anything to go with here,   until you decide to add proper instructions on how to replicate the issue. 

And no,  "it doesn't work on my passport"  isn't going to help debugging it.   Nor have I any russian passports to play with .

In the end,  its down to you to solve this issue.   We can't.

ezhevita, 2024-02-14:
I’ve tried debugging this issue, the firmware itself is alive — i’ve added debug output in the AppMain method in the main loop (that receives new packets from usb) and it goes through without any problems, so it looks like after authentication proxmark just doesn’t receive any packets

also i’ve tried capturing usb interface on macOS using wireshark and i still can see the last sent packet to proxmark there (despite it being non-existent for proxmark)

marcellp, 2024-03-02:
I can reproduce this on an Intel Mac running 14.0 Sonoma.

I tested this with three separate eMRTDs issued by different countries. Seems like the issue is that some of these newer documents (from what I can see from the countries impacted, those manufactured by Gemalto/Thales, in particular), and perhaps related to the type of elliptic curve being used for ECDH, as it seems all reports are so far for the `secp256r1` curve. 

**Document 1:** Passport issued by Hungary in 2019. PACE-enabled, with `BrainpoolP256r1` curve, no CAN number. Works perfectly fine.

```
[usb] pm3 --> hf emrtd info -n xyz -d xyz -e xyz
[=] ..
[=] Authentication is enforced
[=] Switching to external authentication...
[=] External authentication with BAC successful.

[=] ------------------ Basic Info ------------------
[+] Communication standard: ISO/IEC 14443(A)
[+] Authentication........: Enforced
[+] PACE..................: Available
[+] Authentication result.: Successful

[=] ----------------- EF_CardAccess ----------------
[+] PACE version..........: 2
[+] PACE algorithm........: ECDH, Generic Mapping, AES-CMAC-128
[+] PACE parameter........: BrainpoolP256r1
[=] ..

[=] -------------------- EF_COM --------------------
[+] EF_DG1 ...............: Details recorded in MRZ
[+] EF_DG2 ...............: Encoded Face
[+] EF_DG3 ...............: Encoded Finger(s)
[+] EF_DG7 ...............: Displayed Signature or Usual Mark
[+] EF_DG11...............: Additional Personal Detail(s)
[+] EF_DG12...............: Additional Document Detail(s)
[+] EF_DG14...............: Security Options
[=] ...................
[=] ..

[=] -------------------- EF_DG1 --------------------
[+] Document Type.........: Passport
[+] Document Form Factor..: TD3
[+] Issuing state.........: HUN
[+] Nationality...........: HUN
[.. omitted ..]
```

**Document 2:** Passport issued by United Kingdom in 2024. PACE-enabled with `secp256r1` curve, no CAN number. Crashes the device.

```
[usb] pm3 --> hf emrtd info -m xyz
[=] ..
[=] Authentication is enforced
[=] Switching to external authentication...
[=] External authentication with BAC successful.

[=] ------------------ Basic Info ------------------
[+] Communication standard: ISO/IEC 14443(B)
[+] Authentication........: Enforced
[+] PACE..................: Available
[+] Authentication result.: Successful

[=] ----------------- EF_CardAccess ----------------
[+] PACE version..........: 2
[+] PACE algorithm........: ECDH, Generic Mapping, AES-CMAC-256
[+] PACE parameter........: NIST P-256 (secp256r1)
[=] You can cancel this operation by pressing the pm3 button
[!!] 🚨 APDU: reply timeout
[!!] 🚨 Failed to secure select 011E
[!!] 🚨 Failed to read EF_COM.
```

**Document 3:** Passport issued by Poland in 2014. No PACE enabled. Works fine.

```
[usb] pm3 --> hf emrtd info -n xyz -d xyz -e xyz
[!!] 🚨 Failed to select 011C
[?] The error above this is normal. It just means that your eMRTD lacks PACE.
[=] Authentication is enforced
[=] Switching to external authentication...
[=] External authentication with BAC successful.

[=] ------------------ Basic Info ------------------
[+] Communication standard: ISO/IEC 14443(A)
[+] Authentication........: Enforced
[+] PACE..................: Not available
[+] Authentication result.: Successful
[=] ..

[=] -------------------- EF_COM --------------------
[+] EF_DG1 ...............: Details recorded in MRZ
[+] EF_DG2 ...............: Encoded Face
[+] EF_DG3 ...............: Encoded Finger(s)
[+] EF_DG14...............: Security Options
[=] ..................
[=] ..

[=] -------------------- EF_DG1 --------------------
[+] Document Type.........: Passport
[+] Document Form Factor..: TD3
[+] Issuing state.........: POL
[+] Nationality...........: POL
[.. omitted ..]
```

marcellp, 2024-03-02:
On further inspection, looks like the main differentiating factor between those that fail and those that don't is the type of NFC algorithm being used.

For the working passports, `Iso7816ExchangeEx`'s `GetISODEPState()` returns `ISODEP_NFCA` and we are sending messages through `ExchangeAPDU14a()`.

For the failing passports, the same is `ISODEP_NFCB` and therefore we are sending stuff through `exchange_14b_apdu()`. I wonder if something is broken in the implementation of that one.