RfidResearchGroup/proxmark3 #1716: texkom chips with another modulation scheme

horrordash, 2022-07-17:
I found texkom chips that use another modulation scheme.
The principle is similar to that used in touchmemory.
In all other respects, the chip corresponds to tk13 (crc, bit order, etc.)
The tag has a poor (but recognizable) signal due to the small antenna size (~12mm)
[image]

merlokk, 2022-07-17:
strange...  it should read it.
Try please move the badge from the antenna - maybe these "double" peaks will be eliminated...
and run the command: `hf texkom read -v`
it should be like this:
[image]

and give me please .pm3 file with several best readers (`data save -f mydatafile`)

horrordash, 2022-07-17:
it's really a different modulation scheme (PWM). your example uses manchester coding.
short spikes do not affect the operation of the algorithm. 
your correlation function is tolerant to spikes.

I slightly modified your program and made a simple decoder with hard-coded thresholds
and here is decoding results:

middle-size-tag.pm3
```
[!] --- indx: 22712, len: 2284, max: 111, noise: 22
[!] --- impulses: 128, lenarray: 127, [20,16]
[=] Length array:
20, 16, 22, 13, 21, 15, 21, 15, 20, 18, 18, 15, 21, 15, 21, 17, 21, 15, 21, 15, 22, 14, 20, 15, 21, 15, 21, 15, 20, 16, 20, 18, 9, 26, 20, 1
5, 21, 15, 9, 26, 9, 26, 9, 28, 18, 18, 18, 18, 20, 16, 9, 26, 9, 26, 9, 26, 9, 25, 9, 26, 9, 26, 9, 28, 21, 17, 21, 13, 20, 16, 20, 16, 9,
25, 10, 25, 21, 15, 20, 18, 9, 26, 9, 26, 20, 18, 9, 26, 9, 23, 21, 15, 9, 26, 21, 17, 9, 26, 9, 26, 20, 16, 9, 26, 11, 24, 20, 16, 9, 28, 7
, 28, 9, 26, 9, 25, 21, 15, 9, 28, 19, 15, 9, 26, 9, 26, 9,
bits: 1111111111111111011000111000000011110011001001010010010000101000
[!] --- indx: 25435, len: 2284, max: 107, noise: 21
[!] --- impulses: 128, lenarray: 127, [20,15]
[=] Length array:
20, 15, 21, 15, 21, 15, 20, 16, 20, 15, 21, 15, 21, 15, 20, 18, 20, 18, 18, 16, 20, 15, 23, 13, 21, 15, 20, 16, 20, 15, 21, 18, 9, 25, 21, 1
5, 21, 15, 9, 26, 9, 26, 11, 24, 20, 15, 21, 17, 21, 15, 9, 28, 7, 26, 9, 26, 9, 26, 9, 26, 9, 26, 9, 28, 20, 18, 18, 16, 20, 15, 21, 15, 11
, 24, 9, 26, 22, 14, 20, 18, 9, 28, 7, 26, 20, 18, 7, 25, 9, 26, 21, 15, 9, 26, 20, 18, 9, 26, 9, 26, 20, 16, 9, 26, 9, 25, 21, 15, 9, 26, 9
, 28, 9, 26, 9, 26, 21, 15, 9, 26, 20, 16, 9, 26, 9, 26, 9,
bits: 1111111111111111011000111000000011110011001001010010010000101000
[=] Texkom: FF FF 63 80 F3 25 24 28
[=] uid       : 80 F3 25 24
[=] crc       : OK
```

small-size-tag.pm3
```
[!] --- indx: 22864, len: 2277, max: 97, noise: 19
[!] --- impulses: 128, lenarray: 127, [19,17]
[=] Length array:
19, 17, 18, 16, 22, 13, 22, 16, 18, 15, 23, 15, 18, 15, 23, 17, 21, 15, 18, 15, 21, 15, 20, 17, 19, 15, 20, 16, 22, 13, 20, 18, 9, 26, 22, 1
3, 22, 16, 7, 26, 9, 27, 7, 28, 18, 17, 21, 17, 21, 13, 9, 28, 7, 25, 9, 26, 9, 28, 7, 27, 9, 26, 7, 28, 20, 16, 20, 15, 22, 16, 18, 17, 9,
26, 7, 26, 20, 15, 20, 18, 9, 28, 18, 17, 9, 24, 20, 16, 20, 15, 20, 16, 22, 13, 9, 28, 20, 16, 20, 15, 21, 17, 18, 16, 11, 25, 7, 26, 11, 2
5, 19, 17, 23, 13, 20, 15, 23, 13, 22, 14, 20, 17, 9, 24, 9, 28, 7,
bits: 1111111111111111011000111000000011110011010111101111000111111000
[!] --- indx: 25580, len: 2277, max: 96, noise: 19
[!] --- impulses: 128, lenarray: 127, [20,14]
[=] Length array:
20, 14, 20, 17, 18, 16, 20, 15, 22, 16, 18, 15, 21, 17, 18, 18, 20, 15, 21, 15, 20, 17, 21, 15, 20, 15, 21, 13, 20, 16, 22, 16, 9, 25, 20, 1
6, 20, 15, 9, 26, 9, 26, 11, 23, 21, 15, 20, 20, 18, 15, 11, 24, 9, 26, 11, 24, 11, 23, 9, 26, 11, 24, 9, 29, 19, 15, 22, 14, 20, 17, 18, 16
, 9, 26, 9, 27, 18, 16, 20, 20, 7, 27, 19, 15, 11, 25, 19, 15, 20, 16, 20, 15, 20, 16, 9, 30, 18, 17, 19, 15, 20, 15, 21, 15, 9, 28, 7, 25,
11, 24, 20, 18, 20, 17, 21, 13, 20, 16, 20, 15, 20, 16, 9, 26, 10, 24, 9,
bits: 1111111111111111011000111000000011110011010111101111000111111000
[=] Texkom: FF FF 63 80 F3 5E F1 F8
[=] uid       : 80 F3 5E F1
[=] crc       : OK
```
[signals.zip](https://github.com/RfidResearchGroup/proxmark3/files/9128517/signals.zip)

merlokk, 2022-07-17:
uncomment please this 2 lines - lets look
```
    //PrintAndLogEx(WARNING, "--- hi: %d, low: %d, threshold: %d", hilength, lowlength, threshold);

    bool biterror = false;
    for (uint32_t i = 0; i < implengthslen; i++) {
        if (TexcomCalculateBit(implengths[i], hilength, threshold))
            strcat(bitstring, "1");
        else if (TexcomCalculateBit(implengths[i], lowlength, threshold))
            strcat(bitstring, "0");
        else {
            //PrintAndLogEx(INFO, "ERROR string [%zu]: %s, bit: %d, blen: %d", strlen(bitstring), bitstring, i, implengths[i]);
         ```

and looks like you make something like `data hpf` to pm3....   give me please raw (the zero of the data should be at the bottom of the graph buffer)

merlokk, 2022-07-17:
looks like it too close to each other and bit length calculator dont works

maybe increase the threshold here....
`uint32_t threshold = (hilength - lowlength) / 3 + 1;`

merlokk, 2022-07-18:
BTW it looks it TK-15. maybe this key has some signs on it?
it like this?
https://ikey.ru/38366-tm15m/

merlokk, 2022-07-18:
check please:
#1720

horrordash, 2022-07-22:
All works fine!
```
[usb] pm3 --> hf texkom read
[=] Texkom: FF FF 63 80 F3 25 24 28
[=] Texkom duplicator: 28 24 25 F3 80 63 FF FF
[=] modulation: TK15
[=] type      : TK15
[=] uid       : 80 F3 25 24
[=] crc       : OK
[usb] pm3 --> hf texkom read
[=] Texkom: FF FF 63 80 F3 81 02 74
[=] Texkom duplicator: 74 02 81 F3 80 63 FF FF
[=] modulation: TK15
[=] type      : TK15
[=] uid       : 80 F3 81 02
[=] crc       : OK
```
I have tags in the form of a naked PCB with a SOT23-6 chip. the marking on the chip is erased so no markings at all.
btw, another interesting tag fell into my hands. this is also some kind of read-only 13.56 tag but has "TAGSYS" marking, so I think it has nothing to do with texkom. can we continue the discussion here or is there a special forum?

iceman1001, 2022-07-22:
Time to close

closed 2022-07-22 by iceman1001

merlokk, 2022-07-22:
> All works fine!
cool
> I have tags in the form of a naked PCB with a SOT23-6 chip. the marking on the chip is erased so no markings at all. btw, another interesting tag fell into my hands. this is also some kind of [...]
discord