RfidResearchGroup/proxmark3 #1722: hf 14a apdufind

atkfromabove, 2022-07-19:
**Describe the bug**
When I go to run the apdufind command in the "hf 14a" section I get the following error and the app exits.

[usb] pm3 --> hf 14a apdufind
./pm3: line 246: 17566 Bus error: 10           $CLIENT "$@"

iceman1001, 2022-07-19:
On latest source?  compiled and flashed for your particular pm3 device?
which os?..   This issue is missing quite much context.   Fill in the bug issue template would been helpful

atkfromabove, 2022-07-20:
***Things to try before submitting bug report***
read the [troubleshooting guide](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to Iceman repo.

-

**Describe the bug**
A clear and concise description of what the bug is.
[usb] pm3 --> hf 14a apdufind
./pm3: line 246: 17566 Bus error: 10 CLIENT"@"

**To Reproduce**
Steps to reproduce the behavior:
1. Run pm3 application start
2. Run hf 14a apdufind
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect the apdufind to start going thorugh and checking for any apdu responses

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: mac OS Monterey 12.4
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 [usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-706-g45ea60c92 2022-07-01 12:57:09 c48a213a1
  compiled with............. Clang/LLVM Apple LLVM 13.0.0 (clang-1300.0.27.3)
  platform.................. OSX / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.14831-802-g6d950e1e4 2022-07-19 13:04:00 7a8ce3a0b
       os: Iceman/master/v4.14831-802-g6d950e1e4 2022-07-19 13:04:17 7a8ce3a0b
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 65% used )
  
 - hw status
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40904
[#]   Available memory........ 40904
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD567A882A79DCC25
[#] Smart card module (ISO 7816)
[#]   version................. v3.11
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 307200
[#]   Transfer Speed PM3 -> Client... 614400 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31840 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare.................. 1397 keys
[#]   T55x7................... 124 keys
[#]   iClass.................. 11 keys
[#] 

 - data tune
[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 66.91 V - 125.00 kHz
[+] LF antenna: 32.40 V - 134.83 kHz
[+] LF optimal: 66.91 V - 125.00 kHz
[+] Approx. Q factor (*): 11.5 by frequency bandwidth measurement
[+] Approx. Q factor (*): 11.7 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 44.48 V - 13.56 MHz
[+] Approx. Q factor (*): 7.8 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

**Additional context**
Add any other context about the problem here.

iceman1001, 2022-07-20:
Bus error,  ok,  you are on RDV4,   on OSX,   

Now,  I tested with my device,  I don't get a crash when running with or w/o card on antenna,  on WSL1 / Ubuntu.

Your client is  706 while your firmware seem to be 802,   are you using the same client / fw version from same compilation?

atkfromabove, 2022-07-20:
Just updated the client to 802 and I get the same issue.

[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-802-g6d950e1e4 2022-07-20 12:42:52 7a8ce3a0b
  compiled with............. Clang/LLVM Apple LLVM 13.1.6 (clang-1316.0.21.2.5)
  platform.................. OSX / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.14831-802-g6d950e1e4 2022-07-19 13:04:00 7a8ce3a0b
       os: Iceman/master/v4.14831-802-g6d950e1e4 2022-07-19 13:04:17 7a8ce3a0b
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 65% used )

[usb] pm3 --> hf 14a apdufind
./pm3: line 246: 11851 Bus error: 10           $CLIENT "$@"

iceman1001, 2022-07-20:
Which card are you trying to run against?

and can you do a `gdb`  output?

iceman1001, 2022-07-21:
gdb is a linux debugger. 

```
iceman@TAU:~/prox2$ gdb -v
GNU gdb (Ubuntu 10.2-0ubuntu1~20.04~1) 10.2
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
iceman@TAU:~/prox2$ gdb -h
This is the GNU debugger.  Usage:

    gdb [options] [executable-file [core-file or process-id]]
    gdb [options] --args executable-file [inferior-arguments ...]

```

atkfromabove, 2022-07-21:
[image]

iceman1001, 2022-07-21:
yes and now you need to debug it...

atkfromabove, 2022-07-21:
Sorry I am not familiar with debugging. I'll look up info about how to work with gdb.

iceman1001, 2022-07-21:
```
# start debugger,  adapt to your serial port
gdb --args /client/proxmark3 /dev/ttyS3

then inside type
run

now the pm3 client starts like normal,  
run the  "hf 14a apdufind"

it will break,

type  BT 

paste the output here.
```

iceman1001, 2022-07-22:
ok,
I have verified the client crashes on OSXl.
It seems to be cliparser, when handling hex2strings.

iceman1001, 2022-07-24:
Try pulling latest,  I believe it fixes it.

commit 51408c89 (2022-07-24): fix #1722,   this should fix the breaking on OSX for apdufind.  Seems u64 triggers something in memory. Passing ref to array and makeing array u32 instead
closed 2022-07-24 by iceman1001