RfidResearchGroup/proxmark3 #1734: Proxmark3 RDV4 stuck at hardnested

przytular, 2022-07-28:
Hello, I have problem with my brand new Proxmark3 RDV4 and pm3 client

**Describe the bug**
After running `hf mf autopwn` command proxmark always stuck on the same lines on hardnested attack:

```
[=]     5073 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[!!] 🚨 Error: No response from Proxmark3
```
(always at 5073)

Full command log:
```
[usb] pm3 --> hf mf autopwn 
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 42 keys from hardcoded default array
[=] running strategy 1
[=] Chunk 0,8s | found 30/32 keys (42)
[=] running strategy 2
[=] Chunk 0,7s | found 30/32 keys (42)
[+] target sector   0 key type B -- found valid key [ FFFFFFFFFFFF ] (used for nested / hardnested attack)
[+] target sector   1 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   1 key type B -- found valid key [ FFFFFFFFFFFF ]
...
[+] target sector  15 key type B -- found valid key [ FFFFFFFFFFFF ]
[-] ⛔ Tag isn't vulnerable to Nested Attack (PRNG is probably not predictable).
[-] ⛔ Nested attack failed --> try hardnested
[=] Hardnested attack starting...
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]          |         |                                                         | Expected to brute force
[=]  Time    | #nonces | Activity                                                | #states         | time 
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]        0 |       0 | Start using 16 threads and AVX2 SIMD core               |                 |
[=]        0 |       0 | Brute force benchmark: 2783 million (2^31,4) keys/s     | 140737488355328 |   14h
[=]        4 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |   14h
...
[=]     5070 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     5071 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     5072 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     5073 |       1 | Apply bit flip properties                               | 140737488355328 |   14h

[!!] 🚨 Error: No response from Proxmark3
``` 
After that all leds are on (constant light, no blinking indicating computing)

**To Reproduce**
Steps to reproduce the behavior:
1. Run `hf mf autopwn` command
2. See error after about 1 hour / at 5073 "time"

**Expected behavior**
Found hidden mifare keys

**Desktop (please complete the following information):**
 - OS: Parrot OS 5.0
 - hw version
```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-667-gb9b6f06bb 2022-06-25 01:02:05 f0adc4a2c
  compiled with............. GCC 10.2.1 20210110
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.14831-667-gb9b6f06bb 2022-06-25 01:01:47 f0adc4a2c
       os: Iceman/master/v4.14831-667-gb9b6f06bb 2022-06-25 01:01:55 f0adc4a2c
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 64% used )
```
 - hw status
```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40920
[#]   Available memory........ 40920
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD5697C309730BC25
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 283648
[#]   Transfer Speed PM3 -> Client... 567296 bytes/s
[#] Various
[#]   Max stack usage......... 4104 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 29770 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#] 
[usb] pm3 --> 
```
 - data tune
```
[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 32,76 V - 125,00 kHz
[+] LF antenna: 38,26 V - 134,83 kHz
[+] LF optimal: 39,15 V - 131,87 kHz
[+] Approx. Q factor (*): 6,0 by frequency bandwidth measurement
[+] Approx. Q factor (*): 6,8 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 47,23 V - 13.56 MHz
[+] Approx. Q factor (*): 8,2 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134,83 kHz, 95 (red) is 125,00 kHz.
```

iceman1001, 2022-07-28:
Doesn't really sound like a source code problem,   sounds more like a user problem,
We have discord and PM3 Forum to ask questions at.

What is the output for `hf 14a info`  ?  
and have you tried some distance between tag and antenna (1-2cm for 14A styled cards)

and I see you haven't followed the setup guides,   which I suggest you do.

przytular, 2022-07-28:
@iceman1001 thanks for reply,

output of `hf 14a info`:
```
[usb] pm3 --> hf 14a info

[+]  UID: E3 CA 13 5D 
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K
[=] proprietary non iso14443-4 card found, RATS not supported
[#] Card didn't answer to CL1 select all
[#] Card didn't answer to CL1 select all
[#] Auth error
[?] Hint: try `hf mf` commands
```
I was using that setup guide: https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md is there anything else I should follow?

iceman1001, 2022-07-28:
Try some distance. 

[image]

atkfromabove, 2022-08-08:
I am having the exact same issue on both OSX and Kali Linux

[image]
[image]

atkfromabove, 2022-08-08:
I ran the test 3 times and it exits at around 5074 and 5075.

przytular, 2022-08-10:
hey @iceman1001 I've tried with some distance, but unfortunately result is exactly the same (stops at 5075).

I found somewhere on web (i can't find link right now) that this may be related to memory management on hardware itself (and as @atkfromabove had same problem on Kali and MacOS it sounds it isn't user issue).

I followed all setup guides on fresh parrot install and result is the same

iceman1001, 2022-08-15:
With 1-2cm distance you shouldn't get the "cant select card"  message any more when running.

@przytular   you have a  `hf 14a info`  output from your card?

przytular, 2022-08-15:
hey @iceman1001 i pasted output in that comment: https://github.com/RfidResearchGroup/proxmark3/issues/1734#issuecomment-1197806384
my problem isn't error "can't select card", but device frozen with error
```
[!!] 🚨 Error: No response from Proxmark3
```

d4g, 2022-08-15:
Just as a question: are you using the blueshark addon? I got the same error when the blueshark battery was low on power. Then running sth stressful would lead to a reset of either the Bluetooth connection or the proxmark.

przytular, 2022-08-16:
@d4g thanks for input, negative, I'm using just pure RDV4 unit, without bluetooth module

iceman1001, 2022-08-17:
so lets eliminate things.
compile and flash device for without btaddon.
run autopwn (use usb cable)

iceman1001, 2022-08-17:
it says you are running parrot os.
on which metal are you running?

iceman1001, 2022-08-17:
and how much RAM do you have on your host OS?

przytular, 2022-08-18:
hey @iceman1001 thanks for reply

> it says you are running parrot os. on which metal are you running?
it's laptop system76 gazelle (https://tech-docs.system76.com/models/gaze15/README.html)
cause i hope by metal you mean hardware?
would you like me to execute specific command, ie. `lshw` ?

> and how much RAM do you have on your host OS?
32 GB

> compile and flash device for without btaddon.
ok, i'll need some time to try that

iceman1001, 2022-08-18:
so you have enough RAM,
you are not on M1/ or a ARM based cpu,
you use 1-2cm of distance between tag and antenna,

I dunno,   I ran out of ideas.

atkfromabove, 2022-08-18:
It's odd when the error occurs since the proxmark is no longer responding. I have to unplug and replug the proxmark for it to start responding again. If I try to continue with other commands they no longer work correctly until I physically cycle the device and restart the pm3 application.

It's tough to test any changes since the program takes so long before the error occurs (~1.5 hours). 

I've re-compiled the source and flashed the firmware without the BTaddon but the error still occurs at the same point (5075).
[image]

iceman1001, 2022-08-18:
This is where I see an issue.
Somehow the hardnested command can't collect any new nonces.  You only get one...

[image]

Can you try running it with the `slow` flag?

iceman1001, 2022-08-18:
if you pull latest,   I pushed a minor fix for the slow param in autopwn command.

`hf mf autopwn --slow --1k  -f mfc_default_keys` 

and do you have a picture of that card?

atkfromabove, 2022-08-19:
> if you pull latest, I pushed a minor fix for the slow param in autopwn command.
> 
> `hf mf autopwn --slow --1k -f mfc_default_keys`
> 
> and do you have a picture of that card?

Pulled the git and ran that command. The error happened even earlier this time at 1535.

`[usb] pm3 --> hf mf autopwn --slow --1k -f mfc_default_keys
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 1465 keys from dictionary file /Users/taylor/Documents/Scripts/proxmark3/client/dictionaries/mfc_default_keys.dic
[=] running strategy 1
[=] ......
[=] Chunk 13.3s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 1.0s | found 2/32 keys (85)
[=] Chunk 0.4s | found 2/32 keys (20)
[=] running strategy 2
[=] ......
[=] Chunk 13.2s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] ......
[=] Chunk 13.0s | found 2/32 keys (85)
[=] .
[=] Chunk 3.2s | found 2/32 keys (20)
[+] target sector   0 key type A -- found valid key [ FFFFFFFFFFFF ] (used for nested / hardnested attack)
[+] target sector   0 key type B -- found valid key [ FFFFFFFFFFFF ]
[-] ⛔ Tag isn't vulnerable to Nested Attack (PRNG is probably not predictable).
[-] ⛔ Nested attack failed --> try hardnested
[=] Hardnested attack starting...
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]          |         |                                                         | Expected to brute force
[=]  Time    | #nonces | Activity                                                | #states         | time 
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]        0 |       0 | Start using 16 threads and AVX2 SIMD core               |                 |
[=]        0 |       0 | Brute force benchmark: 2796 million (2^31.4) keys/s     | 140737488355328 |   14h
[=]        4 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |   14h
[=]        7 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]        8 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]        9 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
...
[=]     1529 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     1530 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     1531 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     1531 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     1532 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     1533 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     1534 |       1 | Apply bit flip properties                               | 140737488355328 |   14h
[=]     1535 |       1 | Apply bit flip properties                               | 140737488355328 |   14h

[!!] 🚨 Error: No response from Proxmark3`

Here is a picture of the card. It was a small wristband for a hotel.
[image]

iceman1001, 2022-08-19:
So I am curious of the output from running these following commands.   I wanna see if the nonce changes,  so run it all and copy pasta the output here.

```
hf mf rdbl --blk 0 -b -k ffffffffffff
hf mf list

hf mf rdbl --blk 0 -b -k ffffffffffff
hf mf list
```

atkfromabove, 2022-08-19:
> So I am curious of the output from running these following commands. I wanna see if the nonce changes, so run it all and copy pasta the output here.
> 
> ```
> hf mf rdbl --blk 0 -b -k ffffffffffff
> hf mf list
> 
> hf mf rdbl --blk 0 -b -k ffffffffffff
> hf mf list
> ```

```
[usb] pm3 --> hf mf rdbl --blk 0 -b -k ffffffffffff

[=]   # | sector 00 / 0x00                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   0 | 73 84 18 C2 2D 08 04 00 03 4D AC 2E 3A 00 46 90 | s...-....M..:.F.

[usb] pm3 --> hf mf list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 188 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
       2116 |       4484 | Tag |04  00                                                                   |     | 
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10564 |      16388 | Tag |73  84  18  c2  2d                                                       |     | 
     112128 |     122656 | Rdr |93  70  73  84  18  c2  2d  8b  ce                                       |  ok | SELECT_UID
     123716 |     127236 | Tag |08  b6  dd                                                               |     | 
     129536 |     134304 | Rdr |61  00  2d  62                                                           |  ok | AUTH-B(0)
     135748 |     140420 | Tag |9e  7e  1e  dd                                                           |     | AUTH: nt 
     150016 |     159392 | Rdr |a3! 1c  d4  fe  90  45! 37! fe                                           |     | AUTH: nr ar (enc)
     160452 |     165188 | Tag |16! bf! 91! 70                                                           |     | AUTH: at (enc)
     171136 |     175904 | Rdr |aa  c8! f4! 8f                                                           |     | 
            |            |  *  |                                              key FFFFFFFFFFFF prng WEAK |     |
            |            |  *  |30  00  02  A8                                                           |  ok | READBLOCK(0)
     176964 |     197828 | Tag |1b  cd  0d! a2! 94! 82! c8  2c  08  24  ba  1f  5b! 66  f8! 27! 21! 87!  |     | 
            |            |  *  |73  84  18  C2  2D  08  04  00  03  4D  AC  2E  3A  00  46  90  E0  47   |  ok | 
     211072 |     215776 | Rdr |63! c5  01  7d                                                           |     | 
            |            |  *  |50  00  57  CD                                                           |  ok | HALT
[usb] pm3 --> hf mf rdbl --blk 0 -b -k ffffffffffff

[=]   # | sector 00 / 0x00                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   0 | 73 84 18 C2 2D 08 04 00 03 4D AC 2E 3A 00 46 90 | s...-....M..:.F.

[usb] pm3 --> hf mf list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 188 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
       2116 |       4484 | Tag |04  00                                                                   |     | 
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10564 |      16388 | Tag |73  84  18  c2  2d                                                       |     | 
      19200 |      29728 | Rdr |93  70  73  84  18  c2  2d  8b  ce                                       |  ok | SELECT_UID
      30788 |      34308 | Tag |08  b6  dd                                                               |     | 
      36608 |      41376 | Rdr |61  00  2d  62                                                           |  ok | AUTH-B(0)
      42820 |      47556 | Tag |b0  9a  9f  73                                                           |     | AUTH: nt 
      57088 |      66464 | Rdr |07! e6! c6  41! d0  d3! 22! b4!                                          |     | AUTH: nr ar (enc)
      67524 |      72260 | Tag |c9  db! 2b  b4!                                                          |     | AUTH: at (enc)
      78208 |      82912 | Rdr |74  f8  9b! 9f                                                           |     | 
            |            |  *  |                                              key FFFFFFFFFFFF prng WEAK |     |
            |            |  *  |30  00  02  A8                                                           |  ok | READBLOCK(0)
      84036 |     104900 | Tag |16  5a! a9! 23  c9! 8d  b5  54  52! 55  16! 40! 0e  f3  d0! 97  11  94   |     | 
            |            |  *  |73  84  18  C2  2D  08  04  00  03  4D  AC  2E  3A  00  46  90  E0  47   |  ok | 
     118144 |     122848 | Rdr |df  e1! fc! 8d                                                           |     | 
            |            |  *  |50  00  57  CD                                                           |  ok | HALT

```

iceman1001, 2022-08-19:
so you get a different nonce each time...
nothing makes sense,   your device is flashed with same firmware as your pm3 client. (from same build)

you don't run on a M1/ Arm based cpu
you have enough RAM
you are connected with USB to pm3

I am afraid I can not figure out what is going on here.
Maybe someone else might find something but I will give up now.

iceman1001, 2022-08-19:
I see that you still haven't run the init script on your device, but that shouldn't matter,   

```
mem info
mem spiffs info
mem spiffs tree
```

przytular, 2022-08-21:
So maybe this may be hardware / problem with device itself?

iceman1001, 2022-08-21:
if you have rdv4,  the spiffs might be of concern.   Hence I wanted you to test it

atkfromabove, 2022-08-22:
> if you have rdv4, the spiffs might be of concern. Hence I wanted you to test it

Just ran the mem spiff commands and re-ran the tests.

Same result. The device errors out still.

iceman1001, 2022-08-23:
I am afraid I have no idea whats wrong.

iceman1001, 2022-08-24:
The others who has had issue with only getting one nonce,   has tried running the key recovery against a MIFARE Plus card.
It depends on which Security Level the card is in.   For instance SL3 will not work with MFC commands.

I will close this issue now.

closed 2022-08-24 by iceman1001

mywalkb, 2022-10-14:
I have the same issue how also #1760, when retrieve nonces they are normal and not predictable, while in MifareAcquireEncryptedNonces receivedAnswer is always 0020F414, maybe @pwpiwi can clear this wrong behaviour?
I have many mifare and only this card has this issue, it's very simple white mifare.

It's SL1 as all others working mifare
```
[=] --- Security Level (SL)
[+]        SL mode: SL1
[=]   SL 1: backwards functional compatibility mode (with MIFARE Classic 1K / 4K) with an optional AES authentication
```

atkfromabove, 2022-10-20:
> I have the same issue how also #1760, when retrieve nonces they are normal and not predictable, while in MifareAcquireEncryptedNonces receivedAnswer is always 0020F414, maybe @pwpiwi can clear this [...]

I'd enjoy seeing this issue re-opened since I am still having the problem

radeksh, 2022-11-10:
@iceman1001 what I see on other device this card is Mifare Classic 1K - is it suitable for proxmark3?
@mywalkb I have same, simple white card
@atkfromabove unfortunatelly it isn't re-opened, and in my opinion it probably should as 3 users already reporting same issue.

I'm trying to find common ground on all these cases

p.s. it's funny because it's my first card i've tried to use with proxmark and it isn't working, what a luck :sweat_smile: 
[image]
[image]
[image]

iceman1001, 2022-11-10:
if its one card which give  "1"  nonce in hardnested,   this is a clone MFC which has shown peculiar behavior which hardnested and static nested can't not solve at the moment. 

Not a issue related to the source code.

radeksh, 2022-11-10:
@iceman1001 yeah, that's the one
now I see, thank's for explanation!

iceman1001, 2022-11-11:
Feel free to find a solution for it

iceman1001, 2022-11-15:
Got another card with a static encrypted nonce.

[image]