RfidResearchGroup/proxmark3 #1753: O's instead of 0's

leeroywking, 2022-08-13:
https://github.com/RfidResearchGroup/proxmark3/blob/003623e5a731a40eb477a2dee964bbebe6477dae/client/dictionaries/mfc_default_keys.dic#L1698-L1725

O's are not hexadecimal values I suspect 0's were intended but I don't want to break the data if they were supposed to be something else. 

@UberGuidoZ  can you take a look and confirm these?

UberGuidoZ, 2022-08-13:
100% correct to correct this. It had been corrected on my own repo but the PR was lost getting over here.

I have an existing PR open to add more keys (already verified to not have the same issue) and will see if I can add this. Thank you!

UberGuidoZ, 2022-08-13:
This has been added to https://github.com/RfidResearchGroup/proxmark3/pull/1749

Small L instead of 1 was also replaced. (This was an OCR from a screengrab, hence the issues.) Thank you @leeroywking!

leeroywking, 2022-08-14:
Good catch I didn't notice that one, and thank you!

closed 2022-08-16 by iceman1001