RfidResearchGroup/proxmark3 #1754: 14b tag not detected. It was in the past.....

tcccorp, 2022-08-15:
hello, 

I bought a RDV4 around 25 of june. it worked fine until middle a July especially for a 14b tag.

after this date,  my proxmark was unable to detect any 14b tags  ( it is OK for other tags )  ... These tags can be read by my flipper or with a proxmark easy.

As I updated firmware each time I runned pm3,  I believed than could be an issue with the soft, so I downgraded firwmare with version around 25 of June, without success.

I was on Discord to try to find an answer and after several exchange, thank Icemann :),  no trouble was found

I saw with the vendor and a new hardware was sent to me and result was...... same problem as with the first RDV4 ( I checked every version from 25 of june to now, I checked with Ubuntu 18 and 20, with Kali 2022.3  on another computer, I tested on a virtual machine and even under windows ( yes I tried too....) ) .... same, this tag is not seen . I can read other tag as mifare  not not this king of tag .

 maybe I am very unlucky...  so I open this issue to see if someone has the same trouble 

about my hardware/software 
[usb] pm3 --> hw version 

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-872-g003623e5a 2022-08-13 10:05:44 27dbc5195
  compiled with............. GCC 9.4.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.14831-872-g003623e5a 2022-08-14 00:17:46 27dbc5195
       os: Iceman/master/v4.14831-872-g003623e5a 2022-08-14 00:17:58 27dbc5195
  compiled with GCC 10.3.1 20210621 (release)

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 65% used )

[usb] pm3 --> hw status 
[#] Memory
[#]   BigBuf_size............. 40904
[#]   Available memory........ 40904
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD5697C3097637023
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 307712
[#]   Transfer Speed PM3 -> Client... 615424 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31192 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#] 

when I try to run a hf search
[usb] pm3 --> hf search 
[!] ⚠️  No known/supported 13.56 MHz tags found

with a another kind of tag
[usb] pm3 --> hf search
 🕓  Searching for ISO14443-A tag...          
[+]  UID: XX XX XX XX
[+] ATQA: 00 XX
[+]  SAK: XX [2]
[+] Possible types:
[+]    MIFARE Classic 1K
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Prng detection: hard
[=] 
[=] --- Tag Signature
[=]  IC signature public key name: NXP Mifare Classic MFC1C14_x
[=] IC signature public key value: xxxxxx
[=]     Elliptic curve parameters: NID_secp128r1
[=]              TAG IC Signature: xxxxxx
[+]        Signature verification: successful
[?] Hint: try `hf mf` commands

[+] Valid ISO 14443-A tag found

[+] ATQA: 00 04
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Prng detection: hard
[=] 
[=] --- Tag Signature
[=]  IC signature public key name: NXP Mifare Classic MFC1C14_x
[=] IC signature public key value: 044F6D3F294DEA5737F0F46FFEE88A356EED95695DD7E0C27A591E6F6F65962BAF
[=]     Elliptic curve parameters: NID_secp128r1
[=]              TAG IC Signature: DA6B812B13C9B21CD4C2C4191F9EFDB5DD51BE06D698C67B092CAAAAC7A7D28F
[+]        Signature verification: successful
[?] Hint: try `hf mf` commands

[+] Valid ISO 14443-A tag found

hw tune 

[image]

iceman1001, 2022-08-23:
I still can't replicate your issue,  since I can read 14B tags with my RDV4 on latest source.

tcccorp, 2022-08-24:
hello Iceman, 

Thanks for your answer 

I tested with another RDV4 and still same issue.... I don't know if it is the same hardware as you. 

last Monday I pulled the latest version from scratch and after a make clean, make -j  and sudo make install, I updated firmawe and run pm3 with result as below 

[image]

if you want I can send to you one RDV4, maybe it is an electronic trouble, maybe it is me the trouble....I don't know

iceman1001, 2022-08-24:
`hf 14b reader -@`   and keep the tag close to the antenna,   try some different positions

iceman1001, 2023-08-29:
the 14B got some love where the shallow mode now works on RDV4 / ISO14443B / ISO15693,   I think this fix solves this issue.

closed 2023-08-29 by iceman1001