RfidResearchGroup/proxmark3 #1760: hardnested random number is always 1

xbebhxx3, 2022-08-19:
Describe the bug

When I run hf mf hardnested --blk 0 -b -k ced0cfc1bbbf random number is always 1.

To Reproduce

run hf mf hardnested --blk 0 -b -k ced0cfc1bbbf.

Expected behavior
Found hidden mifare keys.

Screenshots
[image]

Desktop (please complete the following information):

OS: windows 10
inside proxmark3 client run the following commands and paste the output here.
hw version
[image]

hw status
[image]

data tune
[image]

iceman1001, 2022-08-19:
Which proxmark3 device do you have?

I see that you are not on latest source,    please pull latest, compile, flash and test.

I have tried it on a RDV2 with firmware PM3_GENERIC and it works.

xbebhxx3, 2022-08-19:
proxmark3 easy
I have recompiled the latest version, but the problem persists

![1660900741481](https://user-images.githubusercontent.com/102897690/185587677-0328b0ea-176f-4661-b56b-ed4460848a81.jp
[image]
g)

xbebhxx3, 2022-08-19:
Only the current card has problems, other cards do not

iceman1001, 2022-08-19:
Which card is it?

xbebhxx3, 2022-08-19:
It may be the 3rd generation without loopholes card made in China

atkfromabove, 2022-08-22:
Is this the same issue as the other Hardnested ticket?

https://github.com/RfidResearchGroup/proxmark3/issues/1734

xbebhxx3, 2022-08-22:
Only 2 I've come across have this situation

xbebhxx3, 2022-08-22:
I would like to know how to read this kind of card

xbebhxx3, 2022-08-22:
> Is this the same issue as the other Hardnested ticket?
> 
> #1734

mine is not stuck

iceman1001, 2022-08-23:
If you can post a picture of the card you are looking at,  that would be helpful to identify which system its being used with.
Otherwise I am afraid I have no idea whats wrong.

xbebhxx3, 2022-08-23:
[image]

iceman1001, 2022-08-24:
`hf mfp info`  ?

xbebhxx3, 2022-08-24:
[=] --- Tag Information ---------------------------
    [=] -------------------------------------------------------------
    [!!] No card response.
    
    [+]  UID: 53 2E 97 B9
    [+] ATQA: 00 04
    [+]  SAK: 08 [2]
    [+] Possible types:
    [+]    MIFARE Classic 1K
    [=] proprietary non iso14443-4 card found, RATS not supported
    [+] Prng detection: weak
    [#] Auth error
    [?] Hint: try `hf mf` commands
    
    [!!] No card response.
    [=] --- Fingerprint
    [=]           SIZE: 2K (4 UID)
    [=]             SAK: 2K 7b UID
    [=] --- Security Level (SL)
    [+]        SL mode: SL1
    [=]   SL 1: backwards functional compatibility mode (with MIFARE Classic 1K / 4K) with an optional AES authentication

iceman1001, 2022-08-24:
ok,  so the card isn't a MFC.   

No luck for you.

closed 2022-08-24 by iceman1001

xbebhxx3, 2022-08-24:
So how do I crack it

iceman1001, 2022-08-24:
use the forum or discord to ask user related questions.

huanyingu, 2022-10-14:
I have the same problem. Is it resolved?