RfidResearchGroup/proxmark3 #1762: make clean && make all  error kali 2022 rolling

cucit, 2022-08-29:
-] CC src/mifare/aiddesfire.c
[-] CC src/aidsearch.c
[-] CC src/atrs.c
[-] CC src/cmdanalyse.c
[-] CC src/cmdcrc.c
[-] CC src/cmddata.c
[-] CC src/cmdflashmem.c
[-] CC src/cmdflashmemspiffs.c
[-] CC src/cmdhf.c
[-] CC src/cmdhf14a.c
[-] CC src/cmdhf14b.c
[-] CC src/cmdhf15.c
[-] CC src/cmdhfcryptorf.c
[-] CC src/cmdhfepa.c
[-] CC src/cmdhfemrtd.c
[-] CC src/cmdhffelica.c
[-] CC src/cmdhffido.c
[-] CC src/cmdhffudan.c
[-] CC src/cmdhfgallagher.c
[-] CC src/cmdhfksx6924.c
[-] CC src/cmdhfcipurse.c
[-] CC src/cmdhficlass.c
[-] CC src/cmdhflegic.c
[-] CC src/cmdhfjooki.c
[-] CC src/cmdhflist.c
[-] CC src/cmdhflto.c
[-] CC src/cmdhfmf.c
[-] CC src/cmdhfmfdes.c
[-] CC src/cmdhfmfhard.c
src/cmdhfmfhard.c:33:10: fatal error: bzlib.h: No such file or directory
   33 | #include <bzlib.h>
      |          ^~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:920: obj/cmdhfmfhard.o] Error 1
make: *** [Makefile:158: client/all] Error 2

mwalker33, 2022-08-29:
Did you try to install something that would provide the bzlib ?
e.g.
https://installati.one/kalilinux/libghc-bzlib-dev/
Note: I have not tested, but being the -dev i would expect to to provide the needed files.

cucit, 2022-08-29:
i tried and it did not work

it:1 http://kali.download/kali kali-rolling InRelease
Reading package lists... Done

???(kali?kali)-[~]
??$ sudo apt-get -y install libghc-bzlib-dev

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libghc-bzlib-dev

???(kali?kali)-[~]
??$ sudo apt update

Hit:1 http://kali.download/kali kali-rolling InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
205 packages can be upgraded. Run 'apt list --upgradable' to see them.

???(kali?kali)-[~]
??$ sudo apt -y install libghc-bzlib-dev

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libghc-bzlib-dev
                                                            ?

https://installati.one/kalilinux/libghc-bzlib-dev/

mwalker33, 2022-08-30:
Sorry, I don't have that os installed so cant check things.
Does this command work ?

sudo yum whatprovides */bzlib.h

iceman1001, 2022-08-30:
Look like some dependencies is not installed.  Which setup guide did you follow ?

This is the correct one to follow instructions 
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md

iceman1001, 2022-09-04:
ping @cucit 
If no answer I will close this issue due to inactivity

cucit, 2022-09-04:
Thanks for the help. All good now
Please close

closed 2022-09-05 by iceman1001