RfidResearchGroup/proxmark3 #1764: Apple Magsafe Support

lsylx, 2022-08-30:
I've just made some sniff with my Apple Magsafe Devices.

It seems that Apple have used some private APDU commands? 

__Apple Magsafe Wallet With Find My__
```
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 8853 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |7a(7)                                                                    |     | MAGSAFE WUPA
      76496 |      77488 | Rdr |7b(7)                                                                    |     | MAGSAFE WUPA
     152992 |     153984 | Rdr |7c(7)                                                                    |     | MAGSAFE WUPA
     155236 |     157604 | Tag |04  00                                                                   |     | 
     164768 |     169536 | Rdr |50  00  57  cd                                                           |  ok | HALT
     192096 |     193088 | Rdr |7f(7)                                                                    |     | 
     194340 |     196708 | Tag |04  00                                                                   |     | 
     203872 |     206336 | Rdr |93  20                                                                   |     | ANTICOLL
     207524 |     213412 | Tag |08  d7  9b  5d  19                                                       |     | 
     220512 |     231040 | Rdr |93  70  08  d7  9b  5d  19  f5  62                                       |  ok | SELECT_UID
     232228 |     235812 | Tag |20  fc  70                                                               |     | 
     243504 |     248272 | Rdr |50  00  57  cd                                                           |  ok | HALT
     270384 |     271376 | Rdr |7c(7)                                                                    |     | MAGSAFE WUPA
     281488 |     282480 | Rdr |7f(7)                                                                    |     | 
     283732 |     286100 | Tag |04  00                                                                   |     | 
     293264 |     295728 | Rdr |93  20                                                                   |     | ANTICOLL
     296916 |     302804 | Tag |08  d7  9b  5d  19                                                       |     | 
     309904 |     320432 | Rdr |93  70  08  d7  9b  5d  19  f5  62                                       |  ok | SELECT_UID
     321620 |     325204 | Tag |20  fc  70                                                               |     | 
     332896 |     337664 | Rdr |e0  80  31  73                                                           |  ok | RATS
     341668 |     349860 | Tag |05  7a  33  a0  02  ce  45                                               |  ok | 
     358032 |     375408 | Rdr |02  00  a4  04  00  07  d2  76  00  00  85  01  01  a6  09               |  ok | 
     406100 |     411924 | Tag |02  90  00  f1  09                                                       |     | 
     419216 |     430832 | Rdr |03  00  a4  00  0c  02  e1  03  d2  af                                   |  ok | 
     438484 |     444308 | Tag |03  90  00  2d  53                                                       |     | 
     451600 |     460912 | Rdr |02  00  b0  00  00  0f  8e  a6                                           |  ok | 
     469460 |     492628 | Tag |02  00  15  20  00  f0  00  f0  04  06  e1  04  00  f0  00  ff  90  00   |     | 
            |            |     |60  b3                                                                   |  ok | 
     499856 |     509232 | Rdr |03  00  b0  00  0f  06  ac  bc                                           |  ok | 
     517716 |     530516 | Tag |03  0f  04  57  81  09  0c  90  00  73  0e                               |  ok | 
     679024 |     680080 | Rdr |26(7)                                                                    |     | REQA
    1563792 |    1564784 | Rdr |7c(7)                                                                    |     | MAGSAFE WUPA
    1566036 |    1568404 | Tag |04  00                                                                   |     | 
    1575568 |    1580336 | Rdr |50  00  57  cd                                                           |  ok | HALT
    1602880 |    1603872 | Rdr |7f(7)                                                                    |     | 
    1605124 |    1607492 | Tag |04  00                                                                   |     | 
    1614656 |    1617120 | Rdr |93  20                                                                   |     | ANTICOLL
    1618308 |    1624196 | Tag |08  06  f7  d3  2a                                                       |     | 
    1631296 |    1641760 | Rdr |93  70  08  06  f7  d3  2a  1c  33                                       |  ok | SELECT_UID
    1643012 |    1646596 | Tag |20  fc  70                                                               |     | 
    1654288 |    1659056 | Rdr |50  00  57  cd                                                           |  ok | HALT
    1681184 |    1682176 | Rdr |7c(7)                                                                    |     | MAGSAFE WUPA
    1692272 |    1693264 | Rdr |7f(7)                                                                    |     | 
    1694516 |    1696884 | Tag |04  00                                                                   |     | 
    1704048 |    1706512 | Rdr |93  20                                                                   |     | ANTICOLL
    1707700 |    1713588 | Tag |08  06  f7  d3  2a                                                       |     | 
    1720688 |    1731152 | Rdr |93  70  08  06  f7  d3  2a  1c  33                                       |  ok | SELECT_UID
    1732404 |    1735988 | Tag |20  fc  70                                                               |     | 
    1743696 |    1748464 | Rdr |e0  80  31  73                                                           |  ok | RATS
    1752468 |    1760660 | Tag |05  7a  33  a0  02  ce  45                                               |  ok | 
    1839312 |    1857840 | Rdr |02  00  a4  04  00  07  d2  76  00  00  85  01  01  00  35  c0           |  ok | 
    1888532 |    1894356 | Tag |02  90  00  f1  09                                                       |     | 
    1943792 |    1955408 | Rdr |03  00  a4  00  0c  02  e1  03  d2  af                                   |  ok | 
    1963060 |    1968884 | Tag |03  90  00  2d  53                                                       |     | 
    2017984 |    2027360 | Rdr |02  00  b0  00  00  00  79  5e                                           |  ok | 
    2035844 |    2065860 | Tag |02  00  15  20  00  f0  00  f0  04  06  e1  04  00  f0  00  ff  0f  04   |     | 
            |            |     |57  81  09  0c  90  00  9b  12                                           |  ok | 
[... 428 lines cut ...]
   46461524 |   46467412 | Tag |08  ca  19  b5  6e                                                       |     | 
   46474512 |   46485040 | Rdr |93  70  08  ca  19  b5  6e  ee  b3                                       |  ok | SELECT_UID
   46486228 |   46489812 | Tag |20  fc  70                                                               |     | 
   46497520 |   46502288 | Rdr |50  00  57  cd                                                           |  ok | HALT
   46524400 |   46525392 | Rdr |7c(7)                                                                    |     | MAGSAFE WUPA
   46535504 |   46536496 | Rdr |7f(7)                                                                    |     | 
   46537748 |   46540116 | Tag |04  00                                                                   |     | 
   46547280 |   46549744 | Rdr |93  20                                                                   |     | ANTICOLL
   46550932 |   46556820 | Tag |08  ca  19  b5  6e                                                       |     | 
[... cut ...]
   46724756 |   46747924 | Tag |02  00  15  20  00  f0  00  f0  04  06  e1  04  00  f0  00  ff  90  00   |     | 
            |            |     |60  b3                                                                   |  ok | 
   46755152 |   46764528 | Rdr |03  00  b0  00  0f  06  ac  bc                                           |  ok | 
   46773012 |   46785812 | Tag |03  0f  04  57  81  09  0c  90  00  73  0e                               |  ok | 
   46934304 |   46935360 | Rdr |26(7)                                                                    |     | REQA
[usb] pm3 --> 
```

lsylx, 2022-08-30:
Here is the 
__Apple Magsafe Battery Pack sniff__
```
[usb] pm3 --> hf 14a list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 778 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |7a(7)                                                                    |     | MAGSAFE WUPA
       2244 |       4612 | Tag |04  03                                                                   |     | 
      11776 |      16544 | Rdr |50  00  57  cd                                                           |  ok | HALT
      39088 |      40080 | Rdr |7d(7)                                                                    |     | MAGSAFE WUPA
      41332 |      43700 | Tag |04  03                                                                   |     | 
      50864 |      53328 | Rdr |93  20                                                                   |     | ANTICOLL
      54516 |      60340 | Tag |08  64  54  8a  b2                                                       |     | 
      67504 |      77968 | Rdr |93  70  08  64  54  8a  b2  13  6f                                       |  ok | SELECT_UID
      79220 |      82804 | Tag |20  fc  70                                                               |     | 
      90496 |      95264 | Rdr |50  00  57  cd                                                           |  ok | HALT
     117392 |     118384 | Rdr |7a(7)                                                                    |     | MAGSAFE WUPA
     128496 |     129488 | Rdr |7d(7)                                                                    |     | MAGSAFE WUPA
     130740 |     133108 | Tag |04  03                                                                   |     | 
     140272 |     142736 | Rdr |93  20                                                                   |     | ANTICOLL
     143924 |     149748 | Tag |08  64  54  8a  b2                                                       |     | 
     156912 |     167376 | Rdr |93  70  08  64  54  8a  b2  13  6f                                       |  ok | SELECT_UID
     168628 |     172212 | Tag |20  fc  70                                                               |     | 
     179904 |     184672 | Rdr |e0  80  31  73                                                           |  ok | RATS
     185860 |     195140 | Tag |06  77  77  71  02  80  be  6a                                           |  ok | 
     207744 |     225120 | Rdr |02  00  a4  04  00  07  d2  76  00  00  85  01  01  a6  09               |  ok | 
     233284 |     239108 | Tag |02  90  00  f1  09                                                       |     | 
     246400 |     258016 | Rdr |03  00  a4  00  0c  02  e1  03  d2  af                                   |  ok | 
     260676 |     266500 | Tag |03  90  00  2d  53                                                       |     | 
     273792 |     283104 | Rdr |02  00  b0  00  00  0f  8e  a6                                           |  ok | 
     289988 |     313092 | Tag |02  00  1d  20  01  00  00  ff  04  06  e1  04  01  00  00  00  90  00   |     | 
            |            |     |ff  96                                                                   |  ok | 
     320384 |     329696 | Rdr |03  00  b0  00  0f  0e  e4  30                                           |  ok | 
     336580 |     358532 | Tag |03  05  06  e1  05  00  80  82  82  0f  04  42  32  00  79  90  00  09   |     | 
            |            |     |42                                                                       |  ok | 
     507136 |     508128 | Rdr |7b(7)                                                                    |     | MAGSAFE WUPA
     583648 |     584640 | Rdr |7c(7)                                                                    |     | MAGSAFE WUPA
     660128 |     661184 | Rdr |26(7)                                                                    |     | REQA
   15016692 |   15019060 | Tag |04  03                                                                   |     | 
   15026240 |   15031008 | Rdr |50  00  57  cd                                                           |  ok | HALT
   15053552 |   15054544 | Rdr |7d(7)                                                                    |     | MAGSAFE WUPA
   15055796 |   15058164 | Tag |04  03                                                                   |     | 
   15065328 |   15067792 | Rdr |93  20                                                                   |     | ANTICOLL
   15068980 |   15074868 | Tag |08  01  90  94  0d                                                       |     | 
   15093684 |   15097268 | Tag |20  fc  70                                                               |     | 
   15131856 |   15132848 | Rdr |7a(7)                                                                    |     | MAGSAFE WUPA
   15142960 |   15143952 | Rdr |7d(7)                                                                    |     | MAGSAFE WUPA
[... cut ...]
            |            |     |ff  96                                                                   |  ok | 
   15334848 |   15344160 | Rdr |03  00  b0  00  0f  0e  e4  30                                           |  ok | 
   15351044 |   15372996 | Tag |03  05  06  e1  05  00  80  82  82  0f  04  42  32  00  79  90  00  09   |     | 
            |            |     |42                                                                       |  ok | 
   15521584 |   15522576 | Rdr |7b(7)                                                                    |     | MAGSAFE WUPA
```

lsylx, 2022-08-30:
Another thing interesting is that if two magsafe devices in the field, the iphone can both recognize.
```
usb] pm3 --> hf 14a list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 10448 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |7a(7)                                                                    |     | MAGSAFE WUPA
       2244 |       4612 | Tag |04  03                                                                   |     | 
      11776 |      16544 | Rdr |50  00  57  cd                                                           |  ok | HALT
      39104 |      40096 | Rdr |7d(7)                                                                    |     | MAGSAFE WUPA
      41348 |      43716 | Tag |04  03                                                                   |     | 
      50880 |      53344 | Rdr |93  20                                                                   |     | ANTICOLL
      54532 |      60420 | Tag |08  36  c4  e9  13                                                       |     | 
      67520 |      78048 | Rdr |93  70  08  36  c4  e9  13  b4  f1                                       |  ok | SELECT_UID
      79236 |      82820 | Tag |20  fc  70                                                               |     | 
      90512 |      95280 | Rdr |50  00  57  cd                                                           |  ok | HALT
     117392 |     118384 | Rdr |7a(7)                                                                    |     | MAGSAFE WUPA
     128496 |     129488 | Rdr |7d(7)                                                                    |     | MAGSAFE WUPA
     130740 |     133108 | Tag |04  03                                                                   |     | 
     140288 |     142752 | Rdr |93  20                                                                   |     | ANTICOLL
     143940 |     149828 | Tag |08  36  c4  e9  13                                                       |     | 
     156928 |     167456 | Rdr |93  70  08  36  c4  e9  13  b4  f1                                       |  ok | SELECT_UID
     168644 |     172228 | Tag |20  fc  70                                                               |     | 
     179920 |     184688 | Rdr |e0  80  31  73                                                           |  ok | RATS
     185876 |     195156 | Tag |06  77  77  71  02  80  be  6a                                           |  ok | 
     207744 |     225120 | Rdr |02  00  a4  04  00  07  d2  76  00  00  85  01  01  a6  09               |  ok | 
     233284 |     239108 | Tag |02  90  00  f1  09                                                       |     | 
     246400 |     258016 | Rdr |03  00  a4  00  0c  02  e1  03  d2  af                                   |  ok | 
     260676 |     266500 | Tag |03  90  00  2d  53                                                       |     | 
     273792 |     283104 | Rdr |02  00  b0  00  00  0f  8e  a6                                           |  ok | 
     289988 |     313092 | Tag |02  00  1d  20  01  00  00  ff  04  06  e1  04  01  00  00  00  90  00   |     | 
            |            |     |ff  96                                                                   |  ok | 
     320384 |     329696 | Rdr |03  00  b0  00  0f  0e  e4  30                                           |  ok | 
     336580 |     358532 | Tag |03  05  06  e1  05  00  80  82  82  0f  04  42  32  00  79  90  00  09   |     | 
            |            |     |42                                                                       |  ok | 
     507120 |     508112 | Rdr |7b(7)                                                                    |     | MAGSAFE WUPA
     583616 |     584608 | Rdr |7c(7)                                                                    |     | MAGSAFE WUPA
     585860 |     588228 | Tag |04  00                                                                   |     | 
     595392 |     600160 | Rdr |50  00  57  cd                                                           |  ok | HALT
     622720 |     623712 | Rdr |7f(7)                                                                    |     | 
     624964 |     627332 | Tag |04  00                                                                   |     | 
     634496 |     636960 | Rdr |93  20                                                                   |     | ANTICOLL
     638148 |     644036 | Tag |08  e0  b2  e3  b9                                                       |     | 
     651136 |     661664 | Rdr |93  70  08  e0  b2  e3  b9  77  e3                                       |  ok | SELECT_UID
     662852 |     666436 | Tag |20  fc  70                                                               |     | 
     674128 |     678896 | Rdr |50  00  57  cd                                                           |  ok | HALT
     701008 |     702000 | Rdr |7c(7)                                                                    |     | MAGSAFE WUPA
[... cut ...]
   50112068 |   50135236 | Tag |02  00  15  20  00  f0  00  f0  04  06  e1  04  00  f0  00  ff  90  00   |     | 
            |            |     |60  b3                                                                   |  ok | 
   50142464 |   50151840 | Rdr |03  00  b0  00  0f  06  ac  bc                                           |  ok | 
   50159940 |   50172740 | Tag |03  0f  04  57  81  09  0c  90  00  73  0e                               |  ok | 
   50321232 |   50322288 | Rdr |26(7)                                                                    |     | REQA
[usb] pm3 --> 
```

iceman1001, 2022-08-30:
Its easier to do `trace save -f xxxx`   and share as file,    then anyone can look at the trace locally.

And the protocol is unknown,  so if you don't have some kind of datasheet,   there isn't much to conclude.

Usually I suggest you join the discord server and involve the community that way.

iceman1001, 2022-12-05:
Ping!
How is this going?

iceman1001, 2025-06-08:
Ping ,   those trace files would have been great to have in the project.

Meanwhile closing because of lack of interest from OP

closed 2025-06-08 by iceman1001