RfidResearchGroup/proxmark3 #1773: HID C15001  KeyScan 36-bit - incorrect Wiegand format displayed

Githubby14, 2022-09-06:
just pulled the latest repo and noticed it no longer displaces the correct format for HID C15001

```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-916-g429139f17-dirty-unclean 2022-09-01 13:39:25 260a57537
  compiled with............. MinGW-w64 10.3.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.14831-916-g429139f17-dirty-unclean 2022-09-01 13:38:41 260a57537
       os: Iceman/master/v4.14831-916-g429139f17-dirty-unclean 2022-09-01 13:38:53 260a57537
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 59% used )

```
example 
```
[usb] pm3 --> lf hid clone -w C15001 --fc 118 --cn 1603
[=] Preparing to clone HID tag
[+] [C15001  ] HID KeyScan 36-bit               FC: 118  CN: 1603  OEM: 900  parity ( ok )
[=] Done
[?] Hint: try `lf hid reader` to verify
```

The raw hex is written correctly but when you do a search it doesn't recognize the correct bit and wiegand format. It now comes things its a possible 37-bit format not the correct 36 bit C15001.

```
[usb] pm3 --> lf search

[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
hid preamble detected
[+] [H10302  ] HID H10302 37-bit huge ID          CN: 32287098435  parity ( fail )
[+] [H10304  ] HID H10304 37-bit                FC: 61582  CN: 394819  parity ( fail )
[+] [P10004  ] HID P10004 37-bit PCSC           FC: 7697  CN: 221284
[+] [MDI37   ] PointGuard MDI 37-bit            FC: 12  CN: 74843715  parity ( fail )
[=] found 4 matching formats
[+] DemodBuffer:
[+] 1D555AAA5595A9A555A59569

[=] raw: 000000000000003f08ec0c86

[+] Valid HID Prox ID found!

[+] Chipset detection: T55xx
[?] Hint: try `lf t55xx` commands
[usb] pm3 -->                      
```

iceman1001, 2023-03-16:
it has a sentinel bit problem.
if you ise  "wiegand decode --bin"   and use the binary data it easier to make one bit changes in front.

MrNorth907, 2023-09-26:
@ehinckleyCTO for the keyscan 36 bit format the basic are easy. Facility code starts at 11 runs to 18 (8bits). Card number starts at 19 runs to 34. (16 bits). 

Just to make sure we are on the same page, the first bit in the sequence is bit zero, not one.

I am looking to find the parity bit for this format. Does anyone one know?

iceman1001, 2025-02-20:
This should be fixed.  Try pulling latest and verify

closed 2025-02-20 by iceman1001