RfidResearchGroup/proxmark3 #1776: EM41x clones should have configurable data rates

afkamalipour, 2022-09-15:
The default option for cloning an EM41x to a T55xx currently sets the data rate to RF/64, but there are configurations that use other timings. The current workaround is to dump a cloned t55xx and manually modify the JSON dump to reflect the actual desired data rate in the configuration page 0 block 0.

Preferably the feature would allow you to use a command line option such as:

`lf t55 clone --id 0F0368568B --dr 32` valid dr values would be 16, 32, 64.

I can add this feature myself and initiate a pull request if that is welcomed!

afkamalipour, 2022-09-15:
Whoops! While looking through the source, I see that there's an argument for clk! Perfect!

Looks like all that could help is an update to the command's help text!

closed 2022-09-15 by afkamalipour

iceman1001, 2022-09-16:
To me its pretty clear that there is a param called --clk

```
[usb] pm3 --> lf em 410x clone -h

clone a EM410x ID to a T55x7, Q5/T5555 or EM4305/4469 tag.

usage:
    lf em 410x clone [-h] [--clk <dec>] --id <hex> [--q5] [--em]

options:
    -h, --help                     This help
    --clk <dec>                    <16|32|40|64> clock (default 64)
    --id <hex>                     EM Tag ID number (5 hex bytes)
    --q5                           optional - specify writing to Q5/T5555 tag
    --em                           optional - specify writing to EM4305/4469 tag

examples/notes:
    lf em 410x clone --id 0F0368568B             -> encode for T55x7 tag
    lf em 410x clone --id 0F0368568B --q5        -> encode for Q5/T5555 tag
    lf em 410x clone --id 0F0368568B --em        -> encode for EM4305/4469
    
  ```