RfidResearchGroup/proxmark3 #1780: src/cmdlfparadox.c:102:13: error: variable 'error' set but not used [-Werror,-Wunused-but-set-variable]
    uint8_t error = 0;

sun1ch, 2022-09-20:
I got that error when Im trying to compile client in Termux

I need solution of this problem.

Device Samsung Galaxy Tab S7 Android 12
[image]

iceman1001, 2022-09-20:
I pushed a fix,  
Try pulling latest and see if it compiles now

sun1ch, 2022-09-20:
Now error is

bignum.c:1442:29: error: variable 't' set but not used [-Werror,-Wunused-but-set-variable]
    mbedtls_mpi_uint c = 0, t = 0;
                            ^
1 error generated.
make[2]: *** [../../Makefile.host:95: ../../client/obj/mbedtls/bignum.o] Error 1
make[1]: *** [Makefile:866: obj/libmbedtls.a] Error 2
make: *** [Makefile:158: client/all] Error 2

[image]

iceman1001, 2022-09-20:
Feel free to contribute

closed 2022-09-20 by sun1ch