RfidResearchGroup/proxmark3 #1782: "UART:: write time-out" issue when hf reading

NikNitro, 2022-09-21:
Hi!

It's been over 8 months without using the Proxmark. I've downloaded the repository, make, make -j and ./pm3-flash-all and all went good.
Now, when I'm trying to scan any mifare card, it returns this error of the title. I'll show you the steps:

`┌──(aml㉿kali)-[~/tools/proxmark3]
└─$ ./pm3                          
[=] Session log /home/aml/.proxmark3/logs/log_20220921.txt
[+] loaded from JSON file /home/aml/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ❄ ]

  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 Kb ( 59% used )

    Client.... Iceman/master/v4.14831-960-gbf143219a 2022-09-21 15:49:53
    Bootrom... Iceman/master/v4.14831-960-gbf143219a 2022-09-21 15:49:33 
    OS........ Iceman/master/v4.14831-960-gbf143219a 2022-09-21 15:50:16 
    Target.... PM3 GENERIC

[usb] pm3 --> hf mf chk
[#] [ 0] key FF FF FF FF FF FF 
[EDITED for not spamming]
[#] [41] key E0 00 00 00 00 00 
[=] Start check for keys...
[=] ....[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out
.[!!] 🚨 UART:: write time-out

[=] time in checkkeys 80 seconds

[=] testing to read key B...

[+] found keys:

[+] -----+-----+--------------+---+--------------+----
[+]  Sec | Blk | key A        |res| key B        |res
[+] -----+-----+--------------+---+--------------+----
[+]  000 | 003 | ------------ | 0 | ------------ | 0
[+]  001 | 007 | ------------ | 0 | ------------ | 0
[+]  002 | 011 | ------------ | 0 | ------------ | 0
[+]  003 | 015 | ------------ | 0 | ------------ | 0
[+]  004 | 019 | ------------ | 0 | ------------ | 0
[+]  005 | 023 | ------------ | 0 | ------------ | 0
[+]  006 | 027 | ------------ | 0 | ------------ | 0
[+]  007 | 031 | ------------ | 0 | ------------ | 0
[+]  008 | 035 | ------------ | 0 | ------------ | 0
[+]  009 | 039 | ------------ | 0 | ------------ | 0
[+]  010 | 043 | ------------ | 0 | ------------ | 0
[+]  011 | 047 | ------------ | 0 | ------------ | 0
[+]  012 | 051 | ------------ | 0 | ------------ | 0
[+]  013 | 055 | ------------ | 0 | ------------ | 0
[+]  014 | 059 | ------------ | 0 | ------------ | 0
[+]  015 | 063 | ------------ | 0 | ------------ | 0
[+] -----+-----+--------------+---+--------------+----
[+] ( 0:Failed / 1:Success )

[!!] 🚨 UART:: write time-out
[!] ⚠  command execution time out
[usb] pm3 --> 
`

Once it happened, I need to restart both, the Proxmark and the client.

Hw status:
`[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 42780
[#]   Available memory........ 42780
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 283648
[#]   Transfer Speed PM3 -> Client... 567296 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31876 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] 
`

Hw tune:
`[usb] pm3 --> hw tune
[#] DEBUG: (setClockGrid) clear settings
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕐  20[#] Got Waiting Time eXtension request 1500 ms
 🕛   9[#] Got Waiting Time eXtension request 1500 ms

[=] ---------- LF Antenna ----------
[+] LF antenna: 41,03 V - 125,00 kHz
[+] LF antenna: 28,92 V - 134,83 kHz
[+] LF optimal: 41,59 V - 122,45 kHz
[+] Approx. Q factor (*): 6,2 by frequency bandwidth measurement
[+] Approx. Q factor (*): 12,1 by peak voltage measurement
[!] ⚠  Contradicting measures seem to indicate you're running a PM3GENERIC firmware on a RDV4
[!] ⚠  False positives is possible but please check your setup
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 28,37 V - 13.56 MHz
[+] Approx. Q factor (*): 8,2 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134,83 kHz, 95 (red) is 125,00 kHz.
`
Note: Although it says that I have a RDV4, if I install that firmware, the Proxmark bricks and I need to reinstall firmware in bootloader mode (pressing the button).

Thanks in advance.

iceman1001, 2022-09-21:
looks like you have set debug configuration a bit too aggressive.   But I can be wrong.

NikNitro, 2022-09-21:
Yes it is, but only because I'm a little desperated about whats happening. However, the "UART" error aldo appears in the normal debog mode.

Thanks for your answering

iceman1001, 2022-09-22:
I would suggest you head over to the discord server and ask.   Try different usb cables and usb ports.  Don't use a usb hub..  etc..

NikNitro, 2022-09-22:
Ok thank you, I'll do it, and if I solve the issue I will comment here.

Have a nice day!

closed 2022-09-22 by iceman1001