RfidResearchGroup/proxmark3 #1786: Trying to compile on Raspberry Pi Debian Jessie

hmax42, 2022-09-25:
**What I want to do:**
Compile Proxmark3 on a Raspberry Pi 2 running Debian Jessie.

**Why do I want to do this on this old platform:**
I have a Pimoroni Hyperpixel Display ( https://github.com/pimoroni/hyperpixel ) for Raspberry Pi which is incompatible with anything beyond Jessie.
I'd like to use this RasPi with the display as a small mobile cli-client for various hardwares.
I do not want to accept that the display is a piece of trash now.

**What did i do:**

1. Run 'apt-get upgrade' (/etc/apt/sources.list was updated to contain the archive and security hosts)
2. Clone https://github.com/RfidResearchGroup/proxmark3 
3. Create makefile.platform with PM3GENERIC
4. run 'make'
> ===================================================================
> Version info:      RRG/Iceman/HEAD/v4.14831-259-g3169bf2-dirty
> Platform name:     Proxmark3 generic target
> PLATFORM:          PM3GENERIC
> PLATFORM_FPGA:     xc2s30
> PLATFORM_SIZE:     512
> Platform extras:   No extra selected
> Included options:  LF HITAG EM4x50 EM4x70 ZX8211 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
> Standalone mode:   LF_SAMYRUN
> ===================================================================
> [*] MAKE client/all
> ===================================================================
> Version info:      RRG/Iceman/HEAD/v4.14831-259-g3169bf2-dirty
> Client platform:   Linux
> GUI support:       QT5 found, enabled (Qt version 5.3.2 in /usr/lib/arm-linux-gnueabihf)
> native BT support: Bluez found, enabled
> Jansson library:   system library not found, using local library
> Lua library:       system library not found, using local library
> Python3 library:   Python3 v3.4 found, enabled
> Readline library:  enabled
> Whereami library:  system library not found, using local library
> Lua SWIG:          wrapper found
> Python SWIG:       wrapper found
> compiler version:  gcc (Debian 4.9.2-10+deb8u2) 4.9.2
> ===================================================================
> [-] CC src/mifare/aiddesfire.c
> src/mifare/aiddesfire.c: In function ‘print_aiddf_description’:
> src/mifare/aiddesfire.c:282:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
>      for (uint32_t idx = 0; idx < json_array_size(root); idx++) {
>      ^
> src/mifare/aiddesfire.c:282:5: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
> Makefile:901: recipe for target 'obj/mifare/aiddesfire.o' failed
> make[1]: *** [obj/mifare/aiddesfire.o] Error 1
> Makefile:157: recipe for target 'client/all' failed
> make: *** [client/all] Error 2

following this advice i get this error

> [-] CC src/cmdhficlass.c
> src/cmdhficlass.c: In function ‘CmdHFiClassDump’:
> src/cmdhficlass.c:1777:9: error: missing initializer for field ‘use_elite’ of ‘i
> class_auth_req_t’ [-Werror=missing-field-initializers]
>          .req.use_elite = elite,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:58:10: note: ‘use_elite’ declared here
>      bool use_elite;
>           ^
> src/cmdhficlass.c:1778:9: error: missing initializer for field ‘use_credit_key’
> of ‘iclass_auth_req_t’ [-Werror=missing-field-initializers]
>          .req.use_credit_key = false,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:59:10: note: ‘use_credit_key’ declared here
>      bool use_credit_key;
>           ^
> src/cmdhficlass.c:1779:9: error: missing initializer for field ‘use_replay’ of ‘
> iclass_auth_req_t’ [-Werror=missing-field-initializers]
>          .req.use_replay = use_replay,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:60:10: note: ‘use_replay’ declared here
>      bool use_replay;
>           ^
> src/cmdhficlass.c:1780:9: error: missing initializer for field ‘send_reply’ of ‘iclass_auth_req_t’ [-Werror=missing-field-initializers]
>          .req.send_reply = true,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:61:10: note: ‘send_reply’ declared here
>      bool send_reply;
>           ^
> src/cmdhficlass.c:1781:9: error: missing initializer for field ‘do_auth’ of ‘iclass_auth_req_t’ [-Werror=missing-field-initializers]
>          .req.do_auth = auth,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:62:10: note: ‘do_auth’ declared here
>      bool do_auth;
>           ^
> src/cmdhficlass.c:1782:9: error: missing initializer for field ‘blockno’ of ‘iclass_auth_req_t’ [-Werror=missing-field-initializers]
>          .end_block = app_limit1,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:63:13: note: ‘blockno’ declared here
>      uint8_t blockno;
>              ^
> src/cmdhficlass.c: In function ‘iclass_write_block’:
> src/cmdhficlass.c:1947:9: error: missing initializer for field ‘use_elite’ of ‘iclass_auth_req_t’ [-Werror=missing-field-initializers]
>          .req.use_elite = elite,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:58:10: note: ‘use_elite’ declared here
>      bool use_elite;
>           ^
> src/cmdhficlass.c:1948:9: error: missing initializer for field ‘use_credit_key’ of ‘iclass_auth_req_t’ [-Werror=missing-field-initializers]
>          .req.use_credit_key = use_credit_key,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:59:10: note: ‘use_credit_key’ declared here
>      bool use_credit_key;
>           ^
> src/cmdhficlass.c:1949:9: error: missing initializer for field ‘use_replay’ of ‘iclass_auth_req_t’ [-Werror=missing-field-initializers]
>          .req.use_replay = replay,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:60:10: note: ‘use_replay’ declared here
>      bool use_replay;
>           ^
> src/cmdhficlass.c:1950:9: error: missing initializer for field ‘send_reply’ of ‘iclass_auth_req_t’ [-Werror=missing-field-initializers]
>          .req.blockno = blockno,
>          ^
> In file included from src/cmdhficlass.h:23:0,
>                  from src/cmdhficlass.c:19:
> ../include/iclass_cmd.h:61:10: note: ‘send_reply’ declared here
>      bool send_reply;
>           ^
> cc1: all warnings being treated as errors
> Makefile:901: recipe for target 'obj/cmdhficlass.o' failed
> make[1]: *** [obj/cmdhficlass.o] Error 1
> Makefile:157: recipe for target 'client/all' failed
> make: *** [client/all] Error 2

i removed the -Werror=missing-field-initializers from makefile.defs which led to 

> [-] CC src/cmdhficlass.c
> [-] CC src/cmdhflegic.c
> [-] CC src/cmdhfjooki.c
> [-] CC src/cmdhflist.c
> src/cmdhflist.c:58:1: error: missing braces around initializer [-Werror=missing-braces]
>  static uint8_t gs_mfuc_authdata[3][16] = {0};
>  ^
> src/cmdhflist.c:58:1: error: (near initialization for ‘gs_mfuc_authdata[0]’) [-Werror=missing-braces]
> cc1: all warnings being treated as errors
> Makefile:901: recipe for target 'obj/cmdhflist.o' failed
> make[1]: *** [obj/cmdhflist.o] Error 1
> Makefile:157: recipe for target 'client/all' failed
> make: *** [client/all] Error 2

i did not find the missing-braces parametre in Makefile.def so that became  the moment i stopped fiddling around and asked on discord, where i was advised to make a ticket.
i did not make a ticket at first, but went to discord, because i suspect my old platform is at fault here

**Further experiments**
I also tried to compile at an older commit revision 3169bf2763d6bfdfafc6b077ffac1cf839ff0a77
but that changes nothing

**Why this old revision:**
it compiles on my old macbook core2duo with kali linux and this is the firmware i have currently on my proxmark 3 easy 512

**Note**
the only c-programming i do usually is on arduino
i am a java-programmer by profession

I hope this is a not doomed cause!

iceman1001, 2022-09-25:
First, 
you say you cloned the repo,  but the source seems to a very old project
You have `259`  compared to my latest source which is  `939` 
```
Version info: RRG/Iceman/HEAD/v4.14831-259-g3169bf2-dirty
                                                                 ^^^
```

Second and the most disturbing is the fact you are using an ancient GCC compiler. 
```compiler version: gcc (Debian 4.9.2-10+deb8u2) 4.9.2``` 

GCC is up on v12 now,  I think we support 7.9 and forward.    I recommend you upgrade to a modem compiler and try again.

iceman1001, 2022-09-25:
On my ubuntu setup:
```
Version info:      Iceman/master/v4.14831-963-gba1def240-dirty
compiler version:  cc (Ubuntu 11.1.0-1ubuntu1~20.04) 11.1.0
``

hmax42, 2022-09-25:
Thank you for explaining

your note about the gcc version explains a lot, and the version of gcc is the main culprit, i think
the gcc i am using is the latest, the apt-get repo for jessie is providing

the revision of the proxmark code does not make any differences,
i started with a clean "newest" clone, then changed to "my" older revision which seems to have the number 259
i just not switch back to the newest version, when i created the log out put today for the git issue

iceman1001, 2022-09-25:
hm, does  `make clean;  make -j;`  do a difference for your revision output?

Let me know how a more modern GCC works for you

hmax42, 2022-09-27:
will try gcc 8.3.0 as a start
it will be interesting, i have never built a compiler with itself.

but i have to take a detour first, because
> configure: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.

iceman1001, 2022-09-27:
usually you can download gcc binaries.  
But compiling it also works.

hmax42, 2022-09-27:
where would i download binaries for a raspberry 2 armhf platform?
i thought that if apt-get cannot supply any newer version of gcc for this old platform than already installed (4.9...), i have to compile them myself.
remember i am not on an regular linux pc, but on a raspberry pi.

and the same goes for the 3 libs gmp, mpfr & mpc.
the versions of these libs apt-get would install were older than what gcc demanded
for each i did
>make
>sudo make install
that worked for all 3

at some point i got
> [libmpc.so.2: cannot open shared object file: No such file or directory](https://stackoverflow.com/questions/19625451/cc1-error-while-loading-shared-libraries-libmpc-so-2-cannot-open-shared-objec)
so i googled and then
>export LD_LIBRARY_PATH="/usr/local/lib"
>sudo ldconfig

now it fails at
>In file included from /usr/include/stdio.h:27,
>                 from ../../../gcc-8.3.0/libgcc/../gcc/tsystem.h:87,
>                 from ../../../gcc-8.3.0/libgcc/libgcc2.c:27:
>/usr/include/features.h:374:12: fatal error: sys/cdefs.h: Datei oder Verzeichnis nicht gefunden

i found this file in gcc-8.3.0/fixincludes/tests/base/sys/cdefs.h
and someone on stackoverflow suggested
>sudo cp /home/pi/gcc/gcc-8.3.0/fixincludes/tests/base/sys/cdefs.h /usr/include/sys/

but now gnu/stubs.h is missing
and i can't find that
so thats all for today

i see it this way:
if i cannot build the old versions of gcc on this old platform, then i surely won't be able to build a more current version as well!
i did download the tar.gz for gcc 8, 9 & 12.

hmax42, 2022-10-02:
i gave up - with jessie

i asked in the github for the hyperpixel(my reason for jessie) about the current state and seems like it works in buster now 

https://github.com/pimoroni/hyperpixel/issues/58

buster comes with gcc 8, which compiles the current version 
>Iceman/master/v4.14831-971-geeb83e112

but thank you for the support anyway

closed 2022-10-02 by hmax42

iceman1001, 2022-10-02:
Things gets EOL:d  eventually.  Hard to keep backward compatibility in the long run.

At least you tried.